Kinovarobotics/Kinova-kortex2_Gen3_G3L #3: send_joint_speeds does not work as expected

Achllle, 2019-02-20:
## Description

send_joint_speeds action doesn't work using code based on examples provided in `api_cpp` and `api_python` and using the auto-generated documentation as a reference

Side note: issue template didn't load automatically.

## Version

KortexAPI : 1.1.3
Kortex Device : Jaco GEN3 7DOF

## Steps to reproduce
### Python
1. edit the [move angular example](
        kortex/api_python/examples/102-BaseGen3_ctrl/01-move_angular_and_cartesian.py)
2. replace the joint angles example in angular movement with joint speeds code, see below
3. set angle_value variable to an array of non-zero values (e.g. [0,0,0,0,0,0,5]) for 5deg/s for the 7th DOF.

### c++
Same as Python steps, except with the [api_cpp example](kortex/api_cpp/examples/102-BaseGen3_ctrl/01-move_angular_and_cartesian.cpp).

## Code example

Note: I didn't change the names of the variables

### Python
When I convert any joint angle reference to joint speed references 1:1, i.e.

```
for joint_id in range(7):
    joint_angle = action.send_joint_speeds.joint_angles.joint_speeds.add()
    # joint_angle = action.send_joint_speeds.joint_speeds.add()
    joint_angle.joint_identifier = joint_id
    joint_angle.value = angle_value[joint_id]
    joint_angle.duration = 5
```
I get an `AttributeError: 'JointSpeeds' object has no attribute 'joint_angles'`. When I comment out the second line in the example above and replace it with the commented line right below it, I don't get the AttributeError (which I believe shouldn't have happened in the first place) and the code executes but doesn't move the last joint.

### c++
in `angularMovement()`:

```
    auto action = k_api::Base::Action();
    action.set_name("Example angular movement");
    action.set_application_data("");

    auto reachJointAngles = action.mutable_send_joint_speeds();
    auto jointAngles = reachJointAngles->mutable_joint_speeds();

    std::vector<float> angles = {0.0f, 0.0f, 0.0f, 0.0f, 0.0f, 0.0f, 5.0f};
    for(size_t i = 0 ; i < angles.size(); ++i)
    {
        auto jointAngle = jointAngles->add_joint_speeds();
        // auto jointAngle = reachJointAngles->add_joint_speeds();
        jointAngle->set_joint_identifier(i);
        jointAngle->set_value(angles.at(i));
    }
```
This results in 
```
error: ‘class google::protobuf::RepeatedPtrField<Kinova::Api::Base::JointSpeed>’ has no member named ‘add_joint_speeds’
         auto jointAngle = jointAngles->add_joint_speeds();
```
Again, commenting out the first line in the for-loop and replacing with the commented line results in a successful build, but results in a segfault during execution on the robot.

## Expected behavior

The final joint rotates at ~5 deg/s for 5 seconds

sjhansen3, 2019-02-20:
@nsaif-kinova Here is the issue we referenced earlier.

alexvannobel, 2019-02-21:
Hi @Achllle !

Thanks for filling the issue template, it really helps troubleshooting your problem!

I found the solution. I'll explain it for the C++ example but it naturally works for the Python too.

Basically, the reach_joint_angles and send_joint_speeds Actions are not built the same way.
 - The reach_joint_angles Action has a [ConstrainedJointAngles field](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/references/msg_Base_ConstrainedJointAngles.md#), which you get a reference to with 
```auto reachJointAngles = action.mutable_reach_joint_angles();```
The ConstrainedJointAngles field, in turn, has a [JointAngles field](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/references/msg_Base_JointAngles.md#), which you get a reference to with 
```auto jointAngles = reachJointAngles->mutable_joint_angles();```
This is the array in which we add individual angles with the ```add_joint_angles()``` method.

 - The send_joint_speeds Action has a [JointSpeeds field](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/references/msg_Base_JointSpeeds.md#), which you could get a reference to with 
```auto jointSpeeds = action.mutable_send_joint_speeds();```. This is the array in which we add individual joint speeds with the ```add_joint_speeds()``` method. 

The fully functional C++ code snippet to control the arm with a send_joint_speeds Action would look like : 
```
auto jointSpeeds = action.mutable_send_joint_speeds();

std::vector<float> speeds = {0.0f, 0.0f, 0.0f, 0.0f, 0.0f, 0.0f, 5.0f}; // Actuator 7 rotates at 5deg/s for 5s
for(size_t i = 0 ; i < speeds.size(); ++i)
{
    auto jointSpeed = jointSpeeds->add_joint_speeds();
    jointSpeed->set_joint_identifier(i);
    jointSpeed->set_value(speeds.at(i));
    jointSpeed->set_duration(5);
}
``` 

As you saw, not all Actions are built the same way, so I strongly recommend you look at the [documentation](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/index.md) first when you start using a new Action, or more generally when you use a new Kortex API function. 

Tell me if this helped resolve your issue!

Cheers,

Alex

alexvannobel, 2019-08-13:
Hi @Achllle ,

The [SendJointSpeedsCommand](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md) function is now available with the 2.0 release. 
I'll close this issue, but feel free to open a new one if there is anything with the functionality.

Best,
Alex

closed 2019-08-13 by alexvannobel