Kinovarobotics/Kinova-kortex2_Gen3_G3L #5: Building cpp examples - kortex api not found

miscz, 2019-03-29:
I followed the setup instruction for the cpp API. Having the recommended folder structure

```
examples/kortex_api  
┬  
├ api_cpp/
└   ┬  
    ├ doc/
    ├ examples/     (Here to build all example)
    └   ┬  
        ├ 000-Getting_started/  (Here to build this specific directory)
        └ .../
```

I noticed an error while building the examples because the kortex api could not be found. Due to the defined path in CMakeLists `set(KORTEX_DIR "${PROJECT_SOURCE_DIR}/../kortex_api/${KORTEX_SUB_DIR}")` cmake was looking in the parent folder of e.g. 000-Geting_started for kortex api. Which does not exist there.

If someone is facing the same issue, just copy the example folders (000-Getting_started to 500-Vision) to the top folder named examples:

```
examples/
┬  
├ kortex_api /
├ 000-Getting_started/

...

└ 500-Vision/
```

Then building should work.

sbrisebois, 2019-03-29:
Hi miscz,

I just want to confirm with you that you didn't miss a step in the build procedure... It seems that you didn't create the build folder before running the cmake .. command.

We assume this in the cmake file. This is a standard way to work with cmake, because it avoid to mess the source folder.

Please confirm me that this was the issue, because we run this many time without the need to move the folder to build.

Sincerely,
Sylvain

miscz, 2019-04-01:
Hi Sylvain,

thank you for your answer. I created the build folder. These are the commands I am using:

```
mkdir build  
cd build  
cmake .. -DCMAKE_BUILD_TYPE=release
make  
```

If I do not move all the example folders to the top "examples" folder like mentioned in my previous post I get this error:

` /home/lenovo/Documents/examples/kortex_api/api_cpp/examples/102-BaseGen3_ctrl/02-sequence.cpp:14:27: fatal error: BaseClientRpc.h: No such file or directory`

In the CMakeLists is said `set(KORTEX_DIR "${PROJECT_SOURCE_DIR}/../kortex_api/${KORTEX_SUB_DIR}")`, which means that all the kortex files like `BaseClientRpc.h` can be found in the parent folder. Unfortunately there are only the other example folders and no kortex-api folder.

My guess is, to either copy all example folders to the top "examples" folder (see previous post) or to change the KORTEX_DIR variable inside CMakeLists to:

 `set(KORTEX_DIR "${PROJECT_SOURCE_DIR}/../../../../kortex_api/${KORTEX_SUB_DIR}")`

sbrisebois, 2019-04-01:
Hi Miscz,

Ok, so I think that the issue is more with the kortex_api folder it'self...
The installation instruction are not enough clear, we work on that :)

You need to create a folder kortex_api in the api_cpp/examples folder and copy in that folder the matching language you are using for compilation from the kortex_api zip file downloaded from the web site.

So as an example for linux compilling, you need to copy the content of the folder : 
kortex_api-1.1.3.zip\kortex_api\cpp\linux_x86\ to the kortex_api folder just created.

Let me know if that help you.
Sincerely,
Sylvain

alexvannobel, 2019-04-23:
Hi @miscz ,

Did you manage to build the examples?
I am closing the issue for inactivity, but feel free to reopen it if any build problem remains!

Cheers,
Alex

closed 2019-04-23 by alexvannobel