Kinovarobotics/Kinova-kortex2_Gen3_G3L #6: Control example without sleep function

miscz, 2019-03-29:
I successfully built and ran the Control examples `01-move_angular_and_cartesian.cpp` and `02-sequence.cpp`. Now I wanted to control the robot with a planned path where I am having all points already. I tried to adapt the code in order to drive trough the calculated points. I noticed there is always a sleep function (e.g. `std::this_thread::sleep_for(std::chrono::milliseconds(10000))` ) that waits until each movement between two points is finished. 

Since it is not always possible to calculate exactly the duration of the movement, I wanted to ask if there is a solution, that does not require a timing information. Something like:

`cartesiantMovement(desired_pose, velocity, acceleration,blend_radius)`
`angularMovement(desired_angles, velocity, acceleration)`

Thanks!

sbrisebois, 2019-04-05:
Hi Miscz,

I'm currently working on a demo that will allow you to do this, but it require some gymnastic, because the correct way to do will be ready in a near release. From now, what I can propose, is to register to the event that is generated when a motion is requested, and then launch the next one. There is no need for timers and delays.

I will provide you assap with this demo code in cpp.

Thank you for your patience,
Sylvain

sbrisebois, 2019-04-16:
Hi Miscz,

Here is an example that show how to send action without a sleep, but waiting for the proper event et the end of the execution.
[001-BaseGen3_action_and_notification.zip](https://github.com/Kinovarobotics/kortex/files/3082482/001-BaseGen3_action_and_notification.zip)

Let me know if that help you.

Sincerely,
Sylvain

miscz, 2019-04-16:
Perfect! Thank you! I will try it right away.

alexvannobel, 2019-04-23:
Hi @miscz ,
Did checking the notifications work for your application?

Cheers,
Alex

alexvannobel, 2019-04-23:
Hi @sbrisebois,
Can you bring this example in the repo with a PR? We don't really have examples that demonstrate notification polling and I believe it would be a great addition to the repo.

Thank you,
Alex

Mi-Scz, 2019-04-25:
Unfortunately it throws an error message:

```
Creating session for communication
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout detected: SessionClient::CreateSession

Aborted (core dumped)
```

I have the newest kortex api installed.

Any suggestions?

alexvannobel, 2019-04-25:
Hi @Mi-Scz ,

We usually see this message when trying to connect to a robot, and the connection fails (either the IP address is wrong, the robot is turned off or it can't be pinged).
I have a few questions/suggestions : 
- You say you have the latest kortex_api installed, I will assume it is v1.1.6 found [here](https://drive.google.com/file/d/1ASbEsulf5cByru8Hy1oBZJyNDBa9H22C/view).
- Do you also have the latest SWU for the arm, that matches the API version? That would be SWU v1.1.6 (Jaco3-1.1.6-17.swu) found [here](https://drive.google.com/file/d/1zed2gbzFgtPKjeeDn3LY8A0SIbGVDk2z/view). If not, you can update your arm's firmware _via_ the Webapp. The error you have can be thrown when there is a mismatch between API and firmware version.
- After you get this message, can you ping the arm successfully? 
  If you can, can you try rebooting the arm? It will restart the API server on the arm and restore the communication. Does the error message persist?
  If you can't, can you verify the arm is turned on and the IP address used is the right one?

Hope this helps,
Cheers,
Alex

Mi-Scz, 2019-04-26:
Hi Alex,

Yes I have the kortex api v1.1.6. I installed. I tried to update the SWU (downloaded from your link) but I am not sure if it was successful. Not all joint seem to be updated:

[image]

But good news, the demo works now. Your hint with the correct IP address helped to find the error. The IP address defined in the demo file (#define IP_ADDRESS "192.168.1.110") did not match the default IP address of the robot that I am using (192.168.1.10).

One more question: Will there be a function, where one can use a blend radius in cartesian movement?

Thank you for your help!
Michael

alexvannobel, 2019-04-26:
Hi @Mi-Scz ,

I confirm the update worked, as the SWU version 1.1.6 contains firmware version 1.1.6 for the base and vision module, and 1.1.5 for the actuators. 

As for the blend radius in cartesian movement, there will eventually be a function, yes, but I cannot confirm it will be in the next release.

Thanks a lot for your feedback,
Cheers,
Alex

closed 2019-04-26 by alexvannobel