Kinovarobotics/Kinova-kortex2_Gen3_G3L #9: Compiling example on windows

michaelchi08, 2019-04-11:
Hello,

I'm trying to compile the cpp examples on my windows 10 PC. I tried the steps to compile but i'm not sure if i got the right files in the right folder?  i downloaded the zip file and git cloned the examples repository.  Could you please post some steps of how to do this?
C:\dev\kinova\kortex_api-1.1.6\cpp\win_msvc_x86-64\api_cpp>cmake .. -G "Visual Studio 15 2017 [arch]" -DCMAKE_SH=CMAKE_SH-NOTFOUND -DCMAKE_BUILD_TYPE=release [-DKORTEX_SUB_DIR=<os_arch dir name>]
Thank you!

nsaif-kinova, 2019-04-12:
Hello Michael,
The steps are as follows :

1-Install cmake >= 3.10 : https://github.com/Kitware/CMake/releases/download/v3.14.1/cmake-3.14.1-win64-x64.msi
Very important to add cmake to the system path

2-Install GCC 5.4 under Windows https://sourceforge.net/projects/mingw-w64/files/latest/download
On the options page ensure that the options look exactly like this (all options are the default values, except for version, which is the version of GCC we want to install - 5.4 in this case) https://github.com/Kinovarobotics/kortex/blob/master/linked_md/MinGW_5_4_install_setting.png

3- Add the binary to the system environnment variable : 
- Right-click on the Start Button
- Select “System” from the context menu.
- Click “Advanced system settings”
- Go to the “Advanced” tab
- Click “Environment Variables…”
- Click variable called “Path” and click “Edit…”
- Click “New”
- Enter the path to the folder containing the binary you want on your PATH (C:\Program Files (x86)\mingw-w64\i686-5.4.0-posix-dwarf-rt_v5-rev0\mingw32\bin)

4-Create a build folder (kortex\api_cpp\examples\build)

5-Create a folder called kortex_api (kortex\api_cpp\examples\kortex_api) and copy in this folder the include and lib folders from (kortex_api-1.1.3.zip\kortex_api\cpp\win_mingw\) which can be downloaded from here : https://drive.google.com/file/d/1ASbEsulf5cByru8Hy1oBZJyNDBa9H22C/view

6-Open a cmd in the build folder and type these two commands :
```
cmake .. -G "MinGW Makefiles"
mingw32-make 
```

michaelchi08, 2019-04-12:
Hello @nsaif-kinova ,

The steps are the same for MSVC? Is there a preference to use gcc instead?

Thanks

sbrisebois, 2019-04-12:
Hello Michael,

The instruction are very similar, the main difference is with the cmake command (see the c++ api doc).
But there is no preference... it up to you. What do you prefer, using GCC with other code that already use GCC or work with visual studio.

Best,
Sylvain

michaelchi08, 2019-04-12:
I get the following error 

C:\dev\kinova\kortex\api_cpp\examples\build>cmake .. -G "Visual Studio 15 2017 [Win64]"
CMake Error: Could not create named generator Visual Studio 15 2017 [Win64]

Generators
* Visual Studio 16 2019        = Generates Visual Studio 2019 project files.
                                 Use -A option to specify architecture.
  Visual Studio 15 2017 [arch] = Generates Visual Studio 2017 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 14 2015 [arch] = Generates Visual Studio 2015 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 12 2013 [arch] = Generates Visual Studio 2013 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 11 2012 [arch] = Generates Visual Studio 2012 project files.
                                 Optional [arch] can be "Win64" or "ARM".
  Visual Studio 10 2010 [arch] = Generates Visual Studio 2010 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Visual Studio 9 2008 [arch]  = Generates Visual Studio 2008 project files.
                                 Optional [arch] can be "Win64" or "IA64".
  Borland Makefiles            = Generates Borland makefiles.
  NMake Makefiles              = Generates NMake makefiles.
  NMake Makefiles JOM          = Generates JOM makefiles.
  Green Hills MULTI            = Generates Green Hills MULTI files
                                 (experimental, work-in-progress).
  MSYS Makefiles               = Generates MSYS makefiles.
  MinGW Makefiles              = Generates a make file for use with
                                 mingw32-make.
  Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Watcom WMake                 = Generates Watcom WMake makefiles.
  CodeBlocks - MinGW Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles = Generates CodeBlocks project files.
  CodeBlocks - NMake Makefiles JOM
                               = Generates CodeBlocks project files.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - MinGW Makefiles   = Generates CodeLite project files.
  CodeLite - NMake Makefiles   = Generates CodeLite project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Sublime Text 2 - MinGW Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - NMake Makefiles
                               = Generates Sublime Text 2 project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
  Kate - MinGW Makefiles       = Generates Kate project files.
  Kate - NMake Makefiles       = Generates Kate project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Eclipse CDT4 - NMake Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - MinGW Makefiles
                               = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.

sbrisebois, 2019-04-12:
Hi,

You need to remove the bracket : 
Visual Studio 15 2017 [Win64] must be like : Visual Studio 15 2017 Win64

michaelchi08, 2019-04-12:
Hello Sylvain,

It works. Thanks for your help and fast support.

closed 2019-04-12 by michaelchi08