Kinovarobotics/Kinova-kortex2_Gen3_G3L #12: Implemented / Not-Implemented-Yet flags

akihikoy, 2019-04-18:
## Summary

Could you categorize the examples into Implemented and Not-Implemented-Yet classes?

## Details

I explored the [examples](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/readme.md), however according to the issues (#3 and #11 ), some features are not implemented yet.  As far as I saw, there are no flags like Not-Implemented-Yet or Incomplete.  We need to execute the code to know if it works.  When an example does not work, we may think a possibility that our operation is wrong.  It is time consuming.  Just putting flags for each example is much helpful.

Many thanks,

Akihiko

alexvannobel, 2019-04-23:
Hi @akihikoy ,

Sorry for the late reply.
Technically, all the examples are functional and we do not use non-implemented functions in the code. We will add and modify the examples as we release new firmware and API versions to reflect the new mechanics, functionalities and RPC calls available.
However, we may have missed a call in one the examples. Can you tell me what example(s) were not working as expected? 

Cheers,
Alex

sjhansen3, 2019-04-30:
+1 to this. Not all RPC's are implemented. It would help us all to be very verbose about which isnt working and which is. For example I know not all gripper interfaces (Torque and velocity) are not implemented. Putting the examples aside - it would be nice to know which RPC's are expected to work and which are not. Nothing is more frustrating than spending a few hours debugging to realize that it was never going to work in the first place. Perhaps you should even throw a not_implemented_error or something incredibly verbose. It would also be super helpful for us developers to know what the timeline is for these things. Is it coming in months/weeks/years? This way we can plan our development roadmaps accordingly. Tagging @nsaif-kinova on this one as well for his thoughts.

alexvannobel, 2019-05-03:
Hi,
We understand the presence of not-yet-implemented RPC's in the proto and generated files create a lot of confusion, and we are planning for the next release to 
a) Increase verbosity for exceptions thrown by the API, especially the exception for "not implemented function".
b) Remove a lot of the not-implemented functions from the proto and generated files to reduce confusion. We'll put them back when they'll be implemented inside the arm. 

Cheers,
Alex

akihikoy, 2019-05-09:
Hi @alexvannobel 

The examples of not-implemented features are discussed in #3 and #11 
that are about the velocity control of the robot.

Thanks,
Akihiko