Kinovarobotics/Kinova-kortex2_Gen3_G3L #13: Increase velocity communication limit through high level api to atleast 100Hz

sjhansen3, 2019-04-30:
## Description

Set joint speeds is limited to 40Hz through the high level API. It would be preferable to be able to send velocity commands at at-least 100Hz through the high level API. 

## Version
KortexAPI : 1.1.7
Kortex Device : Gen3

alexvannobel, 2019-05-03:
Hi Stephen,
Thanks for the feedback.
It is planned in the roadmap to support controlling the robot at higher rates while maintaining the high-level control protections. 
However, it will not be available in the upcoming release but probably in the next.

Cheers,
Alex

polthergeist, 2020-02-28:
@sjhansen3 If I may add,

Meanwhile, you could use the low level control that let you send joint position at 1 Khz. Maybe that can suits your needs?

grizzi, 2021-01-20:
What is the current status about this?

felixmaisonneuve, 2021-06-14:
The high level API is still running at 40Hz and this cannot be changed as of now. 

This is something that will be adressed eventually, so I will keep this issue open

rojikada, 2023-02-22:
+1 Also looking for this enhancement, I would like to avoid doing the IK, singularities and protection zones by myself - while sending velocity commands to the robot at higher rate.

martinleroux, 2024-07-18:
Hi all,

I am currently in the process of going through all of our issues for a much needed clean up - with the intention of increasing our responsiveness. In this process, I will keep open issues related to the content of the repo itself (i.e. the examples, the documentation) as well as issue regarding the API.

I will however close this issue since it consist of a request for improvement on the product. Rest assured that the request has been formally logged into our customer feedback system. For further requests for improvements to the robot itself, please contact us at support@kinova.ca.

Cheers,

closed 2024-07-18 by martinleroux