Kinovarobotics/Kinova-kortex2_Gen3_G3L #14: Increase velocity limit through high level API to 100Deg/s.

sjhansen3, 2019-04-30:
## Summary

Allow for 100 deg/s commanded velocities through the high level API. Currently the commanded values are limited to 50 deg/s. This limits control performance.

## Use case

Increased control performance while maintaining safeties implemented by the high level API.

What are the use case covered by this feature?

Positional control using velocity feedback.

## Alternative
Send the commands at 40Hz.

## Possible issues

Crappy control performance at 40Hz.

alexvannobel, 2019-05-03:
Hi Stephen,

In the upcoming months, we will indeed be looking at ways (both mechanical and software) to allow for higher commanded velocities through the high-level API. 

Thanks a lot for the feedback.
Cheers,
Alex

alexvannobel, 2020-02-28:
Hi @sjhansen3 ,

This is to let you know that the soft speed and acceleration limits are now configurable as of release 2.2 for Gen3, and the hard limits have been increased to 80 deg/s for the big actuators and to 70 deg/s for the small actuators. The Cartesian twist hard limits are now 50 cm/s (linear) and 100 deg/s (angular).

It is not currently in the roadmap to make the Gen3 arm faster (so 100 deg/s is not planned for now), but I'll let the issue open anyway to let you know in the future if there are improvements in the Gen3 joint or Cartesian speed.

Best regards,
Alex

sjhansen3, 2020-03-02:
Awesome!

martinleroux, 2024-07-17:
Tests were recently run to validate whether it would be possible to increase the twist hard limits and it was determined that limitations of the hardware will prevent us from making it happen. I will therefore close this issue.

closed 2024-07-17 by martinleroux