Kinovarobotics/Kinova-kortex2_Gen3_G3L #17: Software release schedule

akihikoy, 2019-06-11:
As the user of Gen3, I strongly request the developers to provide us the **software release schedule**.

At least, I need the following FUNDAMENTAL functions of Gen3 control.
- Trajectory control (controlling by specifying multiple via-points).
- Velocity control.

Through the discussions in the other issues, it turned out that these functions are not implemented yet, and the developers always answered: "they will be available in the next release".  However, after waiting for a month, the new release hasn't been provided.  Please **describe the software release schedule**.  Otherwise, we cannot make our  R&D plan.

Also, let me point out the issue of your development plan.  The first things you need to provide are the fundamental functions to control the robot, such as above features.  If we have the above features, we can implement a trajectory avoidance motion by ourselves since the URDF is already provided.  At the early stage, we do not need this kind of fancy features.  All what we need are the fundamental features.  Even for non-professional users, such advanced features can be implemented by using ROS if the fundamental functions are provided.  However if the fundamental features are missing, we can do nothing.  Please change your development plan to **provide the fundamental features as soon as possible**.

alexvannobel, 2019-06-12:
Hi @akihikoy ,

We began the Verification and Validation phase for the 2.0 release, which will include, amongst other features, the trajectory control and the angular velocity control. 
The tests are currently ongoing and we aim at releasing around mid-July.

Cheers,
Alex

akihikoy, 2019-06-13:
Hi @alexvannobel ,
That's good to know.  Thank you so much for sharing the release schedule.
If possible, I would be happy if I can try a beta version of it.
--Akihiko

alexvannobel, 2019-06-13:
Hi @akihikoy ,

We currently do not plan on releasing a beta version before the 2.0 release, but if the plans change we will keep you posted.  

Cheers,
Alex

alexvannobel, 2019-08-13:
Hi @akihikoy ,

The [version release 2.0.0](https://github.com/Kinovarobotics/kortex/releases/tag/v2.0.0) is now released. 

I'll close this issue as it was mainly for asking when the 2.0.0 release would be made, but don't hesitate to create a new one if there is anything. 

Thanks for your patience,
Best,
Alex

closed 2019-08-13 by alexvannobel