Kinovarobotics/Kinova-kortex2_Gen3_G3L #18: Set TCP Values

miscz, 2019-06-13:
Where can I find the command to set other tool center point values (translation AND rotation) other than the default values for the Robotiq Gripper? In order to attach a custom tool we need to change those values.

Thanks a lot!

alexvannobel, 2019-06-13:
Hi @miscz ,

Unfortunately the TCP of the Robotiq Gripper is hardcoded in the firmware's configuration files as of release 1.1.7, but the API function to change the TCP will be available in release 2.0, coming in mid-July.

Sorry for the inconvenience and I will make sure to update this issue whenever the TCP will be exposed in the API.

Best,
Alex

miscz, 2019-07-29:
Hello!

It's almost August and I was wondering if there are any updates or a 2.0 release?

Best regards,
Michael

alexvannobel, 2019-07-30:
Hi @miscz ,

The V&V on the arm's 2.0 firmware is finishing up this week for a targeted release early next week, if everything goes well.

Chers,
Alex

miscz, 2019-07-30:
Nice! Thank you for your answer. I am excited to test the new functionalities!

luangnp98, 2019-08-09:
Hello, there is an estimate to 2.0 firmware release?
Chers!

alexvannobel, 2019-08-13:
Hi @miscz  and @LvOg118 ,

The [version 2.0 is now released](https://github.com/Kinovarobotics/kortex/releases/tag/v2.0.0).

You can now set the TCP from the WebApp or from the API with the [SetToolConfiguration](SetToolConfiguration) function of the ControlConfig service. 

I'll close this issue for now, but feel free to open a new one if there is anything with the functionality.

Thanks for your patience,
Best,
Alex

closed 2019-08-13 by alexvannobel