Kinovarobotics/Kinova-kortex2_Gen3_G3L #19: Is there a way to directly control the actuators?

henrycharlesworth, 2019-07-09:
## Summary

As far as I can tell the only actions that can be applied are to send the arm to fixed positions or to control individual joint speeds (it seems from the other issues that currently being able to control all joint speeds simultaneously is not available yet). Is there a way to directly control the actuator torques, or if not will this be possible in the future?

## Use case

We are planning to use the arm for reinforcement learning experiments. Most likely this will involve transferring policies trained on a Mujoco simulator to the real arm, and it is most natural to learn in the simulator directly acting on all of the actuators.

alexvannobel, 2019-07-12:
Hi @henrycharlesworth ,

The Torque Control mode (torque commands on every actuator) will be available as an experimental feature in the next release (coming in the next few weeks). 
I'll tag this issue as "not implemented yet" to let you know when it's released.

Best,
Alex

alexvannobel, 2019-08-13:
Hi @henrycharlesworth ,

The [version 2.0 is now released](https://github.com/Kinovarobotics/kortex/releases/tag/v2.0.0).

You can see an example of controlling the actuators torque with low-level commands in this [C++ example](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-BaseGen3_torque_control/01-torque_control_cyclic.cpp) and in this [Python example](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/108-BaseGen3_torque_control/01-torque_control_cyclic.py). 

I'll close this issue for now, but feel free to open a new one if there is anything with the functionality.

Best,
Alex

closed 2019-08-13 by alexvannobel