Kinovarobotics/Kinova-kortex2_Gen3_G3L #20: Controlling a Robotiq gripper

RaduCorcodel, 2019-08-09:
I'm trying to get the Robotic gripper to open and close in position mode. Here is the code I am using:
```
#!/usr/bin/env python

import sys
import rospy
from kortex_driver.srv import *
from kortex_driver.msg import *

def Gripper_client():

    rospy.wait_for_service('SendGripperCommand')
    rospy.wait_for_service('GetMeasuredGripperMovement')

    try:
        function_Gripper = rospy.ServiceProxy('SendGripperCommand', SendGripperCommand)
        function_GetMeasuredGripper = rospy.ServiceProxy('GetMeasuredGripperMovement',GetMeasuredGripperMovement)

        current_gripper = function_GetMeasuredGripper()
        request = SendGripperCommandRequest()

        request.input.gripper.finger = current_gripper.output.finger
        request.input.mode = 3
        request.input.gripper.finger[0].value = 0.1

        function_Gripper(request)

    except rospy.ServiceException as e:
        print "Service call failed: %s"%e

if __name__ == "__main__":
    Gripper_client() `
```
Though I set the gripper mode to 3 (per [documentation](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/references/enm_Base_GripperMode.md#))  the gripper either fully closes if I give it a negative value, and fully opens when I give it a positive value. This also contradicts the documentation where it says that in position mode, admissible values for each finger is between 0 and 1.0, where 0 is closed and 1.0 is fully open. 

Any ideas? Maybe the code I'm using has a bug?
Thanks.

alexvannobel, 2019-08-13:
Hi @RaduCorcodel ,

Since the [version 2.0 has just been released](https://github.com/Kinovarobotics/kortex/releases/tag/v2.0.0), can you try either the [low-level gripper control example](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/107-BaseGen3_gripper_low_level/01-gripper_low_level_command.py) or the [high-level gripper control example](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/106-BaseGen3_gripper_command/01-gripper_command.py)? The latter may be more useful to you since you use the SendGripperCommand function in your own code. Note that this code is not ROS based and uses the Kortex Python API (ROS uses the C++ API in the driver).

Since this is ROS code, I also invite you to look at the [ros_kortex 2.0 release](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.0.0) and the [full_arm_movement example](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/python/full_arm/example_full_arm_movement.py#L151-L168), which controls the gripper in position].

Best,
Alex

RaduCorcodel, 2019-08-13:
Thank you. I pulled the new release and will try that. So far I have upgraded the firmware on the robot but the vision module fails to update. I know it's an unrelated topic so I won't go into details here. I submitted a ticket and hopefully will get sorted out soon.

alexvannobel, 2019-08-15:
Hi @RaduCorcodel ,

I invite you to take a look at #22 for the vision module problem.

Best,
Alex

alexvannobel, 2019-10-18:
Hi @RaduCorcodel 
Was that problem solved for you?

RaduCorcodel, 2019-10-24:
Hi @alexvannobel, so sorry for the late reply. Yes, it is all sorted out since the 2.0 release. Thank you for the help.

Closing

closed 2019-10-24 by RaduCorcodel