Kinovarobotics/Kinova-kortex2_Gen3_G3L #22: Vision module firmware does not upgrade to version 2.0.0

eduardgr, 2019-08-14:
## Description

When upgrading firmware, all modules have been correctly updated, but vision module fails and continues at version 1.1.7-31

## Version

-- BASE: id = 0 --
     firmwareVersion: 1006764032
-----------------------------
-- VISION: id = 9 --
     firmwareVersion: 520159495
-----------------------------
-- INTERCONNECT: id = 8 --
     firmwareVersion: 1006764032
-----------------------------
-- SMALL_ACTUATOR: id = 7 --
     firmwareVersion: 1006764032
-----------------------------
-- SMALL_ACTUATOR: id = 6 --
     firmwareVersion: 1006764032
-----------------------------
-- SMALL_ACTUATOR: id = 5 --
     firmwareVersion: 1006764032
-----------------------------
-- BIG_ACTUATOR: id = 1 --
     firmwareVersion: 1006764032
-----------------------------
-- BIG_ACTUATOR: id = 2 --
     firmwareVersion: 1006764032
-----------------------------
-- BIG_ACTUATOR: id = 3 --
     firmwareVersion: 1006764032
-----------------------------
-- BIG_ACTUATOR: id = 4 --
     firmwareVersion: 1006764032

KortexAPI : 2.0.0

alexvannobel, 2019-08-14:
Hi @eduardgr ,

These are the steps you can follow : 
- Upload [this special SWU load](https://artifactory.kinovaapps.com/artifactory/generic-local-public/kortex/gen3/custom-vision-update/2.0.0/Gen3-KOR-1808-Fix-Vision-Update-2.0.0-1.swu) to the arm
- Refresh the WebApp with CTRL+F5
- Upload the 2.0.0 SWU again
The special load basically only fixes the vision module in the arm (which won't have this problem again in future releases). That's why it's so lightweight compared to a real SWU.

Let me know how this goes,
Alex

eduardgr, 2019-08-14:
Hi @alexvannobel 

Thanks for your response!!

The link you sent is not opening. Just like all GitHub download links after yesterday's update.
I made the downloads through the site: https://www.kinovarobotics.com/en/knowledge-hub/gen3-ultra-lightweight-robot

Maybe it is some Artifactory error. Is it possible to make the file available in Drive?

Thank you!

alexvannobel, 2019-08-14:
Hi @eduardgr ,

This is weird, the links are all working on my side and the artifacts are on the public side of our Artifactory server, so there is no need for a login to access them. 
I gave you the direct download link, but can you try [the Artifactory link](https://artifactory.kinovaapps.com/artifactory/webapp/#/artifacts/browse/tree/General/generic-local-public/kortex/gen3/custom-vision-update/2.0.0/Gen3-KOR-1808-Fix-Vision-Update-2.0.0-1.swu)? This should bring you in Artifactory, where you can hit the Download button on the top right side of this artifact.
Meanwhile, I'll see about the Artifactory settings and why the links don't work for you.

Let me know how this work,
Alex

eduardgr, 2019-08-14:
Hi,

I still can't have access.
Error: ERR_EMPTY_RESPONSE

alexvannobel, 2019-08-15:
Hi @eduardgr ,

I checked with the system admin and he modified a couple options so now every region should have access to our Artifactory server. We made some tests on a VM before the change, then got the same response as you. 
After the changes, we were able to access the Artifactory links normally and download the files from the VM.

Can you check the links once more?

eduardgr, 2019-08-15:
Hi @alexvannobel 

Now I was able to download it. 
I will perform the steps to update the vision firmware and then I'll tell you if it worked.

Thank you so much for your attention!

eduardgr, 2019-08-15:
Hi @alexvannobel 

Now everything works. The vision firmware has been updated!

Thank you!

closed 2019-08-15 by alexvannobel

RaduCorcodel, 2019-08-15:
Thank you. Fix works on our robot too.