Kinovarobotics/Kinova-kortex2_Gen3_G3L #23: How to Set ProtectionZone using API-C++, Is there some example codes?

gumpfly, 2019-08-22:
## Summary

A short description of the requested feature.

## Use case

Why is this feature needed?

What are the use case covered by this feature?

## Alternative

Do you have an idea how on how best to implement this enhancement?

If this feature is not implemented, do you have a workaround?

## Possible issues

Anything that could be affected by the new feature.

## Additional context

Any other information you would like us to know about the requested enhancement.

alexvannobel, 2019-08-23:
Hi @gumpfly ,
It's true that a proper Protection Zones example is missing (both for C++ and Python), so we'll work on adding that in the repo in the coming days.

For now, you can look into the [Base service documentation](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md) for the `CreateProtectionZone` function (to add a protection zone), the `UpdateProtectionZone` function (to modify an existing zone) and the `ReadProtectionZone` and `ReadAllProtectionZones` functions to retrieve the zones. The `DeleteProtectionZone` function may also be useful.

You can learn about the ProtectionZone data structure [in the corresponding documentation](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/Base/ProtectionZone.md#).

I'll update this issue when the example will be pull-requested and when it will be added to the repo.

Best,
Alex

alexvannobel, 2019-09-13:
Hi @gumpfly ,

We just added a [C++](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/100-BaseGen3_overview/02-protection_zones_configuration.cpp) and a [Python](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/100-BaseGen3_overview/02-protection_zones_configuration.py) example to show users how to use protection zones with the API.
I'll close the issue, but feel free to reopen it or create a new one if you encounter a problem!

Best,
Alex

closed 2019-09-13 by alexvannobel