Kinovarobotics/Kinova-kortex2_Gen3_G3L #26: A number of questions about using torque control

henrycharlesworth, 2019-10-02:
I've spent the last couple of days trying to get my head around how the Python API works, and in particular looking at the torque control example. Our group is hoping to use the arm for research in reinforcement learning, and so we need a sensible way for the robot to be able to try random actions so it can learn. The most natural way to do this is by being able to directly control the torques that are applied to the actuators. I have a number of questions about the python example that uses torque control as there are parts that I don't understand, and also some related questions about what we're going to need to do to get the robot to work in the way we need it to. I'm not completely sure if this is the best place to post this, so let me know if there is somewhere else more appropriate.

So firstly, I should start by saying that when I run the example ([this one](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/108-BaseGen3_torque_control/01-torque_control_cyclic.py)) it actually does nothing. I don't get any errors but by eye the arm doesn't appear to move at all - so it seems like a bit of a strange example. 

In the code, the first bit I don't understand is in RunCyclic:
```
# Position command to first actuator is set to measured one to avoid following error to trigger
# Bonus: When doing this instead of disabling the following error, if communication is lost and first
# actuator continue to move under torque command, resulting position error with command will
# trigger a following error and switch back the actuator in position command to hold its position
self.base_command.actuators[0].position = self.base_feedback.actuators[0].position
```

I don't understand the comment here and don't get the motivation for this  at all - if all you're doing is applying torques how can you set the actuator position?

We then set the torque relative to the first value `init_first_torque = -self.base_feedback.actuators[0].torque`:

`self.base_command.actuators[0].torque_joint = init_first_torque + \
                                                              self.torque_amplification * (self.base_feedback.actuators[
                                                                                               6].torque - init_last_torque)`
I guess it's not clear to me really what we're trying to do here. Given that we're not applying  a torque to actuator 6, `self.base_feedback.actuators[6].torque` should be the same as `init_last_torque`, so this will always just equal init_first_torque - which would explain why it doesn't do anything.

I modified the example by replacing this with:
`curr_torque = self.base_feedback.actuators[0].torque
self.base_command.actuators[0].torque_joint = -curr_torque - 0.5`

(I also removed the next line `self.base_command.actuators[6].position = self.base_feedback.actuators[0].position - init_delta_position` because I have no idea why this is there/what it's doing.) When I run this it rotates around the first joint as you would expect. I have a couple of questions here as well though - is it sensible for me to be adding a torque relative to -1 times the last measured torque? It seems to work, but I'm not really sure. I guess I don't understand exactly what the command is doing - is it right to say that each joint measures the torque it's experiencing (which is in self.base_feedback.actuators[i].torque), and then the torque we send will be applied on top of this (so -curr_torque should cancel out the current torque being experienced). The reason I'm not sure is that I modified the example to try and send torque commands to all actuators at once and it didn't seem to work as I expected. Firstly, what is the simplest way to do this? I can see there's a [joint torques](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/messages/Base/JointTorques.md) class but I'm not sure how to use this. What I ended up doing was:
```
for i in range(self.expected_number_of_actuators):
    self.base_command.actuators[i].position = self.base_feedback.actuators[i].position
    curr_torque = self.base_feedback.actuators[i].torque
    self.base_command.actuators[i].torque_joint = -curr_torque + actuator_torques[i]
```
where `actuator_torques` is just an array of 7 numbers I define. I then do
```
try:
    for i in range(self.expected_number_of_actuators):
        self.base_feedback = self.base_cyclic.Refresh(self.base_command, i, self.SendOption)
```
If I choose `actuator_torques=[1.0,0.0,0.0,0.0,0.0,0.0,1.0]`, I get rotation around the first and last actuator as I expect, but the arm also starts to fall down gradually - so the resultant torques in the other five actuators definitely aren't zero.

Another issue that I've found is that applying torques in this way (just to a single actuator, not all at once) often leads to the arm going into a fail state (red light) so that it has to be reset. Most of the time this happens at the end of the time interval when we finish applying the torque (but it is quite inconsistent and doesn't always happen), but it sometimes happens during the interval when the torque is being applied too. What is the best way to stop this happening?

I've also noticed that protection zones don't seem to work with the torque commands. Is this because they're operating at a low-level through the cyclic client? Is it possible to issue torque commands using the normal base client? For our purposes I don't think we need to be sending updated torque commands at super-high frequency (so long as the applied torques are applied constantly until the next command arrives), and so I think the speed of the normal base client would probably be OK, but it would be very useful if we could still use protection zones.

So I guess I can summarise my main questions as follows:

- Why do we ever need to set the actuator positions when sending torque commands?
- How exactly does the torque-offset work? What is the most natural way to apply "random torques" to the current state?
- What can we do to ensure that the arm doesn't go into a fail state during and after applying a torque command?
- What is the easiest way to send torque commands to every actuator in one go?
- Can we use protection zones with torque commands?
- Can we send torque commands with the non-cyclic base?

hlamontagne, 2019-10-09:
Hi @henrycharlesworth ,
Sorry for the late reply. I can answer some of your question and I'll get back to you about those I can't answer be cause I need to verify a few things with the control team.

**Why do we ever need to set the actuator positions when sending torque commands?**
It is for security purpose. By providing the position feedback in your command, the actuator can validate that the position still "follow" your command. If something bad happens in the application, the actuator will stop when its error(position - command) is greater than 5 degrees.

**What is the easiest way to send torque commands to every actuator in one go?**
By using the function **feedback Refresh(command)**. The input argument is a BaseCyclic_pb2.Command object that contains a list of torque command. One for each actuator. The function refresh will send all those command in one go. You can use the function refresh with this import: 
from kortex_api.autogen.client_stubs.BaseCyclicClientRpc import BaseCyclicClient

**Can we use protection zones with torque commands?**
Nope

**Can we send torque commands with the non-cyclic base?**
Nope

I will get back to you shortly with the remaining answers,

Hugo

hlamontagne, 2019-10-10:
Hi @henrycharlesworth ,

Here are the missing answer from my last answer:

**How exactly does the torque-offset work? What is the most natural way to apply "random torques" to the current state?**
Are you talking about the function SetTorqueOffset() available in the ActuatorConfig API?
This function is used to set a calibration offset.
I don't think this is what you are looking for. In fact, I suggest that you don't use it, it is used to re calibrate the actuator.

For now since the torque control are only available in the BaseCyclic, the high level kinematic functions are not used. The torque commands are directly sent to the actuators without gravity compensation hence the fall you observed when sending 0 torque.

**What can we do to ensure that the arm doesn't go into a fail state during and after applying a torque command?**
To avoid going in fault state, switch back the control mode to POSITION using the function SetControlMode(). This way, you will avoid unwanted movement when you are done with the torque control.

I hope this answer all your questions correctly.
Don't hesitate if it is not clear of if you need anything else.

Hugo

henrycharlesworth, 2019-11-11:
Hi Hugo,

Sorry for the late follow up, I've been away for a while and also working on other things, thank you for the reply and the useful information! I still have a few problems/questions about the best way to do what we need to do.

Firstly, you said:

> What is the easiest way to send torque commands to every actuator in one go?
By using the function feedback Refresh(command). The input argument is a BaseCyclic_pb2.Command object that contains a list of torque command. One for each actuator. The function refresh will send all those command in one go. You can use the function refresh with this import:
from kortex_api.autogen.client_stubs.BaseCyclicClientRpc import BaseCyclicClient

Unless I'm misunderstanding something here the Refresh function requires a device_id, or is there a way to tell it to send to all devices? At the moment I have been looping over actuator devices as I mentioned in my first question, but it seems strange to be able to define a Command object that has torques for all actuators and then have to use a loop to send this command repeatedly to each actuator.

> For now since the torque control are only available in the BaseCyclic, the high level kinematic functions are not used. The torque commands are directly sent to the actuators without gravity compensation hence the fall you observed when sending 0 torque.

OK, what would be the easiest way to compensate for gravity? I think it is extremely preferable to be applying torques relative to the torque required for the arm to be in equilibrium (at the moment I've been applying them relative to the current torque the actuator is experiencing, i.e. -curr_torque + torque_to_apply, but this does not properly compensate for gravity.

> To avoid going in fault state, switch back the control mode to POSITION using the function SetControlMode(). This way, you will avoid unwanted movement when you are done with the torque control.

This is what I've been doing, but it doesn't always work. Sometimes it goes into a fail state even during the torque control being applied. I would say particularly if we are applying quite strong torques, and then stop and switch the control mode to POSITION, the stop is very abrupt and the arm often goes into a fail state.

Another thing I was wondering is can we automatically get the arm to recover from a fail state? e.g. if it fails, can we use the API to send it back to a given "safe" position? Because if we have the arm trying out random actions so that it can learn, it will be extremely frustrating if we have to manually reset it every time it gets itself into a fail state.

Appreciate you taking the time to help! Henry

henrycharlesworth, 2019-11-22:
Hi, just following up on this. Is it possible to get any more information about these questions?

gstlaurentkinova, 2019-11-25:
Hi @henrycharlesworth ,

Sorry for the late reply.
Hugo has left for a paternity leave, so I will be following with you on this issue. 

I have read your concerns and have asked the low-level engineering team for the answers. They should answer me today or tomorrow. 

I'll get back to you as soon as possible.

Gabriel

gstlaurentkinova, 2019-11-26:
Hi @henrycharlesworth ,

Regarding the `Refresh(self, command, deviceId, options)` function, the `deviceId` argument is optional and defaults to zero. 
This is the value you should use, as this will effectively send the command to the base, which will in turn send it to all acutators.
Populating the `command` with as many ActuatorCommands as there are actuators (as you are doing right now), the base will update all actuators in one command.

Regarding compensating for gravity, the team has confirmed that gravity compensation is not done in low-level mode at the moment.  
The only solution is to rebuild a gravity compensation system based off the URDF file of the robot (containting all information about the robot model) and the 
actuator positions.

Regarding the robot going into a fail state during torque commands, this is probably caused by torque commands that jerk the robot above its hard limits.
Try stabilizing the actuator by sending decreasing torque values to allow for a smoother stop. Also, ensure you have sent a position command before switching to
POSITION control mode, as the robot would fail if no position command were sent when switching.

Regarding automatically recovering from a fail state, this is currently not possible in the low-level torque mode you are using. Sending the robot back to a 
'safe' position could be dangerous to the robot in certain fail state scenarios.

I hope this answers your questions. If you have further questions/comments, feel free to ask them here.
If you prefer, you could start new issues with selected questions, since they are mostly unrelated topics.
We could address those individually.

After confirmation from your part that your questions have been answered. I will close this issue. 
As I have previously said, feel free to open up new issues if you have future questions.

Best regards,

Gabriel

alexvannobel, 2020-01-28:
Hi @henrycharlesworth  ,

I'll close this issue for inactivity, but feel free to re-open it if it persists or to open a new issue.

Best regards,
Alex

closed 2020-01-28 by alexvannobel