Kinovarobotics/Kinova-kortex2_Gen3_G3L #27: Kortex Connection Error

heretonjneer, 2019-10-03:
I've been using linux and the C++_api on kortex_api_2.0.0. The web app runs perfectly well, however when I attempt to run 01-api_creation.cpp in a terminal, I receive the following output.
```
~/kortex/api_cpp/examples/build$ ./000-Getting_started_01-api_creation 
Creating session for communication
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout detected: SessionClient::CreateSession

Aborted (core dumped)
```
Here's the code as well: We changed the IP address to 10.10.10.251 for both the web app and the .cpp file. with port 80 also being used by both through an Ethernet connection.

```

/*
 * KINOVA (R) KORTEX (TM)
 *
 * Copyright (c) 2018 Kinova inc. All rights reserved.
 *
 * This software may be modified and distributed
 * under the terms of the BSD 3-Clause license.
 *
 * Refer to the LICENSE file for details.
 *
 */

#include <SessionManager.h>
#include <DeviceConfigClientRpc.h>
#include <BaseClientRpc.h>

#include <RouterClient.h>
#include <TransportClientTcp.h>

namespace k_api = Kinova::Api;

#define IP_ADDRESS "10.10.10.251"
#define PORT 80

void example_api_creation()
{
    // -----------------------------------------------------------
    // How to create an API with the SessionManager, DeviceConfigClient and BaseClient services
    auto error_callback = [](k_api::KError err){ cout << "_________ callback error _________" << err.toString(); };
    auto transport = new k_api::TransportClientTcp();
    auto router = new k_api::RouterClient(transport, error_callback);
    transport->connect(IP_ADDRESS, PORT);

    // Set session data connection information
    auto create_session_info = k_api::Session::CreateSessionInfo();
    create_session_info.set_username("NERVE-Kinova-Gen3");
    create_session_info.set_password("NERVE1308");
    create_session_info.set_session_inactivity_timeout(6000000);   // (milliseconds)
    create_session_info.set_connection_inactivity_timeout(2000000000); // (milliseconds)

    // Session manager service wrapper
    std::cout << "Creating session for communication" << std::endl;
    auto session_manager = new k_api::SessionManager(router);
    session_manager->CreateSession(create_session_info);
    std::cout << "Session created" << std::endl;

    // Create DeviceConfigClient and BaseClient
    auto device_config = new k_api::DeviceConfig::DeviceConfigClient(router);
    auto base = new k_api::Base::BaseClient(router);

    // -----------------------------------------------------------
    // Now you're ready to use the API
    // ...

    // -----------------------------------------------------------
    // After you're done, here's how to tear down the API

    // Close API session
    session_manager->CloseSession();

    // Deactivate the router and cleanly disconnect from the transport object
    router->SetActivationStatus(false);
    transport->disconnect();

    // Destroy the API
    delete base;
    delete device_config;
    delete session_manager;
    delete router;
    delete transport;
`}`
int main(int argc, char **argv)
{
    example_api_creation();
}
```

  I feel like there might be a simple fix I'm not seeing at the moment, Any help would be greatly appreciated! 

 Thanks, Alex

gstlaurentkinova, 2019-10-09:
Hi @heretonjneer ,

The issue here seems to be that the script is using the WebApp Port (`80`) when trying to reach the Kortex API directly. This port is not suitable for communicating directly with the Kortex API.

The WebApp listens to client actions on port `80`, hence why one would use this port during WebApp connection.

When dealing with the Kortex API directly (without using the WebApp), it is mandatory that one uses these ports for communication :
 - `10000` for high-level (TCP) access (Most of the services)
 - `10001` for low-level (UDP) access (BaseCyclic and other *Cyclic services)

I suggest you replace your port definition
```
#define PORT 80
```
with 
```
#define PORT 10000
```
Tell me if this fixes your problem,

Gab

hlamontagne, 2019-10-18:
@heretonjneer I was wondering, is your problem solved?

heretonjneer, 2019-10-18:
It is! Thanks again for your help

closed 2019-10-18 by heretonjneer