Kinovarobotics/Kinova-kortex2_Gen3_G3L #28: ARM support (raspberry pi)

Achllle, 2019-10-08:
## Summary

I'd like to run kortex & ros_kortex on a Raspberry Pi which is based on ARM processors. I tried installing through ros_kortex' installation instructions but get the following error:

```
[ 99%] Linking CXX shared library /home/base-1/Documents/test_ws/devel/lib/libkortex_driver_generated_files.so
/home/base-1/Documents/test_ws/src/ros_kortex/kortex_driver/../kortex_api/lib/release/libKortexApi.a: error adding symbols: File format not recognized
collect2: error: ld returned 1 exit status
ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:946: recipe for target '/home/base-1/Documents/test_ws/devel/lib/libkortex_driver_generated_files.so' failed
make[2]: *** [/home/base-1/Documents/test_ws/devel/lib/libkortex_driver_generated_files.so] Error 1
CMakeFiles/Makefile2:7275: recipe for target 'ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all' failed
make[1]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

In my limited understanding of this, it seems like this is failing because I'm running this on a RPi 3 which has an ARM chip. Is there a short-term fix for this?

## Use case

Running the controller/interfacer on separate compute to ensure isolation as well as meet lower power consumption constraints.

alexvannobel, 2019-10-08:
Hi @Achllle ,

- For the longer term: 
We don't currently test the ARM version of our API and were planning to test it and release it with the new version of the API around the end of the year.
- For the shorter term:
We could provide you with a link to download the untested version of the Kortex API 2.0 built for arm64 architecture. I want to emphasise it's untested, but it could probably deblock you on your Raspberry Pi until the next release.

Let me know if you want the link, and if you do I'll make sure to contact you to provide it.

Best,
Alex

Achllle, 2019-10-08:
Hi Alex, thanks for your swift response!
It would be great if you could send over that link and I'll let you know if I get it to work.
Cheers,
Achille

alexvannobel, 2019-10-10:
Hi @Achllle , 

I've sent you an email with the link at your Peanut Robotics email address. 

Let me know if you received it or if you prefer another email address,
Best,
Alex

polthergeist, 2020-02-28:
@Achllle In the last release package there is a bunch of armv7 version of the API. Maybe it can help you.

Achllle, 2020-02-28:
Wonderful. We got this running on a Raspberry Pi 4. Closing

closed 2020-02-28 by Achllle