Kinovarobotics/Kinova-kortex2_Gen3_G3L #29: Null Space Admittance activated while navigating to a cartesian Pose

Refctrl, 2019-10-24:
Is there any possibility of commanding the robot to a desired cartesian pose while the Null Space Admittance Mode is enabled?
I tried to enable first the Null Space Admittance and afterwards give a cartesia target pose (both via the API). So the admittance mode is activated in the beginning, but when the robot start to reach the pose, the Admittance mode is immediately disabled.

alexvannobel, 2019-10-28:
Hi @Refctrl ,

The cartesian trajectory mode and Null-space admittance mode are two separate modes. You cannot be in both at the same time. 
I suggest that once the arm arrives at the desired position, you switch immediately to Null Space Admittance mode.

Best,
Alex

alexvannobel, 2019-10-30:
Hi again @Refctrl ,

Is it possible for you to explain your use case? Maybe we could suggest you a workaround to get you the behaviour you need with the current set of features.

Best,
Alex

Refctrl, 2019-10-31:
Hi @alexvannobel,

Thanks for the answer. Yes, I can give you some further information: I try to implement an individually tuned admittance control (giving cartesian velocity and position inputs through the API using the high level mode) to move the robot endeffector manually. While moving the endeffector to a desired position, simultaneously it should be possible to move the elbow (joint 4) of the robot arm to a changed position. The reason for that is, that there might be an obstacle or an area, which the robot elbow is not allowed to pass through. 

Kind regards!

alexvannobel, 2019-10-31:
Hi @Refctrl ,
As I understand it, you want to specify a Cartesian goal (PlayCartesianTrajectory if you use the direct RPC, or REACH_POSE action type if you send an action), then the robot starts moving towards the goal, and during the trajectory you want to be able to change the position of the elbow with admittance (while the robot doesn't change his goal position, and keeps moving towards it).
Unfortunately, as I said, we don't support admittance + trajectory at the same time.

I see two workarounds in this situation:

1. If you really don't want the arm to stop, you'll have to implement your own low-level control algorithm, that calculates joint angle increments every 1ms in order to reach the Cartesian goal, and that also takes into account the torque readings from each joint to apply the admittance to move the elbow. You would have to calculate the IK and send the joint position increments through the Refresh() function in the BaseCyclic service (the low-level). This is a lot of work, but I think it's the only way to implement the control mode as it is needed, as I understand. 
2. If you can accept stops while reaching the Cartesian goal, then you could Pause the trajectory, switch to Nullspace Admittance mode, move the elbow, and then Resume the action. You could map a keyboard interrupt, or another type of interrupt that would pause/resume the trajectory and switch the modes, and simply hit the interrupt whenever you need the admittance. This is the more easy way of implementing your mode, but clearly it pauses the trajectory, which is something you don't want, as I understand.

Let me know what you think,
Best,
Alex

Refctrl, 2019-11-01:
Hi @alexvannobel,

Yes, I guess possibility 1 would be the better option, but the implementation is really a lot of work. 
So I have one more question regarding the IK: Is there any function, which can be used to get the Jacobian of the actual joint configuration of the robot?

Kind regards!

alexvannobel, 2019-11-01:
Hi @Refctrl ,

Unfortunately, the Kortex API doesn't provide that as of right now, and it's not in the short-term pipeline. 
However, you have access to the robot's [URDF file](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_description/arms/gen3/urdf/GEN3_URDF_V12.urdf) in the ROS Kortex repository, if you want to use a framework that needs the URDF as an input. You won't be able to do 1ms real-time low-level control from our ROS driver, though.

Hope that helps,
Best,
Alex

Refctrl, 2019-11-02:
Hi @alexvannobel,

okay, all right, thanks for your advices!

Kind regards!

alexvannobel, 2019-11-04:
Hi @Refctrl ,

You're welcome! As this is not something we plan on providing in the short or mid term, I'll close the issue but still tag it as not implemented.
Don't hesitate to comment, or reopen it.

Best,
Alex

closed 2019-11-04 by alexvannobel