Kinovarobotics/Kinova-kortex2_Gen3_G3L #35: What is the difference between the joint velocity control in low level control and high level control?

XuRuoyuRoy, 2020-01-09:
I see two joint velocity controller examples in both high level and low level. What is the difference? Can I use the velocity controller in high-level control to realize the cartesian pose control and track an object in realtime?

polthergeist, 2020-01-09:
Hi @XuRuoyuRoy ,
When the High level API is used, your command is sent to the robot's base and then the base's controller create a bunch of low level command and send them to the actuators. Benefit of using this technique is that you don't need to bother about the synchronisation or the real time programming stuff.
When the level is used, your command is directly received by a specific actuator. The benefit of using this technique is that your are totally free as long as you don't trigger a safety measure on the actuator(max acceleration, speed, current peak, temperature, ...) But, if you want to use the low level api, you need to take care of all the real time stuff on your end, mainly make sure that you send a command every ms to the actuator. 

I hope this help you and don't hesitate if this was not clear enough :)

Hugo

alexvannobel, 2020-01-28:
Hi @XuRuoyuRoy ,

I'll close this issue for inactivity, but feel free to re-open it if you still have questions or to open new issues.

Best,
Alex

closed 2020-01-28 by alexvannobel