Kinovarobotics/Kinova-kortex2_Gen3_G3L #36: Is feedback from IMU on the base still available in low level control mode?

XuRuoyuRoy, 2020-01-14:
I am using low level control these days. I find that after I apply "basecyclic->RefreshFeedback", the data from IMU and tool attitudes are not updated, however, the feedback from actuators still works?
So, is the feedback from base not available in low level control mode?

polthergeist, 2020-02-28:
Hi @XuRuoyuRoy ,
I tried to reproduce your problem with this code:
`k_api::BaseCyclic::Feedback base_feedback;

    base_feedback = base_cyclic->RefreshFeedback();

    cout << base_feedback.DebugString() << endl;
    cout << "----------------------------------------------" << endl;

    auto servoingMode = k_api::Base::ServoingModeInformation();

    servoingMode.set_servoing_mode(k_api::Base::ServoingMode::LOW_LEVEL_SERVOING);
    base->SetServoingMode(servoingMode);
    usleep(1000000);
    base_feedback = base_cyclic->RefreshFeedback();
    cout << base_feedback.DebugString() << endl;`

The printout was like this:
base {
  active_state_connection_identifier: 27
  active_state: ARMSTATE_SERVOING_READY
  arm_voltage: 23.8840961
  arm_current: 1.43667579
  temperature_cpu: 55.0000229
  temperature_ambient: 37.9370193
  imu_acceleration_x: 0.0568432212
  imu_acceleration_y: -0.0853758827
  imu_acceleration_z: -9.561759
  imu_angular_velocity_x: 0.844627738
  imu_angular_velocity_y: 0.369404465
  imu_angular_velocity_z: -0.39157936
  tool_pose_x: 1.15522816e-05
  tool_pose_y: 0.00135016267
  tool_pose_z: 1.17662501
  tool_pose_theta_x: -4.71990802e-08
  tool_pose_theta_y: -0.00225453
  tool_pose_theta_z: 179.997894
  tool_external_wrench_force_x: -0.502831876
  tool_external_wrench_force_y: -5.83500287e-06
  tool_external_wrench_force_z: 4.86387e-06
  tool_external_wrench_torque_x: 1.6207583e-05
  tool_external_wrench_torque_y: -0.15345116
  tool_external_wrench_torque_z: -0.0296663884
  fault_bank_a: 16
  commanded_tool_pose_x: 1.82945197e-08
  commanded_tool_pose_y: 0.00135016302
  commanded_tool_pose_z: 1.17662501
  commanded_tool_pose_theta_x: 9.95501533e-13
  commanded_tool_pose_theta_y: -1.86494802e-08
  commanded_tool_pose_theta_z: 179.997406
}
actuators {
  command_id: 2147520019
  status_flags: 33557552
  jitter_comm: 2638629207
  position: 359.999512
  torque: -0.0932033584
  current_motor: 0.0889885
  voltage: 22.9966297
  temperature_core: -0.800184608
}
actuators {
  command_id: 2147585555
  status_flags: 33557552
  jitter_comm: 2638593970
  position: 0.000610351562
  torque: -1.09148872
  current_motor: -0.0966796875
  voltage: 22.9966297
  temperature_core: 0.869343817
[... 157 lines cut ...]
  temperature_core: -1.56769133
}
interconnect {
  feedback_id {
    identifier: 2147877893
  }
  status_flags: 1040
  jitter_comm: 2759620667
  imu_acceleration_x: 0.1237446
  imu_acceleration_y: -0.276183605
  imu_acceleration_z: 9.89179707
  imu_angular_velocity_x: 0.90125
  imu_angular_velocity_y: -0.14875
  imu_angular_velocity_z: 1.26
  voltage: 23.0355473
  temperature_core: 47.1428566
  gripper_feedback {
    motor {
      motor_id: 1
      voltage: 23.0355473
    }
  }
}

In both cases IMU data is fine? Just top be sure, do you say that on your robot, once the lowlevel mode is set, you don't have access to the IMU data?

polthergeist, 2020-04-03:
Hi @XuRuoyuRoy ,
Any news on this?

XuRuoyuRoy, 2020-04-15:
> Hi @XuRuoyuRoy ,
> Any news on this?

It seems your data from IMU won't change too. That happened to my robot too. But IMU on base sometimes works after I reboot the robot arm. That is really strange. Now I will print the data everytime when I start my programm to make sure the IMU is still working.

polthergeist, 2021-02-18:
Hi @XuRuoyuRoy I think we can close this one but just to be sure, on the data I sent to you, the value was refreshing correctly:
[image]
Do you say that on your side it still does not?

felixmaisonneuve, 2021-06-14:
Hi @XuRuoyuRoy, do you still have an issue with data not updating?
I will wait for your response for a couple of days. If I do not get any news I will close this issue

closed 2021-07-30 by felixmaisonneuve