Kinovarobotics/Kinova-kortex2_Gen3_G3L #37: Package discovery for kortex library

rmaronson, 2020-01-14:
## Summary

It would be very convenient if the bundled binary/header files came with appropriate configuration files for ease of compiling. For example, a FindKortex.cmake module or a pkgconfig file.

## Use case

The recommended way to compile against the API (as seen in the example section) seems to be to extract the API include/lib directories to a specific subdirectory of that examples folder before compiling. This method requires manually hard-coding the location of the downloaded Kortex binary for each build. Adding automatic discovery means multiple packages can build against the Kortex library, eases porting across computers that may have installed the api differently, etc.

## Alternative

Do you have an idea how on how best to implement this enhancement?
- See [https://cmake.org/cmake/help/v3.12/manual/cmake-developer.7.html#find-modules]

If this feature is not implemented, do you have a workaround?
- It is possible to develop a 3rd-party cmake find file

## Possible issues

N/A

## Additional context

N/A

gstlaurentkinova, 2020-02-27:
hi @rmaronson,

Starting with the 2.2 version release, the Kortex Api will make use of the Conan Package Manager to help build against it. A Kortex Api Conan package will be made available when the 2.2 version releases. 

Using the CMake Conan utilities (https://github.com/conan-io/cmake-conan), it is possible to generate CMake targets from Conan packages, and to fetch said packages automatically.

We will however still support the old build method, providing pre-built libraries for common architectures.

There is no plan for releasing a FindKortex.cmake file at the moment, as Conan will become the standard build method for the Kortex Api.

Best,

Gabriel

alexvannobel, 2020-03-10:
Hi,

I'll close this issue, as Conan acts as a FindPackage and subsequent releases will automatically fetch the correct versions of the API.
Don't hesitate to reopen the issue if you have anymore questions @rmaronson .

Best,
Alex

closed 2020-03-10 by alexvannobel