Kinovarobotics/Kinova-kortex2_Gen3_G3L #38: Torque control, offsets and calibration

tanguy-s, 2020-01-21:
## Description

We are implementing our own admittance torque controller as the one out-of-the box still 
makes the robot too hard to move to be called compliant. 
This is I believe due to the high 1:100 strain in the actuators.

To do such we implement a simple spring-damper torque control function:
```
tau_d = tau_task + coriolis + gravity
```
with
```
tau_task = J.T() * (Ks * error - Kd * (J * dq)
```

In a static configuration all terms but the gravity term are null.

Thus I expect that feeding tau_d = gravity to the robot should lead to no motion.

However, some joints move and not due to underestimated gravity as they rotate without stopping beyond there equilibrium angle. 
Thus, it seems that there is an offset in torque command/reading depending on the joint. 

## Details

In our case, the robot is wall-mounted with the gravity vector being (-9.81, 0, 0). 
The gravity torques are computed using both [orocos-kdl](https://github.com/orocos/orocos_kinematics_dynamics) and [pinocchio](https://github.com/stack-of-tasks/pinocchio) (same results).

The biggest offset is measured in joint 4. Following are some specific joints configuration:

|    |                                         | j1  | j2  | j3 | j4  | j5 | j6 | j7 |
|:----:|:-----------------------------------------:|:-----:|:-----:|:----:|:-----:|:----:|:----:|:----:|
| q0 |               joints zero               | 0   | 0   | 0  | 0   | 0  | 0  | 0  |
| q1 |          joint 4 axis vertical          | 90° | 0   | 0  | 0   | 0  | 0  | 0  |
| q2 | joint 4 axis horizontal link 5 vertical | 0   | 45° | 0  | 45° | 0  | 0  | 0  |

The following table is the difference between the measured torque and the gravity torque. (averaged on 100 runs)
In **q1** and **q2** there is no (or little) gravity effect on **joint 4**. 

|    |     j1    |     j2    |     j3     |    j4        |     j5    |    j6    |     j7    |
|:--:|:---------:|:---------:|:----------:|:------------:|:---------:|:--------:|:---------:|
| q0 | -0.559599 | -0.290294 | -0.0112868 | **-1.54809** | -0.215382 | 0.399982 | 0.0203352 |
| q1 | -0.486184 |  0.722002 |  -0.504018 | **-1.16791** | 0.0149556 | 0.379552 | 0.0466501 |
| q2 | -0.568791 |  -0.60019 |  -0.264473 | **-1.55995** | -0.175545 |  0.43287 | 0.0350407 |

Digging a little in the actuator config API to find any torque related configuration gives:
```
## Actuator 1
 - global_gain: 1.00013
 - global_offset: -0.00102985
 - gains: 1.90554e-05 1.56956e-05 1.92241e-05 1.58185e-05 
 - offsets: 5.83966 15.1692 2.36108 -13.818 
 - TorqueOffset: 0
## Actuator 2
 - global_gain: 1.00021
 - global_offset: 0.000597942
 - gains: 1.82402e-05 1.88354e-05 1.80694e-05 1.91584e-05 
 - offsets: 23.7048 5.78224 -2.59742 -10.4677 
 - TorqueOffset: 0
## Actuator 3
 - global_gain: 1.00006
 - global_offset: 0.00124747
 - gains: 1.84215e-05 1.74807e-05 1.8309e-05 1.79478e-05 
 - offsets: -47.7051 29.2975 -15.6556 -69.0774 
 - TorqueOffset: 0
## Actuator 4
 - global_gain: 0.999326
 - global_offset: -0.00593934
 - gains: 1.83251e-05 1.68976e-05 1.86598e-05 1.60773e-05 
 - offsets: 38.1596 52.3693 19.5519 23.1122 
 - TorqueOffset: 0
## Actuator 5
 - global_gain: 1.00023
 - global_offset: 0.000457619
 - gains: 1.12083e-05 1.25528e-05 1.09894e-05 1.24981e-05 
 - offsets: 25.8188 15.179 -39.9176 0.23256 
 - TorqueOffset: 0
## Actuator 6
 - global_gain: 1.00009
 - global_offset: 0.00244833
 - gains: 1.07365e-05 1.2134e-05 1.05185e-05 1.23181e-05 
 - offsets: -1.93482 27.189 30.364 22.674 
 - TorqueOffset: 0
## Actuator 7
 - global_gain: 1.00009
 - global_offset: -0.0004263
 - gains: 1.20273e-05 1.22568e-05 1.1448e-05 1.31377e-05 
 - offsets: 8.41774 12.5052 3.48515 -0.515961 
 - TorqueOffset: 0
```

Seems like the TorqueOffset that can be modified from the web-app are all set to 0. 
However, the torque calibration information is not set to 0, and interestingly shows the lowest global gain on actuator 4. 

My questions are the following:
1. How can we use the calibration data or set the user-defined torque offset value in order to properly achieve any kind of torque control ? I believe that it's something already done in your low-level controllers.
2. If the torque calibration data is not usefull, can you describe a methodology to reliably estimate the remaining offsets ? 
3. I am also wondering if disabling the following error by setting ```base_command.mutable_actuators(i)->set_position(base_feedback.actuators(i).position());``` has an impact on the robot's stiffness regardless of the torque command.
4. *Bonus:* did someone manage to achieve proper admittance control with low resistance ? I'd be happy to know more about the controller.

Thank you !

ehocquard, 2020-01-24:
Hi Tanguy,

I'm going to develop similar control in coming days, I can let you know what I learn once I get results.

How did you get Jacobian and Coriolis values? From the lib? I can't find anything in the doc....

Many thanks.

ehocquard, 2020-01-28:
Hi @tanguy-s ,

I'm going to explore torque control on Kinova too.
So many thanks for your feedback, it is valuable information.

I was wondering if you could share your torque controller so I can experiment?
I would understand if you want to protect intellectual property and not share, let me know ;)

Cheers

grizzi, 2020-02-05:
I am also wondering how to achieve proper admittance control. 
We have a controller which is doing basically the same as @tanguy-s. Moreover, there is always a mismatch between the commanded torque and the measured one. People from kinova technical support answered that the torque is measured at the output shaft, so torque control should compensate for friction.

As a toy example, setting a torque of even 10Nm on the last joint (it should rotate like hell), makes the joint only move slowly. Can someone explain this?

Cheers

Eglen24, 2020-02-21:
Hi,
Currently the admittance modes of Gen3 have deadbands to compensate for small errors in torque measurements or offsets which prevent the arm from moving on its own in static positions. These deadbands are to compensate for any noise and bending induced torque measures. This affects the arms stiffness in admittance making it ‘harder to move’. We have also tuned the admittance modes to ensure the stability of the arm in all positions and payloads. We aim for acceptable phase and gain margins that we know ensure the safety of the arm. You can perhaps tune your controller in a way that allows the arm to be less ‘hard to move’ while maintaining acceptable stability margins (but perhaps less restrictive than ours).
The web app and api provide the option to set the torque offset. You can go to the configuration page and click the button Set Zero Torque while the arm is in the zero position (Vertically, to have Zero gravity induced torque). This will automatically adjust the offsets in each joint torque sensor to account for any drifts or offsets that may have occurred.

Another option is to use the Kortex API to set the torque offsets (SetTorqueOffset command) for either each actuator individually. This can be done at any position by calling the API function, and sending the value of torque that you expect the arm should have in that static position (using your own gravity model). We recommend doing this in the zero position. however, if you want to optimise the torque offsets around a specific position of the arm, you can with this function.
Disabling the following error is not recommended as it removes a safety function of the arm. You can directly feed the feedback of the actuator position as the command, however note that the following error protection will now only protect the arm in the case of a communication issue, and not any errors from your controller or software. This functionality only provides a safety and should not affect the performance in any way.
do you answer with that?

tanguy-s, 2020-03-05:
Hi @Eglen24,

Thank you for the answer. 
I did try setting the torque offsets through the Web-app which works fine, however, I still have some small offsets compared to my gravity model on specific joints leading to small residual motions. 
Setting the offsets given my gravity model should solve my issue here, however I am unable to set those offsets per actuator through the Kortex API.  

This code should work fine :
```
auto torque_offset_message = k_api::ActuatorConfig::TorqueOffset();
torque_offset_message.set_torque_offset(0.01f);
actuator_config->SetTorqueOffset(torque_offset_message, 7);

auto torque_offset = actuator_config->GetTorqueOffset(7); // ERROR
std::cout << "TorqueOffset: " << torque_offset.torque_offset() << std::endl;
```

however I get on the read operation :
```
terminate called after throwing an instance of 'Kinova::Api::KDetailedException'
  what():  Server protocol error, Error sub type=DEVICE_NOT_READY => <srv: 10, fct: 30, msgType: 3>
description: Non-parsable payload received from device.
```

Which happens specifically on the TorqueOffset, as for example, setting the ControlMode per actuator works fine (see below minimal code to reproduce).

Is anyone able to reproduce ? Do you have an idea of what is wrong ?

Thanks

```
#include <math.h>
#include <iostream>
#include <string>
#include <vector>

#include <KDetailedException.h>

#include <ActuatorConfigClientRpc.h>
#include <BaseClientRpc.h>
#include <BaseCyclicClientRpc.h>
#include <SessionClientRpc.h>
#include <SessionManager.h>

#include <RouterClient.h>
#include <TransportClientTcp.h>
#include <TransportClientUdp.h>

#include <google/protobuf/util/json_util.h>

#if defined(_MSC_VER)
#include <Windows.h>
#else
#include <unistd.h>
#endif
#include <time.h>

namespace k_api = Kinova::Api;

#define IP_ADDRESS "192.168.1.10"

#define PORT 10000
#define PORT_REAL_TIME 10001

int main(int argc, char** argv) {
    // Create API objects
    auto error_callback = [](k_api::KError err) {
        cout << "_________ callback error _________" << err.toString();
    };

    auto transport = new k_api::TransportClientTcp();
    auto router = new k_api::RouterClient(transport, error_callback);
    transport->connect(IP_ADDRESS, PORT);

    auto transport_real_time = new k_api::TransportClientUdp();
    auto router_real_time =
        new k_api::RouterClient(transport_real_time, error_callback);
    transport_real_time->connect(IP_ADDRESS, PORT_REAL_TIME);

    // Set session data connection information
    auto create_session_info = k_api::Session::CreateSessionInfo();
    create_session_info.set_username("admin");
    create_session_info.set_password("admin");
    create_session_info.set_session_inactivity_timeout(
        60000);  // (milliseconds)
    create_session_info.set_connection_inactivity_timeout(
        2000);  // (milliseconds)

    // Session manager service wrapper
    std::cout << "Creating sessions for communication" << std::endl;
    auto session_manager = new k_api::SessionManager(router);
    session_manager->CreateSession(create_session_info);
    auto session_manager_real_time =
        new k_api::SessionManager(router_real_time);
    session_manager_real_time->CreateSession(create_session_info);
    std::cout << "Sessions created" << std::endl;

    // Create services
    auto base = new k_api::Base::BaseClient(router);
    auto base_cyclic =
        new k_api::BaseCyclic::BaseCyclicClient(router_real_time);

    auto actuator_config =
        new k_api::ActuatorConfig::ActuatorConfigClient(router);

    // Set joint 7 to velocity control mode
    auto control_mode_message = k_api::ActuatorConfig::ControlModeInformation();
    control_mode_message.set_control_mode(
        Kinova::Api::ActuatorConfig::ControlMode::VELOCITY);
    actuator_config->SetControlMode(control_mode_message, 7);

    auto torque_offset_message = k_api::ActuatorConfig::TorqueOffset();
    torque_offset_message.set_torque_offset(0.01f);
    actuator_config->SetTorqueOffset(torque_offset_message, 7);

    auto control_mode = actuator_config->GetControlMode(7);
    std::cout << "ControlMode: " << control_mode.control_mode() << std::endl;
    
    auto torque_offset = actuator_config->GetTorqueOffset(7);
    std::cout << "TorqueOffset: " << torque_offset.torque_offset() << std::endl;

    // Close API session
    session_manager->CloseSession();
    session_manager_real_time->CloseSession();

    // Deactivate the router and cleanly disconnect from the transport object
    router->SetActivationStatus(false);
    transport->disconnect();
    router_real_time->SetActivationStatus(false);
    transport_real_time->disconnect();

    // Destroy the API
    delete base;
    delete base_cyclic;
    delete session_manager;
    delete session_manager_real_time;
    delete router;
    delete router_real_time;
    delete transport;
    delete transport_real_time;
}
```

grizzi, 2020-03-09:
Quoting @Eglen24 

> Currently the admittance modes of Gen3 have deadbands to compensate for small errors in torque measurements or offsets which prevent the arm from moving on its own in static positions. These [...]

In terms of a control algorithm, we would like to command x torque at the joint and get x torque. Does what you say mean that I cannot rely on torque tracking at all because of all this additional post processing?

tanguy-s, 2020-03-12:
@grizzi You should not have any issues to achieve accurate torque control using the robot's feedback if you manage to set all torque offsets w.r.t to your control model.
@Eglen24 is commenting on the specifics of the admittance mode underlying torque controller which is designed with large safety margins in order to work out of the box in all configurations.

grizzi, 2020-03-12:
@tanguy-s  What do you mean by "using the robot feedback"? I was expecting that I can send in open loop e specific torque value and have it tracked. Instead what I experience is that I send a torque command (let's say 4 Nm on joint 5) and the measured torque is not what I command. Instead it is way smaller (around 0.8 Nm) and the joint starts moving like there were a huge friction or additional damping.

P.S. Before sending the command the arm is in the up position and the torque offsets are set to 0. 

P.P.S. If, after sending the command, I mechanically impede the joint to move, the measured torque instead, matches the command. This jusify why I think there is either a big uncompensated friction or this is done internally by kinova torque controller.

nberthelot-kino, 2020-04-21:
@tanguy-s 
Looking through the code sample you sent us, I see one possible cause for the error you are experiencing. Note that the ID of the various devices on the robot are not allocated in any specific order (as briefly mentionned here: https://github.com/Kinovarobotics/kortex/blob/master/linked_md/cpp_device_routing.md), so by hard coding the ID to 7, you may be polling another or even non-existent device. Setting the value publishes a message to the robot with no valid destination, but using the getter waits for an answer of this invalid ID, hence the error. The proper way to get device identifiers is to use the device manager ReadAllDevices() function (see previous link). 

However, even if you were to do this, you would find that your sample code still wouldn't work, because setting individual actuator torque offsets overwrites the values previously set for the entire system. You would need to first get the values of the offsets and then modify them according to the leftover offsets you are measuring, and then set them again using the API.

tanguy-s, 2020-04-22:
@nberthelot-kino thanks for the reply.

- Indeed, the list of devices returned by ReadAllDevices() is not ordered by device physical position nor device_identifier as the docs specifies. However, the device_identifiers themselves are ordered by physical position. Base is id 0, actuators are in range 1-7, interconnect 8 and vision 9 (on gen3). 
Testing with both methods leads to same results. 

- I did find the origin of the error, which corresponds to the "DEVICE_NOT_READY" returned. 
Adding a delay (1.5sec) after setting the TorqueOffset and before getting it gets around the problem:
```
auto torque_offset_message = k_api::ActuatorConfig::TorqueOffset();
torque_offset_message.set_torque_offset(0.01f);
actuator_config->SetTorqueOffset(torque_offset_message, 7);

usleep(1500000); // 1.5 second

auto torque_offset = actuator_config->GetTorqueOffset(7);
std::cout << "TorqueOffset: " << torque_offset.torque_offset() << std::endl;
```

I believe that this is a bug, as I'd expect the blocking function `SetTorqueOffset` to block until the actuator has successfully operated the modification (even though, 1.5 second seems quite long). 

This bug however is not in the scope of that issue, which can be closed.

alexvannobel, 2020-04-22:
Hi @tanguy-s ,

Glad you found the error's origin. The Kortex API calls are blocking, but return directly after receiving an answer from the device. It seems in this case the actuator applies the torque offset after having answered the RPC call, but by design actuators are forced to answer all calls within the millisecond, which enforces this kind of behaviour. The issue should probably be worked around in a Kortex API helper function, which would sleep for some time after receiving the response and before returning to the user. We plan on providing such helpers for the API in a future release, so I'll definitely bring this issue up internally.

Thanks for the feedback.
Best regards,
Alex

closed 2020-04-22 by alexvannobel