Kinovarobotics/Kinova-kortex2_Gen3_G3L #39: _ITERATOR_DEBUG_LEVEL

drhalftone, 2020-01-28:
I am trying to link the C++ API into a Qt project using Qt 5.12.5 MSVC2017 64bit and regardless of whether I compile release or debug, I can't make this error go away:

KortexApi.lib(common.cc.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2' in main.obj

Has anyone successfully linked the C++ library against a Qt5 project on Windows?

gstlaurentkinova, 2020-02-27:
Hi @drhalftone,

Can you specify which version of the Kortex Api you are currently using? Can you also specify what version of the run-time library you are using (/MD, /MDd, /MT, /MTd, etc.) ? 

There is a known issue specifically about this. It is fixed in the 2.2 version release.

I hope this answers your questions.

Best,

Gabriel

drhalftone, 2020-03-01:
I was not specifying the /MT flag, any of them.  I added them and it fixed my issue.

closed 2020-03-01 by drhalftone