Kinovarobotics/Kinova-kortex2_Gen3_G3L #42: Velocity control in low-level mode causes the robot to drift

DjoleMNE, 2020-02-26:
## Version
Kortex Device: 7DOF Gen3
Firmware Version: 2.0.1
KortexAPI: 2.0.0

We have two of these arms in our labs and the same problem occurs on both of them.

## Description

We have modified the provided example "01- actuator_low_level_velocity_control.cpp" in order to test velocity control via low-level API. More specifically, instead of commanding joints to move in this way:
```
commands[i] += (0.001f * velocity);
base_command.mutable_actuators(i)->set_position(fmod(commands[i], 360.0f));
````
, we have modified the code to switch to the explicit velocity interface. 

```
base_command.mutable_actuators(i)->set_velocity(velocity_commands[i]);
````
Here, the "velocity_commands" vector is constant and contains elements: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 5.0]. However, when the robot is controlled using this interface, the commanded joint velocities are not executed correctly. Namely, instead of staying still and only moving the last (7th) joint, the robot is drifting (falling downwards). Here, the second and fourth joints are moving due to gravity effects, which should not happen in a velocity control mode.

## Code example
Here, a minimal example is provided. Major changes were only made in the "example_actuator_low_level_velocity_control" method of the aforementioned example that is provided in this repository.

```
bool example_actuator_low_level_velocity_control(k_api::Base::BaseClient* base, k_api::BaseCyclic::BaseCyclicClient* base_cyclic, k_api::ActuatorConfig::ActuatorConfigClient* actuator_config)
{
    bool return_status = true;

    // Move arm to ready position
    example_move_to_home_position(base);
    
    k_api::BaseCyclic::Feedback base_feedback;
    k_api::BaseCyclic::Command  base_command;

    std::vector<float> velocity_commands(7, 0.0);
    velocity_commands[6] = 5.0;

    auto servoingMode = k_api::Base::ServoingModeInformation();

    int timer_count = 0;
    int64_t now = 0;
    int64_t last = 0;

    int timeout = 0;

    std::cout << "Initializing the arm for velocity low-level control example" << std::endl;
    try
    {
        // Set the base in low-level servoing mode
        servoingMode.set_servoing_mode(k_api::Base::ServoingMode::LOW_LEVEL_SERVOING);
        base->SetServoingMode(servoingMode);
        base_feedback = base_cyclic->RefreshFeedback();

        int actuator_count = base->GetActuatorCount().count();

        // Initialize each actuator to its current position
        for(int i = 0; i < actuator_count; i++)
        {
            base_command.add_actuators()->set_position(base_feedback.actuators(i).position());
        }

        // Define the callback function used in Refresh_callback
        auto lambda_fct_callback = [](const Kinova::Api::Error &err, const k_api::BaseCyclic::Feedback data)
        {
            // We are printing the data of the moving actuator just for the example purpose,
            // avoid this in a real-time loop
            std::string serialized_data;
            google::protobuf::util::MessageToJsonString(data.actuators(data.actuators_size() - 1), &serialized_data);
            std::cout << serialized_data << std::endl << std::endl;
        };

        auto control_mode_message = k_api::ActuatorConfig::ControlModeInformation();
        control_mode_message.set_control_mode(Kinova::Api::ActuatorConfig::ControlMode::VELOCITY);        
        for (int actuator_id = 1; actuator_id < actuator_count + 1; actuator_id++)
        {
            actuator_config->SetControlMode(control_mode_message, actuator_id);
        }

        // Real-time loop
        while(timer_count < (time_duration * 1000))
        {
            now = GetTickUs();
            if(now - last > 1000)
            {
                base_feedback = base_cyclic->RefreshFeedback();
                for(int i = 0; i < actuator_count; i++)
                {
                    base_command.mutable_actuators(i)->set_position(base_feedback.actuators(i).position());
                    base_command.mutable_actuators(i)->set_velocity(velocity_commands[i]);
                }

                // Incrementing identifier ensures actuators can reject out of time frames
                base_command.set_frame_id(base_command.frame_id() + 1);
                if (base_command.frame_id() > 65535) base_command.set_frame_id(0);

                for (int idx = 0; idx < actuator_count; idx++)
                {
                    base_command.mutable_actuators(idx)->set_command_id(base_command.frame_id());
                }

                try
                {
                    base_cyclic->Refresh(base_command, 0);
                }
                catch(...)
                {
                    timeout++;
                }
                
                timer_count++;
                last = GetTickUs();
            }
        }
    }
    catch (k_api::KDetailedException& ex)
    {
        std::cout << "Kortex error: " << ex.what() << std::endl;
        return_status = false;
    }
    catch (std::runtime_error& ex2)
    {
        std::cout << "Runtime error: " << ex2.what() << std::endl;
        return_status = false;
    }
 
    // Set back the servoing mode to Single Level Servoing
    servoingMode.set_servoing_mode(k_api::Base::ServoingMode::SINGLE_LEVEL_SERVOING);
    base->SetServoingMode(servoingMode);

    // Wait for a bit
    std::this_thread::sleep_for(std::chrono::milliseconds(2000));

    return return_status;
}

```
In the "main" method:

 - A new line is added: 
    ```
     auto actuator_config = new k_api::ActuatorConfig::ActuatorConfigClient(router);
    ```
 - The "example_actuator_low_level_velocity_control" method is now called with:
   ```
    auto isOk = example_actuator_low_level_velocity_control(base, base_cyclic, actuator_config);
   ```
I hope I can get some help from you in identifying a solution to the aforementioned problem.
Thank you in advance!

Best regards,
Djordje Vukcevic

alexvannobel, 2020-03-10:
Hi @DjoleMNE ,

The actual implementation of velocity control in the actuators causes your velocity to be directly feeded to the velocity control loop, which can cause a certain delay in the command and the drifting you have seen. We may address this in an upcoming release, so I will let this story opened.

In the meantime, you obtain a correct behaviour if you calculate yourself the position increments to command, based on your commanded velocity (as the example already did with the `velocity` variable). This will give the result you want.

Hope this helps,
Best,
Alex

DjoleMNE, 2020-03-11:
Hi Alex,

thanks for the reply! 

However, it is important for our controllers to have the ability to alternate between the low-level (real) velocity and torque control, during the task execution. 

Can you share with us (approximate) release date of the upcoming firmware?

Best regards,
Djordje

alexvannobel, 2020-04-21:
Hi @DjoleMNE ,

Sorry to come back to you so late. The COVID-19 situation has slowed down the development here so I don't have a release date for you. I'll make sure to keep this post updated when I get the information.

Sorry for the inconvenience,

Alex

DjoleMNE, 2020-04-21:
Hi @alexvannobel,

thanks for the update.

Best regards,
Djordje

colinator, 2020-12-11:
Any word on this? I also see this drift phenomenon in low-level velocity control.

DjoleMNE, 2021-06-09:
Hi @alexvannobel,

was this issue addressed in this new 2.3 firmware release?

Best regards,
Djordje

felixmaisonneuve, 2021-06-14:
Hi @DjoleMNE, 

This issue was not addressed in this new release

Sorry for the inconvenience,

Felix

colinator, 2021-11-05:
Any word on this? We would really like to have low-level velocity control.

closed 2024-07-17 by martinleroux