Kinovarobotics/Kinova-kortex2_Gen3_G3L #43: mismatch detected for 'RuntimeLibrary'

drhalftone, 2020-03-04:
This is a follow up to issue #39, which is closed.  In that issue, you recommended upgrading my libraries to 2.2.  I did that; however, I would point out that it only includes Release libraries.  That aside, I'm using the following text in my Qt .pro file:

INCLUDEPATH += $$PWD/kortex_api/MT/include
INCLUDEPATH += $$PWD/kortex_api/MT/include/client
INCLUDEPATH += $$PWD/kortex_api/MT/include/client_stubs
INCLUDEPATH += $$PWD/kortex_api/MT/include/common
INCLUDEPATH += $$PWD/kortex_api/MT/include/google
INCLUDEPATH += $$PWD/kortex_api/MT/include/messages

DEPENDPATH  += $$PWD/kortex_api/MT/include
DEPENDPATH  += $$PWD/kortex_api/MT/include/client
DEPENDPATH  += $$PWD/kortex_api/MT/include/client_stubs
DEPENDPATH  += $$PWD/kortex_api/MT/include/common
DEPENDPATH  += $$PWD/kortex_api/MT/include/google
DEPENDPATH  += $$PWD/kortex_api/MT/include/messages

LIBS        += -L$$PWD/kortex_api/MT/lib/release -lKortexApiCpp

Now I've tried MD as well as MT folders.  So keep that in mind when I compile.  I now receive the following errors (a lot of them).  These are the first three:

KortexApiCpp.lib(generated_message_reflection.cc.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in main.obj
KortexApiCpp.lib(Frame.pb.cc.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in main.obj
KortexApiCpp.lib(KError.cpp.obj) : error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in main.obj

I'm using VS2017 x64 with Qt 5.12.5

alexvannobel, 2020-03-05:
Hi @drhalftone ,
We double checked on our side and our MT and MD builds are correct.
I have trouble reading the error, but from what I understand in `error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in main.obj`:
- MT_StaticRelease is a value in the Kortex object
- MD_DynamicRelease is a value in main.obj (which is yours I'm guessing)

Can you show your whole .pro file? It seems in your snippet that you defines where is Kortex API, but I don't see a line where you set either the /MT or the /MD flag.

If you could show the errors you get when using our MT version and also the errors you get when using our MD version that would be very helpful.

Thanks in advance, 
Cheers,
Alex

drhalftone, 2020-03-05:
Do you have qt installed on your side? I can send you the whole project. Or I can make my machine available to you through chrome Remote Desktop or similar vnc tool. Or I could ship you the whole laptop.

alexvannobel, 2020-03-05:
I contacted you by email.
I'm installing the VS2017 build tools and I already have Qt, I'll be able to test your project when you send it to me.

drhalftone, 2020-03-05:
I’m away from my windows machine that has the latest version of the PRO file; however, I do have a two-day old version on my One Drive, which is available on my Mac.  So here are the files making up the project.  Inside the same folder as these files, I have a kortex_api folder.  Inside of this, I have the MD and MT folders from the 2.20 version release.  So inside the MD and MT folders include and lib.  For actually calling Kortex API commands, I’ve simply copied over one of the example files into the laukinovawidget.cpp file.

If you have any questions or need the latest and greatest version, feel free to reach out to me using the contact info below, but email is the best way to reach me.

Daniel L. Lau
President and CEO and Certified Professional Engineer
Lau Consulting Inc.
774 Bravington Way
Lexington, KY 40503

phone: 1-859-312-8047
email: lauconsulting@windstream.net
web: http://www.drhalftone.com

P Please consider the environment before printing this email.
 
CONFIDENTIALITY NOTICE: The information in this email may be confidential and/or privileged. This email is intended to be reviewed by only the individual or organization named above. If you are not the intended recipient or an authorized representative of the intended recipient, you are hereby notified that any review, dissemination or copying of this email and its attachments, if any, or the information contained herein is prohibited. If you have received this email in error, please immediately notify the sender by return email and delete this email from your system.

alexvannobel, 2020-03-05:
I don't see the files you are referring to. Can you email them at me at avannobel@kinova.ca? I tried messaging you but I get a weird "address not found error" when I send the mail. 
I will probably just try and build the simple C++ Kortex API example at first with VS2017 inside Qt with a .pro file.

drhalftone, 2020-03-05:
In five minutes I’ll resend

alexvannobel, 2020-03-05:
Hi again @drhalftone ,
I built the code just fine by modifying the .pro file you sent me just a little. Here is the complete file I used (the source code was left unchanged):
```
QT     += core gui widgets

CONFIG         += c++11

# The following define makes your compiler emit warnings if you use
# any Qt feature that has been marked deprecated (the exact warnings
# depend on your compiler). Please consult the documentation of the
# deprecated API in order to know how to port your code away from it.
DEFINES += QT_DEPRECATED_WARNINGS

# You can also make your code fail to compile if it uses deprecated APIs.
# In order to do so, uncomment the following line.
# You can also select to disable deprecated APIs only up to a certain version of Qt.
#DEFINES += QT_DISABLE_DEPRECATED_BEFORE=0x060000    # disables all the APIs deprecated before Qt 6.0.0

SOURCES += main.cpp \
           laukinovawidget.cpp

HEADERS += laukinovawidget.h

TRANSLATIONS += LAUKinovaWidget_en_US.ts

# Default rules for deployment.
qnx: target.path = /tmp/$${TARGET}/bin
else: unix:!android: target.path = /opt/$${TARGET}/bin
!isEmpty(target.path): INSTALLS += target

DEFINES     += _OS_WINDOWS

#QMAKE_CXXFLAGS_RELEASE += /MT

INCLUDEPATH += $$PWD/kortex_api/MD/include
INCLUDEPATH += $$PWD/kortex_api/MD/include/client
INCLUDEPATH += $$PWD/kortex_api/MD/include/client_stubs
INCLUDEPATH += $$PWD/kortex_api/MD/include/common
INCLUDEPATH += $$PWD/kortex_api/MD/include/google
INCLUDEPATH += $$PWD/kortex_api/MD/include/messages

DEPENDPATH  += $$PWD/kortex_api/MD/include
DEPENDPATH  += $$PWD/kortex_api/MD/include/client
DEPENDPATH  += $$PWD/kortex_api/MD/include/client_stubs
DEPENDPATH  += $$PWD/kortex_api/MD/include/common
DEPENDPATH  += $$PWD/kortex_api/MD/include/google
DEPENDPATH  += $$PWD/kortex_api/MD/include/messages

LIBS        += -L$$PWD/kortex_api/MD/lib/release -lKortexApiCpp -lwinMM -lws2_32
```
I built with the MD flag (dynamic threading), which is the default. I added two Windows libraries and corrected a path in the LIBS line. 
I also added a line to build with static threading, but I got a single error concerning a system library complaining that is is built with dynamic support. I don't know how to fix this one, but clearly it does not relate to Kortex API.
Anyway, here is my build output:
```
12:55:21: Running steps for project LAUKinovaWidget...
12:55:21: Starting: "C:\Qt\5.7\msvc2015_64\bin\qmake.exe" C:\stash\drhalftone\LAUKinovaWidget.pro -spec win32-msvc2015 "CONFIG+=qtquickcompiler"
12:55:23: The process "C:\Qt\5.7\msvc2015_64\bin\qmake.exe" exited normally.
12:55:23: Starting: "C:\Qt\Tools\Preview\Qt Creator 4.11.0-rc1\bin\jom.exe" -f C:/stash/build-LAUKinovaWidget-Qt_5_7_1_MSVC2017_64bit-Release/Makefile qmake_all

jom 1.1.3 - empower your cores

12:55:23: The process "C:\Qt\Tools\Preview\Qt Creator 4.11.0-rc1\bin\jom.exe" exited normally.
12:55:23: Starting: "C:\Qt\Tools\Preview\Qt Creator 4.11.0-rc1\bin\jom.exe" 
	"C:\Qt\Tools\Preview\Qt Creator 4.11.0-rc1\bin\jom.exe" -f Makefile.Release
	cl -c -nologo -Zc:wchar_t -FS -Zc:strictStrings -Zc:throwingNew -O2 -MD -GR -W3 -w34100 -w34189 -w44996 -w44456 -w44457 -w44458 -wd4577 -wd4467 -EHsc -DUNICODE -DWIN32 -DQT_DEPRECATED_WARNINGS -D_OS_WINDOWS -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DNDEBUG -I..\drhalftone -I. -I..\drhalftone\kortex_api\MD\include -I..\drhalftone\kortex_api\MD\include\client -I..\drhalftone\kortex_api\MD\include\client_stubs -I..\drhalftone\kortex_api\MD\include\common -I..\drhalftone\kortex_api\MD\include\google -I..\drhalftone\kortex_api\MD\include\messages -I..\..\Qt\5.7\msvc2015_64\include -I..\..\Qt\5.7\msvc2015_64\include\QtWidgets -I..\..\Qt\5.7\msvc2015_64\include\QtGui -I..\..\Qt\5.7\msvc2015_64\include\QtANGLE -I..\..\Qt\5.7\msvc2015_64\include\QtCore -Irelease -I..\..\Qt\5.7\msvc2015_64\mkspecs\win32-msvc2015 -Forelease\ @C:\Users\AVANNO~1\AppData\Local\Temp\main.obj.6760.16.jom
main.cpp
	C:\Qt\5.7\msvc2015_64\bin\moc.exe -DUNICODE -DWIN32 -DQT_DEPRECATED_WARNINGS -D_OS_WINDOWS -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DNDEBUG -D_MSC_VER=1900 -D_WIN32 -IC:/Qt/5.7/msvc2015_64/mkspecs/win32-msvc2015 -IC:/stash/drhalftone -IC:/stash/drhalftone/kortex_api/MD/include -IC:/stash/drhalftone/kortex_api/MD/include/client -IC:/stash/drhalftone/kortex_api/MD/include/client_stubs -IC:/stash/drhalftone/kortex_api/MD/include/common -IC:/stash/drhalftone/kortex_api/MD/include/google -IC:/stash/drhalftone/kortex_api/MD/include/messages -IC:/Qt/5.7/msvc2015_64/include -IC:/Qt/5.7/msvc2015_64/include/QtWidgets -IC:/Qt/5.7/msvc2015_64/include/QtGui -IC:/Qt/5.7/msvc2015_64/include/QtANGLE -IC:/Qt/5.7/msvc2015_64/include/QtCore -I. ..\drhalftone\laukinovawidget.h -o release\moc_laukinovawidget.cpp
	cl -c -nologo -Zc:wchar_t -FS -Zc:strictStrings -Zc:throwingNew -O2 -MD -GR -W3 -w34100 -w34189 -w44996 -w44456 -w44457 -w44458 -wd4577 -wd4467 -EHsc -DUNICODE -DWIN32 -DQT_DEPRECATED_WARNINGS -D_OS_WINDOWS -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -DNDEBUG -I..\drhalftone -I. -I..\drhalftone\kortex_api\MD\include -I..\drhalftone\kortex_api\MD\include\client -I..\drhalftone\kortex_api\MD\include\client_stubs -I..\drhalftone\kortex_api\MD\include\common -I..\drhalftone\kortex_api\MD\include\google -I..\drhalftone\kortex_api\MD\include\messages -I..\..\Qt\5.7\msvc2015_64\include -I..\..\Qt\5.7\msvc2015_64\include\QtWidgets -I..\..\Qt\5.7\msvc2015_64\include\QtGui -I..\..\Qt\5.7\msvc2015_64\include\QtANGLE -I.. [... cut ...]
12:55:28: The process "C:\Qt\Tools\Preview\Qt Creator 4.11.0-rc1\bin\jom.exe" exited normally.
12:55:28: Elapsed time: 00:07.
```
And here is my Kit configuration for the build, with VS2017 build tools:
[image]

A difference I see between our setups is that I am using Qt 5.7, but I don't think it relates to the particular errors you have. 

Can you try building with the .pro file I sent you? If you still get errors, can you send your kit configuration in Qt Creator and the complete output of your build with all the errors?

Thanks,
Cheers,
Alex

closed 2020-04-21 by alexvannobel