Kinovarobotics/Kinova-kortex2_Gen3_G3L #44: end-effector force control support

huiwenzhang, 2020-03-18:
Hi, 
Does this repository have APIs for end-effector force control? Alternatively, can I get the end-effector force feedback by using forward dynamics?
Thanks

polthergeist, 2020-04-03:
Hi @huiwenzhang,
Unfortunately, the API to send end-effector has not been release yet. As soon as I have a date for you I will keep you posted. 
About the force feedback, you can have it by calling the function RefreshFeedback from the BaseCyclic service. It returns an object that contains the complete feedback of the robot. Nested in that object there is the feedback from the base. Once there, look for:

tool_external_wrench_force_x
tool_external_wrench_force_y
tool_external_wrench_force_z
tool_external_wrench_torque_x
tool_external_wrench_torque_y
tool_external_wrench_torque_z

I hope this answer your question and don't hesitate if I was not clear enough or if you have any other question.

huiwenzhang, 2020-04-03:
@polthergeist Hi, thanks for your answer. I think it is exactly what I want. By the way, does the accompanying ros package have similar interface (topics or services) which allows to get the end-effector force information? I quickly tried the code, it seems there is no such kind of information available.

polthergeist, 2020-04-03:
I never tried it myself but the base feedback is published in the topic named: base_feedback. You can find the message structure [here](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/msg/generated/base_cyclic/BaseFeedback.msg).
Again, don't hesitate if you have any other question. ;)

alexvannobel, 2020-04-21:
Hi @huiwenzhang ,

I can add that yes, you can have this informatioon in the ROS driver if you look into the [base_feedback topic documentation](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_driver#robot-feedback-topics-out). The values have the same name as @polthergeist already pointed out.

If these wrench values are what you need, can this issue be closed?

Best,

Alex

huiwenzhang, 2020-04-22:
Thanks, I will try it later. Feel free to close this issue right now

alexvannobel, 2020-04-22:
Don't hesitate to reopen it or create a new one if you have trouble with the `base_feedback`.
Best,
Alex

closed 2020-04-22 by alexvannobel