Kinovarobotics/Kinova-kortex2_Gen3_G3L #45: Control dual arms with one PC

LongfeiProjects, 2020-03-31:
Hi there,

I am wondering if gen3 api allows one PC to control multiple arms, such as what gen2 did. 
 - If positive, I assume some configurations are needed, a small example would be very appreciated. 
 - If negative, is it already planned and when it might happen?

Regards!

polthergeist, 2020-03-31:
Hi Longfei!
It's been a long time. :) Yup no problem. When you initialize a connection with a robot, you need to create a transport object tha contains informations like the IP address of the robot. So if you want to communicate with, let's say 2 robot, you instantiate 2 transports with different addresses like this:

```
#define IP_ADDRESS "192.168.1.10"
#define PORT 10000
#define IP_ADDRESS2 "10.10.10.10"
#define PORT2 10000

auto error_callback = [](k_api::KError err){ cout << "_________ callback error _________" << err.toString(); };
    auto transport = new k_api::TransportClientTcp();
    auto router = new k_api::RouterClient(transport, error_callback);
    transport->connect(IP_ADDRESS, PORT);

auto error_callback2 = [](k_api::KError err){ cout << "_________ callback error 2 _________" << err.toString(); };
    auto transport2 = new k_api::TransportClientTcp();
    auto router2 = new k_api::RouterClient(transport2, error_callback2);
    transport2->connect(IP_ADDRESS2, PORT2);
```

As you can guess, IP_ADDRESS and IP_ADDRESS2 must be modified acordign to your needs but this is how you do it. If you want more, you can refer to our API Creation example [here](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/000-Getting_started/01-api_creation.cpp).

Dont hesitate if you need something else or I was not clear enough ;)

LongfeiProjects, 2020-04-01:
Haha Hugo! Glad to hear from you! Thanks for confirming that.
The code snippet provided is clear and straight forward. Just a little more regarding to the session. Do I need to have two session_managers as well?

A verbose version of code after your example:

```
auto create_session_info = k_api::Session::CreateSessionInfo();
create_session_info.set_username("admin");
create_session_info.set_password("admin");

// Do I need one or two sessions to handle two robots?
auto session_manager = new k_api::SessionManager(router);
session_manager->CreateSession(create_session_info);
auto session_manager2 = new k_api::SessionManager(router2);
session_manager2->CreateSession(create_session_info);

// I'm sure here two BaseClients needed to handle two robot motions
auto base = new k_api::Base::BaseClient(router);
ConstrainedJointAngles crazy_stuff;
base->PlayJointTrajectory(crazy_stuff); // first robot is commanded to do something interesting

auto base2 = new k_api::Base::BaseClient(router2);
Base::Action applaud;
base2->ExecuteAction(applaud); // second robot highly agree with that
```

polthergeist, 2020-04-02:
Hi Longfei ,
Yes you need to have a different session manager for each connection. Your code is right.
Can't wait to see what this "crazy_stuff" is!  :)

Don't hesitate if you have other questions,

LongfeiProjects, 2020-04-03:
Thanks, I'll be able to apply that once the second arm arrives :-) 
I'll "join" this thread for now.

closed 2020-04-03 by LongfeiProjects

DjoleMNE, 2020-04-16:
Hi Hugo and Longfei,

thanks for providing the code examples for control of multiple robot arms on one PC!

Here is a simple video-demo of the above-mentioned control (the video requires some time to load, once you press play): https://slack-files.com/T3BP7R57W-F011VLESEFL-e6e22a6156

In the first phase, robots are commanded using high-level trajectory following control. Once the goal states are reached, the control switches to the low-level torque control and compensates gravity based on externally implemented simple (Recursive-Newton-Euler based) gravity-compensator.

Cheers,
Djordje

LongfeiProjects, 2020-04-28:
Hi @DjoleMNE, that's great to see you have already had some results on this topic. Is your code available on github somewhere? My second arm may arrive in two weeks, I'd like to try master-slave following with force/torque control as well.

DjoleMNE, 2020-05-01:
Hi @LongfeiProjects, 

yes, it is available and the specific application file that deals with the dual-arm system is located at https://github.com/DjoleMNE/MT_testing/blob/master/Controller/src/main_dual_kinova.cpp. In this file, the methods of interest are "go_to" , "run_test". The software mediator created for this particular robot is located at https://github.com/DjoleMNE/MT_testing/blob/master/Controller/src/kinova_mediator.cpp.

"Master-slave following with force/torque control" sounds like a cool project. However, I didn't yet test the **external-wrench** estimates/feedback (for the end-effector)  provided by the BaseCyclic service.

Nevertheless, it would be useful to get the information from Kinova engineers on how they compute these estimates/feedback. My guess is that they perform a forward propagation of the (sensed) external torques on the joints, from joint space to the Cartesian space. The approach there might be similar to the work in:
- Emanuele Magrini, Fabrizio Flacco, Alessandro De Luca, "Estimation of Contact Forces using a Virtual Force Sensor", International Conference on Intelligent Robots and Systems (IROS), 2014.

Maybe, @polthergeist and @alexvannobel can share with us more information on the external force-torque estimations for the end-effector. :) 

Best regards,
Djordje