Kinovarobotics/Kinova-kortex2_Gen3_G3L #49: How to use depth stream on Python

jonenfabian, 2020-05-28:
Hi,
I'm trying to get depth images from the KINOVA's Intel® RealSense™ Depth Module D410 using Python. I have already attempted to use cv2.VideoCapture, but this only works for the RGB camera. 
Python version: 3.6.7
Firmware versions: 2.0.1-72
Kortex API version: 2.0.0
Thank you in advance!

nberthelot-kino, 2020-05-28:
Hi @jonenfabian, 

I invite you to take a look at:
https://github.com/Kinovarobotics/kortex/tree/master/api_python/examples/500-Gen3_vision_configuration

Hope this can help!

jonenfabian, 2020-05-28:
Hello @nberthelot-kino, thanks for your reply.
Sadly these examples only demonstrate how to read and set intrinsic parameters like image size, or check which sensors are available, but not how to retrieve an actual frame.
I still have no clue.

alexvannobel, 2020-05-28:
Hi @jonenfabian ,

Sorry about pointing you to the wrong place.
I'm not very familiar with acquiring the color/depth stream in Python so I'll ask a colleague more proficient than I and come back to you shortly!

Best,
Alex

jonenfabian, 2020-05-28:
@alexvannobel, many thanks, looking forward to your reply!

Greetings
Fabian

alexvannobel, 2020-05-28:
Hi again @jonenfabian ,

Basically, Python has OpenCV bindings and OpenCV has bindings under the hood to GStreamer so you can acquire 8-bit color images with `cv2.VideoCapture` as you did.
However, OpenCV doesn't have the under-the-hood bindings to GStreamer to get the depth images from the 16-bit raw stream.

I think the most simple solution here for you to access both the depth and color image from Python would be to use the [ros_kortex_vision](https://github.com/Kinovarobotics/ros_kortex_vision) bindings we developed and construct a Python ROS node, because we don't have pure Python bindings for getting the depth stream in Python. You can have a look at the repo's [readme](https://github.com/Kinovarobotics/ros_kortex_vision/blob/master/README.md) file to learn on usage and see examples.

I hope this answers your question.
Best regards,

Alex

jonenfabian, 2020-05-28:
Excellent, I will try that. Thanks for the quick response!
@eduardgr check out the above answer.

closed 2020-05-28 by jonenfabian

eduardgr, 2020-05-28:
> Excellent, I will try that. Thanks for the quick response!
> @eduardgr check out the above answer.

Thanks for letting me know!!