Kinovarobotics/Kinova-kortex2_Gen3_G3L #51: End-effector orientation offset Z-axis

ngeerlingsriwo, 2020-06-03:
## Description

The orientation of the end-effector has a 90 deg offset on the Z-axis.

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI : 2.1.1-130

Kortex Device : WO521050-1 (Gen3lite with 2f gripper)

## Steps to reproduce

1. Use the Actions menu in the web-app to move the robot to Zero configuration.
2. Verify that the robot is in the exact same configuration as in Figure 63 in the Kinova Gen3 lite robot User Guide (page 122). The tool and base frame are now aligned, therefore the rotation matrix should be the identity matrix. Thus the extrinsic Z-Y-X Tait-Bryan angles should be 0(X), 0(Y) and 0(Z).
3. Go to the Monitoring page in the webapp or open the actuation windown in Pose mode. Here you will see that the angles are 0(X), 0(Y) and 90(Z). How can this be?

4. Run the ROS launch file (roslaunch kortex_driver kortex_driver.launch arm:=gen3_lite ip_address:=192.168.2.10 start_moveit:=false vision:=false gripper:=gen3_lite_2f).
5. Go to rViz and flip open the TF/Frames/end_effector_link/Orientation tab. Here you will see that the quaternion is 0 (X), 0(Y), 0(Z), 1(W), which corresponds to the identity matrix, but conflicts with the angles in the web-app.

## Code example

## Expected behavior

I would expect the angles to be 0-0-0 in this case.

## Any other information

I would like to have the option to send and receive the end-effector orientation as a quaternion, since this is the default way to describe orientations in ROS. I am currently figuring out how to do the conversion between your convention and quaternions.

alexvannobel, 2020-06-08:
Hi @ngeerlingsriwo ,

We changed the orientation of the end-effector to match the Gen3's end effector orientation just before the release and it looks like the user guide and a parameter in the ROS URDFs didn't follow along. The arm's feedback is correct. We're working on fixing the documentation and I'll fix the URDF.
I'll come back to you very shortly with the fix.

Thanks for the feedback,
Best,
Alex

alexvannobel, 2020-06-09:
Hi @ngeerlingsriwo ,

There were two causes to this : 
1. The user guide was not correctly updated after the tool frame change before the release.
2. The `tool_frame` was not correctly referenced in the robot_description and MoveIt. The PR referenced in this issue added the tool_frame as it is considered by the arm's firmware.

This is the updated frame definition for the arm : 

[image]

This is the updated transforms : 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Unfortunately, changing the user guide takes more time and internal steps than answering on Github, so I'll let the issue open and pin it until the user guide is updated and put on our website.

Best,
Alex

felixmaisonneuve, 2021-06-11:
The documention of the Gen3 lite shows the sames values as the ones @alexvannobel mentionned. 

I will close this issue

closed 2021-06-11 by felixmaisonneuve