Kinovarobotics/Kinova-kortex2_Gen3_G3L #52: Tool wrench feedback not updating when low level control mode

rshum19, 2020-06-19:
## Description

When in low level control mode I'm trying to retrieve the tool wrench information however the values are not updating. I'm using the the following commands to retrieve the data:

```
feedback = base_cyclic->RefreshFeedback();
feedback.base().tool_external_wrench_torque_z();
```

Further, when in low-level control end-effector values in the Web App don't' get updated either. 

This issue resembles that from #36. Can you confirm if base feedback updates when in low-level control?

alexvannobel, 2020-07-14:
Hi @rshum19 ,
Sorry for the delay.

Since the low level control mode completely bypasses the Control library, the calculations are no longer made, including the tool wrench values (basically all the values that need the URDF to calculate them). There is no actual workaround for that, but you do have access to the individual actuators's torque sensing values.

Alex

DjoleMNE, 2020-09-07:
Hi @rshum19, this issue also exists on the Kinova Gen3 robots in my lab.

However, I have implemented a standalone method for estimating the external wrench on the robot's tool-tip. The computations are made using solvers from the KDL library and I got solid results from this estimator:

```
// Momentum-observer for wrench estimation
int estimate_external_wrench(const KDL::JntArray &joint_position_measured,
                             const KDL::JntArray &joint_velocity_measured,
                             const KDL::JntArray &joint_torque_measured, 
                             KDL::Wrench &ext_force_torque)
{
    /**
    * ==========================================================================
    * Momentum observer, implementation based on:
    * S. Haddadin, A. De Luca and A. Albu-Schäffer,
    * "Robot Collisions: A Survey on Detection, Isolation, and Identification,"
    * in IEEE Transactions on Robotics, vol. 33(6), pp. 1292-1312, 2017.
    * ==========================================================================
    */
    int solver_result = this->dynamic_parameter_solver_->JntToMass(joint_position_measured, jnt_mass_matrix_);
    if (solver_result != 0) return solver_result;
    solver_result = this->dynamic_parameter_solver_->JntToCoriolis(joint_position_measured, joint_velocity_measured, coriolis_torque_);
    if (solver_result != 0) return solver_result;
    solver_result = this->dynamic_parameter_solver_->JntToGravity(joint_position_measured, gravity_torque_);
    if (solver_result != 0) return solver_result;

    jnt_mass_matrix_dot_.data = (jnt_mass_matrix_.data - previous_jnt_mass_matrix_.data) / DT_SEC_;
    previous_jnt_mass_matrix_.data = jnt_mass_matrix_.data;

    total_torque_estimation_.data = robot_state_.control_torque.data - gravity_torque_.data - coriolis_torque_.data + jnt_mass_matrix_dot_.data * joint_velocity_measured.data;
    estimated_momentum_integral_.data += (total_torque_estimation_.data + filtered_estimated_ext_torque_.data) * DT_SEC_;

    model_based_jnt_momentum_.data = jnt_mass_matrix_.data.lazyProduct(joint_velocity_measured.data);
    estimated_ext_torque_.data = wrench_estimation_gain_.asDiagonal() * (model_based_jnt_momentum_.data -
                                                                         estimated_momentum_integral_.data -
                                                                         initial_jnt_momentum_.data);

    // First order low-pass filter
    double alpha = 0.5;
    for (int i = 0; i < NUM_OF_JOINTS_; i++)
        filtered_estimated_ext_torque_(i) = alpha * filtered_estimated_ext_torque_(i) + (1.0 - alpha) * estimated_ext_torque_(i);

    /**
    * ========================================================================================
    * Propagate joint torques to Cartesian wrench using a pseudo inverse of Jacobian-Transpose
    * ========================================================================================
    */
    solver_result = jacobian_solver_.JntToJac(joint_position_measured, jacobian_end_eff_);
    if (solver_result != 0) return solver_result;

    solver_result = fk_pos_solver_->JntToCart(joint_position_measured, tool_tip_frame_full_model_);
    if (solver_result != 0) return solver_result;

    // Transform the jacobian from the base to the tool-tip frame
    jacobian_end_eff_.changeBase(tool_tip_frame_full_model_.M.Inverse());

    // Compute SVD of the jacobian using Eigen functions
    Eigen::JacobiSVD<Eigen::MatrixXd> svd(jacobian_end_eff_.data.transpose(), Eigen::ComputeThinU | Eigen::ComputeThinV);

    Eigen::VectorXd singular_inv(svd.singularValues());
    for (int j = 0; j < singular_inv.size(); ++j) singular_inv(j) = (singular_inv(j) < 1e-8) ? 0.0 : 1.0 / singular_inv(j);
    jacobian_end_eff_inv_.noalias() = svd.matrixV() * singular_inv.matrix().asDiagonal() * svd.matrixU().adjoint();

    // Compute End-Effector Cartesian forces from joint external torques
    Eigen::VectorXd wrench = jacobian_end_eff_inv_ * filtered_estimated_ext_torque_.data;
    for (int i = 0; i < NUM_OF_CONSTRAINTS_; i++) ext_force_torque(i) = wrench(i);

    return 0;
}
```

Hope it helps.

polthergeist, 2021-02-18:
Hi @rshum19 ,can we consider this issue closed? Do you have other questions regarding the low level control?

closed 2021-02-19 by rshum19

DjoleMNE, 2021-11-03:
The algorithm is now implemented/available in OROCOS KDL library: https://github.com/orocos/orocos_kinematics_dynamics/blob/master/orocos_kdl/src/chainexternalwrenchestimator.cpp

> Hi @rshum19, this issue also exists on the Kinova Gen3 robots in my lab.
> 
> However, I have implemented a standalone method for estimating the external wrench on the robot's tool-tip. The computations are made using solvers from the KDL library and I got solid results from this estimator:
> 
> ```
> // Momentum-observer for wrench estimation
> int estimate_external_wrench(const KDL::JntArray &joint_position_measured,
>                              const KDL::JntArray &joint_velocity_measured,
>                              const KDL::JntArray &joint_torque_measured, 
>                              KDL::Wrench &ext_force_torque)
> {
>     /**
>     * ==========================================================================
>     * Momentum observer, implementation based on:
>     * S. Haddadin, A. De Luca and A. Albu-Schäffer,
>     * "Robot Collisions: A Survey on Detection, Isolation, and Identification,"
>     * in IEEE Transactions on Robotics, vol. 33(6), pp. 1292-1312, 2017.
>     * ==========================================================================
>     */
>     int solver_result = this->dynamic_parameter_solver_->JntToMass(joint_position_measured, jnt_mass_matrix_);
>     if (solver_result != 0) return solver_result;
>     solver_result = this->dynamic_parameter_solver_->JntToCoriolis(joint_position_measured, joint_velocity_measured, coriolis_torque_);
>     if (solver_result != 0) return solver_result;
>     solver_result = this->dynamic_parameter_solver_->JntToGravity(joint_position_measured, gravity_torque_);
>     if (solver_result != 0) return solver_result;
> 
>     jnt_mass_matrix_dot_.data = (jnt_mass_matrix_.data - previous_jnt_mass_matrix_.data) / DT_SEC_;
>     previous_jnt_mass_matrix_.data = jnt_mass_matrix_.data;
> 
>     total_torque_estimation_.data = robot_state_.control_torque.data - gravity_torque_.data - coriolis_torque_.data + jnt_mass_matrix_dot_.data * joint_velocity_measured.data;
>     // total_torque_estimation_.data = -joint_torque_measured.data - gravity_torque_.data - coriolis_torque_.data + jnt_mass_matrix_dot_.data * joint_velocity_measured.data;
> 
>     estimated_momentum_integral_.data += (total_torque_estimation_.data + estimated_ext_torque_.data) * DT_SEC_;
> 
>     model_based_jnt_momentum_.data = jnt_mass_matrix_.data * joint_velocity_measured.data;
>     estimated_ext_torque_.data = wrench_estimation_gain_.asDiagonal() * (model_based_jnt_momentum_.data -
>                                                                          estimated_momentum_integral_.data -
>                                                                          initial_jnt_momentum_.data);
> 
>     // First order low-pass filter
>     double alpha = 0.5;
>     for (int i = 0; i < NUM_OF_JOINTS_; i++)
>         filtered_estimated_ext_torque_(i) = alpha * filtered_estimated_ext_torque_(i) + (1.0 - alpha) * estimated_ext_torque_(i);
> 
>     /**
>     * ========================================================================================
>     * Propagate joint torques to Cartesian wrench using a pseudo inverse of Jacobian-Transpose
>     * ========================================================================================
>     */
>     solver_result = jacobian_solver_.JntToJac(joint_position_measured, jacobian_end_eff_);
>     if (solver_result != 0) return solver_result;
> 
>     solver_result = fk_pos_solver_->JntToCart(joint_position_measured, tool_tip_frame_full_model_);
>     if (solver_result != 0) return solver_result;
> 
>     // Transform the jacobian from the base to the tool-tip frame
>     jacobian_end_eff_.changeBase(tool_tip_frame_full_model_.M.Inverse());
> 
>     // Compute SVD of the jacobian using Eigen functions
>     Eigen::JacobiSVD<Eigen::MatrixXd> svd(jacobian_end_eff_.data.transpose(), Eigen::ComputeThinU | Eigen::ComputeThinV);
> 
>     Eigen::VectorXd singular_inv(svd.singularValues());
>     for (int j = 0; j < singular_inv.size(); ++j) singular_inv(j) = (singular_inv(j) < 1e-8) ? 0.0 : 1.0 / singular_inv(j);
>     jacobian_end_eff_inv_.noalias() = svd.matrixV() * singular_inv.matrix().asDiagonal() * svd.matrixU().adjoint();
> 
>     // Compute End-Effector Cartesian forces from joint external torques
>     Eigen::VectorXd wrench = jacobian_end_eff_inv_ * filtered_estimated_ext_torque_.data;
>     for (int i = 0; i < NUM_OF_CONSTRAINTS_; i++) ext_force_torque(i) = wrench(i);
> 
>     return 0;
> }
> ```
> 
> Hope it helps.