Kinovarobotics/Kinova-kortex2_Gen3_G3L #60: Code for gamepad mappings on the Gen3

karthikm-0, 2021-01-25:
## Summary

I am not sure if this is the right place to post this, so I apologize ahead of time. I was looking through the Gen3 manual and saw that there are 3 modes of teleoperation that are possible through a gamepad: Twist linear, Twist angular, and Joint (https://www.kinovarobotics.com/sites/default/files/UG-014_KINOVA_Gen3_Ultra_lightweight_robot-User_guide_EN_R01.pdf). I am trying to mimic these mechanisms in a Unity simulation with an imported Gen3 urdf through the simple controller script Unity provides (https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md##Guide-to-write-your-own-controller). With this I can do the per-joint control, but I'm trying to build the other two mechanisms out (Twist linear and angular). I was wondering if there is any sample code on this that I can leverage as I am lost on how to implement this. Any help would be greatly appreciated!

polthergeist, 2021-02-18:
Hi @karthikm-0 , I'm very sorry for the late reply. You can find an example of twist command here:
https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/102-Movement_high_level/03-twist_command.cpp

Let me know if it is not what you are looking for.

karthikm-0, 2021-02-18:
Thanks for the response @polthergeist! I am wondering whether you have a Python API to perform IK (specific to Gen3). I am currently piping commands from a joystick and I'd just like to generate the correct joint angles without having to interact with the Gazebo simulation.

felixmaisonneuve, 2021-06-11:
Hi @karthikm-0,
With last week's update to 2.3, you can use the Python API to compute forward kinematics and inverse kinematics.

An example is currently on review and should be merge somewhere next week in the master branch.

In the mean time, you can find it in the feature/kinematics-example [here](https://github.com/Kinovarobotics/kortex/tree/feature/kinematics-example)

I will close this issue, if you still have a question regarding this feature, we will reopen it

Best,
Felix

closed 2021-06-11 by felixmaisonneuve