Kinovarobotics/Kinova-kortex2_Gen3_G3L #62: How to control 2F-85 and FT-300 on Kinova Gen3 via Kortex?

JDchen-Jaden, 2021-02-24:
## Summary

Are there examples of controlling 2F-85 and FT-300 via Kortex? It would be greatly appreciated if anyone can share an example code that includes the basics of allowing communication between Kinova Gen3 and the above two components.

felixmaisonneuve, 2021-06-11:
Hi @JDchen-Jaden,

Kinova provides examples on how to control grippers in high level ([106-gripper_command](https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/examples/106-Gripper_command)) and in low level ([107-Gripper_low_level_command](https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/examples/107-Gripper_low_level_command)). I have linked the C++ examples, but you can find similar examples in the python API.

Is this what you are looking for?

felixmaisonneuve, 2021-07-30:
Hi @JDchen-Jaden,

Is this what you were looking for? 
If I don't get an answer in the next weeks I will close this issue

Best,
Felix