Kinovarobotics/Kinova-kortex2_Gen3_G3L #64: Enable Joint Admittance Mode (freedrive) with Python API

jonenfabian, 2021-04-22:
| Arm type       | Firmware     | Release notes      | API |
| :------------- | :----------: | :-----------: | :-----------:
|  Gen3 | [2.2.1] | [2.2.0] | [2.2.0]

Hello everyone,

I'd like to activate the joint admittance mode of my Kinova Gen-3 with the Python API, so that i can push the robot wherever I want by applying forces.

To do so I thought I might just use ```base.SetAdmittance()``` like the following:
```
from kortex_api.autogen.messages import Base_pb2

my_admittance = Base_pb2.Admittance()
my_admittance.admittance_mode = Base_pb2.JOINT_ADMITTANCE # = 7
base.SetAdmittance(my_admittance)
```
But ```base.SetAdmittance``` throws the error:
```kortex_api.Exceptions.KServerException.KServerException: Server error name=ERROR_DEVICE, sub name=INVALID_PARAM => Invalid Control Mode```

Then I realized I might have to change the control mode.
I tried to change it with:
```
from kortex_api.autogen.client_stubs.ActuatorConfigClientRpc import ActuatorConfigClient
from kortex_api.autogen.messages import ActuatorConfig_pb2

control_mode_info = ActuatorConfig_pb2.ControlModeInformation()
control_mode_info.control_mode = 7 # from ControlConfig_pb2.ControlMode.JOINT_ADMITTANCE?
control_mode = ActuatorConfigClient(router)
control_mode.SetControlMode(control_mode_info)
```

But the line `SetControlMode` throws the error:
`Server error name=ERROR_PROTOCOL_SERVER, sub name=UNSUPPORTED_SERVICE => Service not registered for service id: 10`

How can I change the control mode in order to enable the admittance mode like pushing the right button on the wrist?

Thanks,

Fabian

jonenfabian, 2021-04-22:
Okay I found the mistake: the ```admittance_mode``` value I got from ```Base_pb2.JOINT_ADMITTANCE``` was 7. However in [the doc](https://github.com/Kinovarobotics/kortex/blob/ed5efb6ed6fbd259a35802a5c8f3f6bf3c3faa4c/api_python/doc/markdown/enums/Base/AdmittanceMode.md#) we can find the value 2 for joint admittance mode. This got me confused.
Setting the ```admittance_mode``` to 2 worked for me. With the value 4, the mode is disabled again. No need to manually change the Control Mode btw.

closed 2021-04-22 by jonenfabian

Myfirsttrying, 2022-12-16:
> Okay I found the mistake: the `admittance_mode` value I got from `Base_pb2.JOINT_ADMITTANCE` was 7. However in [the [...]

hi,  I want to achieve Cartesian admittance control , only need to change 7 to 1 ? Like this : And how i can quit this control mode whenever i want in my code ?
from kortex_api.autogen.client_stubs.ActuatorConfigClientRpc import ActuatorConfigClient
from kortex_api.autogen.messages import ActuatorConfig_pb2

control_mode_info = ActuatorConfig_pb2.ControlModeInformation()
control_mode_info.control_mode = 1
control_mode = ActuatorConfigClient(router)
control_mode.SetControlMode(control_mode_info)