Kinovarobotics/Kinova-kortex2_Gen3_G3L #65: Can't connect to robot

chrischu83117, 2021-05-20:
Hi everyone, recently I've tried run the api_python example, but somehow it's always report ConnectionRefusedError: [Errno 111] Connection refused.
My ipv4 ip is 192.168.100.100, so in utilities.py I  set ip to "192.168.100.100", and TCP_port I set to 55000.
Down below shows the setting of my Ethernet and robot:

[image]

[image]

I aslo tried several combination of port value but still can't connect to robot.
(I have turn off wifi, and there is only Ethernet connect to the computer)

Please give me some advice, thanks.

alexvannobel, 2021-06-04:
Hi @chrischu83117 ,

Do you have a Gen2 arm or a Gen3 arm?

The screenshots you sent are from DevelopmentCenter, which is our application to configure a Gen2 arm. This repository is our examples repository for Gen3 arms. Both arm generations work on different software platforms which are not compatible.

- If you have a Gen2, I suggest you have a look at the [user guide](https://www.kinovarobotics.com/en/resources/gen2-technical-resources) and also at [kinova-ros](https://github.com/Kinovarobotics/kinova-ros), our ROS repo for Gen2.

- If you have a Gen3, it is not the right way to connect to it and the [user guide](https://www.kinovarobotics.com/en/resources/gen3-technical-resources) would be a good place to start.

At any time, you can also reach out to support@kinova.ca for any question.

Let me know if this helps,
Best,
Alex

felixmaisonneuve, 2021-07-30:
Hi @chrischu83117 

Do you still have a problem connecting to your arm?
If I don't get an answer in the next weeks I will close this issue

Best,
Felix

chrischu83117, 2021-08-03:
@felixmaisonneuve Hi, now I am using usb connect method, but for Ethernet connection I still don't have any clue to do that.
Thank you for responding anyway.

closed 2021-08-03 by chrischu83117