Kinovarobotics/Kinova-kortex2_Gen3_G3L #72: Moveit implementation on Gen3 slow

sadofatai, 2021-07-29:
## Description

I am using Moveit Python Interface to control the Gen3 arm for a manipulation task. The movement of the arm is very slow compared to the simulation of the arm in gazebo. Any suggestion what can be the likely cause of this? I disabled the velocity limits but speed is still slow.

felixmaisonneuve, 2021-07-30:
Hi @sadofatai 

Can you provide an example of the code you are using? 
Are you using the ROS interface or are you using a python script that uses API calls directly to move the robot. 
The Issue is in our kortex repo, but you mentionned Gazebo wich I assume comes from our ros_kortex repo, so I just want to make this clear.

Regards,
Felix

sadofatai, 2021-10-07:
Hi Felix,
sorry for getting back late. This issue was resolved for me using the fix here: https://github.com/Kinovarobotics/ros_kortex/commit/db7a30acf753ca5559cbbc05de09733e95c20d66

Thank you for your support