Kinovarobotics/Kinova-kortex2_Gen3_G3L #74: Kinova Gen3 lite:  Base joystick / expansion connector pinout usage

alexlt97, 2021-09-09:
Good afternoon,

I was wondering if there some documentation about how to control the connector pinouts of the Base joystick that are shown in table 4 of Gen3_lite_USER_GUIDE.pdf. The idea is to use the expansion GPI pins to communicate with sensors via ROS to Kinova Gen3 lite so I can use  really simple sensors, pneumatic sensors.

[image]

felixmaisonneuve, 2021-09-10:
Hi @alexlt97 

Unfortunately, there is no usable connector in the base panel (except if you want to plug in a Xbox controller). 
It is not possible for you to communicate with connector pinout of the base joystick.

Sorry for the inconveniance,
Felix

closed 2021-09-10 by felixmaisonneuve

alexlt97, 2021-09-10:
But is clearly stated that you can use the expansion GPI pins to place a E-Stop button,  then why not similar inputs?

felixmaisonneuve, 2021-09-10:
What you are asking has not been made accessible throught the Kortex API. You cannot communicate with any of the connectors in the base panel, the current behaviour is pre-programmed and cannot be changed by the user.

Sorry,
Felix