Kinovarobotics/Kinova-kortex2_Gen3_G3L #81: API Support: Admittance mode with api driven position or torque offsets

phuicy, 2021-10-19:
Hi,

I am just struggling to follow the documentation, and I could do with some support. Realistically, the easiest way would be to test the code on the robot, but I am currently self-isolating sans robot.

I am trying to impose some virtual forces on the admittance mode of the robot, so that a user can still move the robot but can feel additional forces in some areas.

Is it possible to send additional joint torques or position commands to the robot whilst staying in admittance mode (in high or low-level mode)?

Ideally, this would be at a high frequency to get a clear resolution of forces.
Thus, is it possible to have the robot in low-level joint torque control mode, and admittance mode simultaneously?
Or how fast (~Hz) can I reliably send joint torque commands in high-level mode?

If I can't use the inbuilt admittance mode, and I need to implement my own admittance controller,  does the low-level torque control pre-compute gravity compensation, or do I need to recompute it? If I need to apply the gravity compensation myself is the robots function that computes it exposed in the API?

Additionally, How stiff is the non-admittance position mode, could a user interact with it?

Many thanks in advance,
Guy

phuicy, 2021-10-19:
[image]

This seems to be related info, presumable if I loop back the current position as a command, I will get a stable gravity compensated position, that I can also add a torque command to. IS that correct?

phuicy, 2021-10-19:
The above graph perhaps suggests that I can do it not in torque mode but in position control mode. Doesn't answer if I can use admittance mode with other control modes.

phuicy, 2021-11-02:
This is excellent thank you. Really useful. 

Looks like I will be implementing my own admittance controller on top of your low-level control mode. I will be able to make a gravity compensation, is there anything else that I need to do: torque ripple compensation or a like?

felixmaisonneuve, 2021-11-02:
As long as you take into account dynamic effects like gravity, accelerations, coriolis... (mainly gravity) it should be enough and you shouldn't have to worry about the rest.

phuicy, 2021-11-03:
Great! I will close for now, and add to the ticket as and when I test.

closed 2021-11-03 by phuicy