Kinovarobotics/Kinova-kortex2_Gen3_G3L #84: A better way to get access of joint speed

JDchen-Jaden, 2021-11-02:
Is there a better way to get access to each joint speed compared to the provided example as follows?

`    // Only print third joint`   
`       for (auto joint_angle : constrained_joint_angles.joint_angles().joint_angles())`   
`       {`   
`           if (joint_angle.joint_identifier() == 3)`   
`           {`   
`               std::cout << "joint 3 = " << joint_angle.value() << std::endl;`   
`               break;`   
`           }`   
`       }`

I am a bit confused about the description of `joint_speeds` (member). Its data type is `JointSpeed`. But how to access value for certain joint, e.g. `joint_angle.joint_identifier() == 3`. 
|Member name|Data type|Description|
|-----------|---------|-----------|
|joint\_speeds| [JointSpeed](JointSpeed.md#)|Array of joint speeds|

JDchen-Jaden, 2021-11-02:
Is it possible to share some examples to show how to access the real-time data of the robot, e.g., joint position, joint speed, joint torque? Thanks!

felixmaisonneuve, 2021-11-03:
Hi @JDchen-Jaden,

What you are looking for is the BaseCyclic function `RefreshFeedback`. This will return an object [Feedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/Feedback.md#) that contains all the information you need (and a lot more). 

You will do something similar to this : 
https://github.com/Kinovarobotics/kortex/blob/0aca9b23b0f04d76bf27e3f7b45befc2c77ebed9/api_cpp/examples/102-High_level_movement/01-move_angular_and_cartesian.cpp#L196-L201

Except you want informations about the actuators, so, in your case, it will be something like this : 
```c++
feedback.actuators(3).velocity()
```
In this example, `feedback` is the `Feedback` object the returned by `base_cyclic->RefreshFeedback()`. 
You then do `feedback.actuators(3)` to get the [ActuatorFeedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/ActuatorFeedback.md#) of the joint 3.

You can look at this exemple to see how to see how BaseCyclic is used, but I think the rest is pretty straight forward.

Regards,
Felix

JDchen-Jaden, 2021-11-03:
Thanks a million! @felixmaisonneuve

JDchen-Jaden, 2021-11-03:
Hi Felix @felixmaisonneuve 

`joint_speeds.clear_joint_speeds();` works well. However, it seems the feedback is only received once in the while loop #86 

```
           std::cout << "Joint 1 Speed: " << feedback.actuators(0).velocity() << std::endl;
            std::cout << "Joint 4 Speed: " << feedback.actuators(3).velocity() << std::endl;
            std::cout << "Joint 6 Speed: " << feedback.actuators(5).velocity() << std::endl;
```

```
Current Time: 0
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 29
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 50
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 75
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 100
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 125
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 150
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0
Current Time: 178
Joint 1 Speed: 0.506392
Joint 4 Speed: -0.235276
Joint 6 Speed: 0

```

JDchen-Jaden, 2021-11-03:
I should call `base_cyclic->RefreshFeedback()` in the loop. Thanks.