Kinovarobotics/Kinova-kortex2_Gen3_G3L #85: Kinova Gen3 lite does not properly power on, the status LED wont turn green.

alexlt97, 2021-11-03:
## Description

Kinova Gen3 lite does not properly power on. After the initial boot, the status LED is blue, the robot stays in that states and it never reaches the status Robot ready for use, status LED green.

I am able to ping the Kinova robot but I cannot access the web application:

[image]

## Any other information

In the booting process, the light turns on blue, then it turns off and again it turns blue and stays in that state.

felixmaisonneuve, 2021-11-03:
Hi @alexlt97,

Please contact support@kinova.ca about this issue.
There is not much I can do to help you with that.

Best,
Felix

alexlt97, 2021-11-04:
Hi @felixmaisonneuve,
Thank your response I contacted them, hopefully, we can discover the issue. This morning I turned it on and it seems to work properly again...

closed 2021-11-04 by alexlt97

DjoleMNE, 2023-05-05:
Hi @alexlt97,

we recently got into a similar issue with Gen3 arm. Can you, please, share with us the solution that you used to fix this problem? Thank you in advance!

Best regards,
Djordje

kyilmazKinova, 2023-08-09:
For people having this issue in the future, a factory reset seems to be the solution.

Here is the steps you need to follow :

1: Prepare an empty USB drive.
(Optional) Format the USB drive to FAT-32. If the drive is not already formatted, please note that formatting will erase all existing data. Make sure to back up any important files before proceeding.
3: Create the "restore" file. On the USB drive, create a new file named "restore" (without any file extension). Please note that on Windows, file extensions might not be visible by default. Ensure the file is completely empty and has no additional extensions like ".txt".
4: Safely remove the USB drive from your computer
5: Power off the robot completely using the power button switch.
6: Insert the USB drive into the designated port at the back of the robot base. Consult the Gen3 lite user guide if you're unsure about the exact location.
7: Power on the robot by flipping the power switch. The robot should automatically initiate the factory reset process.
8: Wait for the reset to complete. Allow the robot sufficient time to perform the factory reset. Once the robot is operational, the Status LED at the back will turn green.
9: Remove the USB drive from the robot base.

Important: Make sure to unplug the USB drive before restarting the robot. Otherwise, the factory reset mechanism will restart.

I would still urge you to contact Kinova support, we'll track the frequency of this issue and take further measures if necessary.