Kinovarobotics/Kinova-kortex2_Gen3_G3L #86: Exception Error: Invalid number of joints

JDchen-Jaden, 2021-11-03:
Hello!

I am learning how to use Kortex API. The example code that appears at the end is to implement high-level joint speed tracking. Vector7 is a 7 by 1 c++ Eigen vector. I did not change the number of joints, but the error shows that the number of joints is 14 as follows. There must be other reasons cause this error. It would be appreciated if anyone could suggest any solutions!

> ERROR
`Time Step: 24775
Current Time: 135344995869
Joint 4 Speed: -0.17066`
`terminate called after throwing an instance of 'Kinova::Api::KDetailedException'
  what():  Device error, Error sub type=VALUE_IS_ABOVE_MAXIMUM => <srv: 2, fct: 132, msgType: 3>`
`description: Invalid number of joints: 14, maximum allowed is 7
Aborted (core dumped)
`

> COMMAND
`    Vector7 q_dot;`  
`      q_dot << 2.0, 0.0, 0.0, -2.0, 0.0, 1.0, 0.0;`  
`      success &= example_send_joint_speeds(base,base_cyclic, q_dot);`

> CODE
`bool example_send_joint_speeds(k_api::Base::BaseClient* base, k_api::BaseCyclic::BaseCyclicClient* base_cyclic, Vector7 qdot_cmd)
{
    int timer_count = 0;
    int64_t now = 0;
    int64_t last = 0;

    std::cout << "Sending the angular velocities to the robot for 10 seconds..." << std::endl;

    k_api::Base::JointSpeeds joint_speeds;
    auto feedback = base_cyclic->RefreshFeedback();    
    int actuator_count = base->GetActuatorCount().count();

    if (actuator_count == 7)
    { 
        now = GetTickUs();
        while(timer_count < (time_duration * 1000))
        {
            for (size_t i = 0 ; i < qdot_cmd.rows(); ++i)
            {
                auto joint_speed = joint_speeds.add_joint_speeds();
                joint_speed->set_joint_identifier(i);
                joint_speed->set_value(qdot_cmd[i]);
                //joint_speed->set_duration(1);
            }
            base->SendJointSpeedsCommand(joint_speeds);        
            last = GetTickUs();
            std::cout << "Time Step: " << last - now << std::endl;
            std::cout << "Current Time: " << now << std::endl;
            std::cout << "Joint 4 Speed: " << feedback.actuators(3).velocity() << std::endl;
            timer_count++;
        }
    }
    // Stop the robot
    std::cout << "Stopping the robot" << std::endl;
    base->Stop();
    return true;
}`

felixmaisonneuve, 2021-11-03:
You loop twice in your `while`, which means you `joint_speeds.add_joint_speeds()` 7 times in your first loop, then 7 more times in the second loop (so `joint_speeds` now has 14 elements) and it fails. 

Adding `joint_speeds.clear()` at the beginning of your while loop should fix it

Best,
Felix

JDchen-Jaden, 2021-11-03:
Thanks a million! @felixmaisonneuve 
Now I have a much better understanding of JointSpeeds. Appreciate!

closed 2021-11-03 by JDchen-Jaden

JDchen-Jaden, 2021-11-03:
I think you meant `joint_speeds.clear_joint_speeds();` 
It solved the loop issue. Thanks again!