Kinovarobotics/Kinova-kortex2_Gen3_G3L #87: Request to read external torques

ShaoxiongYao, 2021-11-04:
## Summary

Request to read external torque at every joints.

## Use case

I would like to use the external force to estimate contact between the robot arm and the environment. This requires knowing detailed mass distribution of the robot arm. The equation to compute external torque can be found here: https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8059840, described in section III.

## Alternative

This external torque can be measured torque subtracting other terms in the dynamics equation. If I know the distribution of mass in the robot arm links, I may be able to compute this by myself.

## Possible issues

No, this only provides an extra value to read.

## Additional context

I am working on to reproduce the result from this paper https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=9561761. They work with a Kuka robot arm, that seems to have this feature.

felixmaisonneuve, 2021-11-04:
Hi @ShaoxiongYao,

I think what you are looking for is all documented in the Gen3 [User Guide](https://artifactory.kinovaapps.com/ui/api/v1/download?repoKey=generic-documentation-public&path=Documentation%2FGen3%2FTechnical%20documentation%2FUser%20Guide%2FUser%20Guide%20Gen3%20-%20R06.pdf). 

In the *Guidance for advanced users* section (found at the end), you can find Mass, Center of Mass, Inertia matrix as well as transformation matrix (and other information you might find usefull) for every joint : 
[image]

Regards,
Felix

DjoleMNE, 2021-11-05:
Hi @ShaoxiongYao,
similar request was discussed in issue #52.

Cheers

ShaoxiongYao, 2021-11-12:
@DjoleMNE Thank you for your reply! I wonder if you rely on the parameters provided in their documents. Did you find the inertial or mass value inaccurate? I tried the robot arm without any load and the measured torques are different from gravitational torques at static state. May I ask if you did any system identification on the mass matric or gravitational term?

felixmaisonneuve, 2022-04-12:
Hi @ShaoxiongYao,

Do you still need support on this topic? 
The torque computed by the arm uses the values found in the user guide, so both your result and the arm's result should be the same.

I will leave this issue open for a bit and close it if I don't get any answer.

Best,
Felix

ShaoxiongYao, 2022-04-12:
Hi @felixmaisonneuve, I had more comparisons on comparing computed and measured torques values. There is generally some error possibly caused by friction. I am not sure if your team may provide some analysis on that issue, like the standard deviation on the torque error per joint. It will be sincerely appreciated.

felixmaisonneuve, 2022-04-12:
The arm provides the `tool_external_wrench_force_x/y/z` and `tool_external_wrench_torque_x/y/z` values in the `BaseFeedback`.
These correspond to an estimate to the force and torque applied on the end effector based on the torque values from each joint. This seems to correspond to what you aim to do. 

There were some un-official tests performed in the past where we compared the estimated force and the actual force and there was an error of about 8% in a static position. No tests were performed in a dynamic environnement. 

If I understand your question correctly, you want to know the difference between the torque value we receive from the arm and the actual torque applied on each individual joint? I can try to find a margin error if you want, but I feel this may vary a lot for each individual actuator since it is impacted by the component used in the joint, wear and tear, variation in the manual assembly...

felixmaisonneuve, 2022-04-12:
Also, torque values can "drift" with time/usage. There a function in the webapp that allows you to reset the torque offset.
First, you put your arm in a vertical position (all joints to 0). The "Zero" preconfigured action available in the web app does that.
Once in a vertical position, all your joint's torque should be ~0 in _Systems->Monitoring_ in the webapp, but it can offset a bit with time.

You then go in _Configurations->Robot->Arm->Base->Set All Actuators Torque Offset to Zero_ and click on "Set Offset"

Now, if you go back to the _Monitoring_ section, you should see that each joint's torque is closer to 0

This might increase a bit your precision

Best,
Felix

ShaoxiongYao, 2022-04-12:
Thank you @felixmaisonneuve ! Your suggestions are very helpful to me :) I do agree that the torque measurements at each joint can be different from the expected value in an actuator-dependent way... I will try to reset the torques offsets to see if that helps. I am good to close the issue now.

closed 2022-04-12 by felixmaisonneuve

lexidliny, 2022-07-13:
> Thank you @felixmaisonneuve ! Your suggestions are very helpful to me :) I do agree that the torque measurements at each joint can be different from the expected value in an actuator-dependent way... I will try to reset the torques offsets to see if that helps. I am good to close the issue now.

hello,  I have the same problem, and I need to compensate the gravity. Have you solved this problem?

felixmaisonneuve, 2022-07-18:
Hi @lexidliny,

What issue are you having? Can you describe what you want and what is the problem you are currently facing?

It might be a good idea to open a new issue since this one was closed. I think the questions aksed here were answered accordingly. 

Best,
Felix