Kinovarobotics/Kinova-kortex2_Gen3_G3L #88: It is not clear that how to return an image (RGBD) array?

huangjiancong1, 2021-11-09:
## Summary

When I read the vision example(https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/500-Gen3_vision_configuration/04-vision_sensor_options.py), it is hard to figure how to return an image correctly.
It is better to share a tutorial that can help developer to use the Gen3 camera easily.#77 

By the way, do the parameters both intrinsics and extrinsic (https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/500-Gen3_vision_configuration/01-vision_intrinsics.py) are fixed to every Gen3 arm? 
If not, some tutorials that can help us find the suitable parameters set is much helpful.

felixmaisonneuve, 2021-11-10:
Hi @huangjiancong1,

There is no API call to access the color stream or the depth stream. 

There is a section in the User Guide (*Accessing Vision module color and depth streams*) that indicates how to access these, but basically it requires using Real Time Streaming Protocol (RTSP). You can stream color images in your web browser (or an external video viewer such as VLC viewer) at the URL `rtsp://192.168.1.10/color` and depth stream at `rtsp://192.168.1.10/depth` (where 192.168.1.10 is your arm's IP address). 

It is also possible to use GStreamer's tool `gst-launch-1.0`, see *Working with camera streams using GStreamer* section in the user guide.
For example, on Windows, you will use : (from the User Guide)
```
gst-launch-1.0.exe rtspsrc location=rtsp://192.168.1.10/color latency=30 ! rtph264depay ! avdec_h264 ! autovideosink
```

For the color stream, you can use OpenCV (with GStreamer) with `cv2.VideoCapture`, but it won't work with depth stream (because it is a 16-bit raw stream instead of an 8-bit color image). See #49.
It would look something like this 
```py
import cv2 as cv
vcap = cv.VideoCapture("rtspsrc location=rtsp://192.168.1.10/color latency=0 ! decodebin ! videoconvert ! appsink sync=false", cv.CAP_GSTREAMER)
ret, imageBGR = vcap.read()
cv.imshow('image', imageBGR)
cv.waitKey(1)
vcap.release()
cv.destroyAllWindows()
```

Finally, we provide [ros_kortex_vision](https://github.com/Kinovarobotics/ros_kortex_vision) that allows you to access both streams. There, you will find more documentation and examples on how to use it. If you must use both color and depth stream and you want to access/use the image data (and not simply stream the images), this is probably the easiest way to do it. 

To answer your second question, both intrinsics and extrinsics are fixed fixed to every Gen3 arm.

Regards,
Felix

huangjiancong1, 2021-11-13:
> Hi @huangjiancong1,
> 
> There is no API call to access the color stream or the depth stream.
> 
> There is a section in the User Guide (_Accessing Vision module color and depth streams_) that indicates how to access these, but basically it requires using Real Time Streaming Protocol (RTSP). You can stream color images in your web browser (or an external video viewer such as VLC viewer) at the URL `rtsp://192.168.1.10/color` and depth stream at `rtsp://192.168.1.10/depth` (where 192.168.1.10 is your arm's IP address).
> 
> It is also possible to use GStreamer's tool `gst-launch-1.0`, see _Working with camera streams using GStreamer_ section in the user guide. For example, on Windows, you will use : (from the User Guide)
> 
> ```
> gst-launch-1.0.exe rtspsrc location=rtsp://192.168.1.10/color latency=30 ! rtph264depay ! avdec_h264 ! autovideosink
> ```
> 
> For the color stream, you can use OpenCV (with GStreamer) with `cv2.VideoCapture`, but it won't work with depth stream (because it is a 16-bit raw stream instead of an 8-bit color image). See #49. It would look something like this
> 
> ```python
> import cv2 as cv
> vcap = cv.VideoCapture("rtspsrc location=rtsp://192.168.1.10/color latency=0 ! decodebin ! videoconvert ! appsink sync=false", cv.CAP_GSTREAMER)
> ret, imageBGR = vcap.read()
> cv.imshow('image', imageBGR)
> cv.waitKey(1)
> vcap.release()
> cv.destroyAllWindows()
> ```
> 
> Finally, we provide [ros_kortex_vision](https://github.com/Kinovarobotics/ros_kortex_vision) that allows you to access both streams. There, you will find more documentation and examples on how to use it. If you must use both color and depth stream and you want to access/use the image data (and not simply stream the images), this is probably the easiest way to do it.
> 
> To answer your second question, both intrinsics and extrinsics are fixed fixed to every Gen3 arm.
> 
> Regards, Felix

The example you refer do not work but I can use the RGBD stream with https://github.com/Kinovarobotics/ros_kortex_vision. Thanks a lot!

In the meantime, do you know the easiest software/way that can achieve a hand-in-eye calibration with Kortex Gen3/Gen Lite arm? We probably need to calibrate the arm with a high quality camera. I am trying this package: https://github.com/IFL-CAMP/easy_handeye/issues/41.

felixmaisonneuve, 2021-11-15:
There is no particular software used to get the hand-in-eye calibration. I've been told you can use the [calibrateHandEye()](https://docs.opencv.org/4.5.1/d9/d0c/group__calib3d.html#gaebfc1c9f7434196a374c382abf43439b) function available with openCV, but looking at the description of the ROS package you sent, this is already what it is using
>  compute the eye-on-base or eye-in-hand calibration matrix through the OpenCV library's Tsai-Lenz algorithm implementation

Whatever you choose, you will probably need the end effector position for the calibration, which you can get with `tool_pose_x/y/z` and `tool_pose_theta_x/y/z` in the [BaseFeedback](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/messages/BaseCyclic/BaseFeedback.md#). If you are using ROS, there a `base_feedback` topic which publishes this information in real time. 

Best,
Felix

felixmaisonneuve, 2022-06-02:
I will assume I answered your question and close this issue. 
If you still need support, make sure to re-open the issue.

Regards,
Felix

closed 2022-06-02 by felixmaisonneuve