Kinovarobotics/Kinova-kortex2_Gen3_G3L #89: Joint speed commands cannot send at 40 Hz in High-level move contorl

JDchen-Jaden, 2021-11-10:
I wish to implement a while loop to update command velocity sent to each joint at each time step. I referred to this example: [https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/102-High_level_movement/04-send_joint_speeds.cpp](url)

However, when I call `example_send_joint_speeds()` in a while loop in `main()`, I noticed that the time step is 75 ms rather than 25 ms. Then I tried to copy the body of the code (as follows) in `example_send_joint_speeds()` to the `while` loop inside `main()` instead call the example function, the time step was reduced to 50 ms. I still have trouble sending the joint speed command at 40 Hz. It would be appreciated if anyone would like to help. Thanks!

```
int main()
{
while( condition)
{
            joint_speeds.clear_joint_speeds();
            for (size_t i = 0 ; i < qdot_cmd.rows(); ++i)
            {
                auto joint_speed = joint_speeds.add_joint_speeds();
                joint_speed->set_joint_identifier(i);
                joint_speed->set_value(qdot_cmd[i]);
                joint_speed->set_duration(1);
            }           
            base->SendJointSpeedsCommand(joint_speeds);
}
}
```

felixmaisonneuve, 2021-11-10:
Hi @JDchen-Jaden,

Normally, you would want a `std::this_thread::sleep_for(std::chrono::milliseconds(some_value));` in a continuous while loop like yours so it doesn't pump your cpu to 100%. I am also surprised your loop is taking 50ms (which is very slow for an infinite loop). I assume something is blocking the thread to prevent your cpu from jumping to 100%.

However, when using High Level commands, you do not need to send a command every 25ms. For instance, with jointSpeed commands, you can send a JointSpeed one time and the arm will keep moving at that speed until another command is received. I don't think you need `base->SendJointSpeedsCommand(joint_speeds);` to be in a loop to send a different JointSpeed every 25ms.

If that's what you need (different JointSpeed every 25ms), I would first recommend using Low Level, which will probably better suit your needs. If you want to stay High Level, I would try to add a `std::this_thread::sleep_for(std::chrono::milliseconds(25));`, because a 50ms loop is crazy slow, so I am pretty sure there is something slowing your loop rate. Maybe it is not the loop but the base that isn't able to send a command faster than 50ms... Anyway, I don't think High Level was intended to be used this way, so I am not sure of the results of this.

Regards,
Felix

JDchen-Jaden, 2021-11-10:
Thanks! @felixmaisonneuve,

I'll try low-level control in the end. But for now, I would like to see if I can figure this out in the high-level control mode. Just in case, I would try low-level when I have a better understanding of this API. 

I still have some confusion: 
(1) As I mentioned, when I call the example function (i.e., `example_send_joint_speeds()`) in `main`, the `while` loop takes 75 ms. When I directly copy the body of `example_send_joint_speeds()` in `main` instead of calling this function, the `while` loop takes 50ms. What makes the 25 ms differences? 
(2) In my while loop, I call `auto feedback = base_cyclic->RefreshFeedback();`. And write the received `feedback` information into a .txt file. Executing these commands should not take an extra 25 ms. I am not sure if `base_cyclic` will take an extra 25 ms. My guess is not, but I am not sure.

Again, thank you very much!

Best regards,
Jaden

JDchen-Jaden, 2021-11-10:
Sorry for the misleading info. `auto feedback = base_cyclic->RefreshFeedback()` is executed in the `while` loop in `main()`

felixmaisonneuve, 2021-11-10:
1. I don't know why it is different, my guess is that something in the computer is intentionally slowing your loop. 
2. It is possible that `base_cyclic->RefreshFeedback()` takes a bit of time, since it refresh every value for the base, the gripper and every actuators.

Why are you trying to send a different command every 25ms?

JDchen-Jaden, 2021-11-10:
Thanks! @felixmaisonneuve,

1, What I did is that I simply copied and pasted the function body into the `while` loop in `main`. It is weird. Since the reduced 25 ms matches the control frequency exactly. 
2, I am a bit surprised that the delayed loop time is exactly 50 ms. It seems Kinova runs an additional loop somewhere. If some code takes computation time like `base_cyclic`, it should not be 25 ms this long. 

I am trying to implement a simple admittance control at a high level. It requires the velocity updated according to input force in a while loop at 40 Hz (a fixed frequency, as you mentioned).

JDchen-Jaden, 2021-11-10:
Hi @felixmaisonneuve 

I've tried my best but still could not find out which parts take an extra 25 ms. There must be a command that Kinova run an additional loop. Could you please check the code? https://github.com/JDchen-Jaden/Gen3_test.git

Thank you very much for your time!

Best regards,
Jaden

felixmaisonneuve, 2021-11-10:
Since you want to create your own admittance control, I would highly recommend using low level control (which is not more complicated than high level and is well documented). 

What I think is happening is you are using blocking calls (e.g. `base->SendJointSpeedsCommand(joint_speeds)`), which means you have to wait for the next loop in the base (every 25ms) before moving to the next statement. see [this](https://github.com/Kinovarobotics/kortex/blob/master/linked_md/cpp_api_mechanism.md) for more info about blocking calls.

In your `main`, your first call (`auto feedback = base_cyclic->RefreshFeedback();`) takes 25ms (since it has to wait for the next base feedback), than you have `base->SendJointSpeedsCommand(joint_speeds);`, for which you have to wait another 25ms (for the next loop in the base). This is why it takes 50ms when copying everything in your `main` function. 
When you call `example_send_joint_speeds`, you add another 25ms because you use `int actuator_count = base->GetActuatorCount().count();`. 

Basically, for each blocking call to the base, you have to wait another 25ms for the next feedback (which is at 40Hz). 

This is (one of the reason) why you should consider using low level to implement your own admitance. We have [this example](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp) on how to use torque control in low level, and [this other example](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/200-Actuator_low_level_control/01-actuator_low_level_velocity_control.cpp) that also shows how to use low level servoing. 

I don't know if this is a clear explanation. I'll try to be more specific if it still doesn't make sense to you.

JDchen-Jaden, 2021-11-10:
Thank you so much for your quick response! @felixmaisonneuve 

It made perfect sense! Thanks a lot. 

I'll go with low-level control. But for curiosity, is there any way I can call two blocks in parallel to make is to be 25 ms? 

Best regards,
Jaden

felixmaisonneuve, 2021-11-10:
In the link I sent about the [API Mechanism](https://github.com/Kinovarobotics/kortex/blob/master/linked_md/cpp_api_mechanism.md), it mentions every API call can be called using `_callback` or `_async`. I haven't tested, but I suppose they will be executed in the same loop (in the base) and you will probably receive every answer at the same time (after 25ms). 

Best,
Felix

JDchen-Jaden, 2021-11-10:
Thanks! @felixmaisonneuve 

I am still having trouble finding a better way to get real-time sensor readings in low-level and high-level control. 

(1), Is there a better way to get sensor data in a loop compared to `auto feedback = base_cyclic->RefreshFeedback();`?
(2), I understand that callback example can print out the sensor data. How to return and store the sensor data for further computation purposes when calling `Refresh_callback(base_command, lambda_fct_callback, 0)`. Thanks!!

```
// Callback function used in Refresh_callback
auto lambda_fct_callback = [](const Kinova::Api::Error &err, const k_api::BaseCyclic::Feedback data)
{
    // We are printing the data for example purpose
    std::string serialized_data;
    google::protobuf::util::MessageToJsonString(data, &serialized_data);
    std::cout << serialized_data << std::endl;
};

base_cyclic->Refresh_callback(base_command, lambda_fct_callback, 0);
```

Thanks,
Jaden

felixmaisonneuve, 2021-11-11:
**(1), Is there a better way to get sensor data in a loop compared to auto feedback = base_cyclic->RefreshFeedback();?** You could use `Refresh()`, that allows you to send a command and receive base feedback at the same time, but no, refresh methods to get sensor data. What is the problem exactly with using `RefreshFeedback()`?

**(2), How to return and store the sensor data for further computation purposes when calling Refresh_callback(base_command, lambda_fct_callback, 0)** I suggest looking at lambda functions documentations (e.g. [this](https://docs.microsoft.com/en-us/cpp/cpp/lambda-expressions-in-cpp?view=msvc-170)). You probably want to use class data members : 
```c++
void doSomethingWithData(const k_api::BaseCyclic::Feedback feedback)
{
    // Doing stuff
    joint1_velocity = feedback.actuators(1).velocity;
}

main
{
// Callback function used in Refresh_callback
auto lambda_fct_callback = [](const Kinova::Api::Error &err, const k_api::BaseCyclic::Feedback data)
{
    doSomethingWithData(data);
};

base_cyclic->Refresh_callback(base_command, lambda_fct_callback, 0);
}
```

JDchen-Jaden, 2021-11-11:
Thanks a million! @felixmaisonneuve 
(1), `RefreshFeedback()` works well but takes 25 ms. As you suggested, I should try low-level control. 
(2), I applied a similar callback function in my high-level example. Applying `Refresh_callback()` in the  `while` loop in `main` makes the while loop run even faster than 1k Hz (high-level), though the `PORT' is defined as '10000` and ServoingMode is `SINGLE_LEVEL_SERVOING`. Maybe I should apply something like this `if(now - last > 25000)`.

Thanks again!
Jaden

closed 2021-11-11 by JDchen-Jaden