Kinovarobotics/Kinova-kortex2_Gen3_G3L #90: Cannot send velocity command in low-level control mode

JDchen-Jaden, 2021-11-11:
I encountered a couple of problems when implementing the low-level velocity control: 

(1), 
- The first block of code works. However, when I call `Refresh()` to send time-varying position velocity in a loop (around 2 ms), the first method (position command) gives a twitch or shake movement. 
- The second block of code does not work. It seems the velocity command cannot be sent to base. 

```
        for(int i = 0; i < actuator_count; i++)
        {
            q_cmd[i] += (h * q_dot_command[i]);
            base_command.mutable_actuators(i)->set_position(fmod(q_cmd[i], 360.0f));
        }
        feedback = base_cyclic->Refresh(base_command,0);
```
```
        for(int i = 0; i < actuator_count; i++)
        {
            base_command.mutable_actuators(i)->set_velocity(q_dot_command(i));
        }
        feedback = base_cyclic->Refresh(base_command,0);
```
(2), Any information that related to the robot end effector (task space), e.g. `feedback.base().tool_twist_linear_x();`, `feedback.base().tool_pose_x();` cannot updated in a loop. The output is simply the initial state. 
But the sensor information (e.g. `feedback.actuators(0).position()` ) for each actuators can be properly read and updated

(3), What is the difference between `Refresh(base_command,0)` and `Refresh(base_command)`? 

Sorry for asking so many questions at once. Thank you for your time!

felixmaisonneuve, 2021-11-12:
1. This is normal behaviour. see issue #42. We recommend using position control calculating the next position based on desired velocity : `next_position = current_position + velocity * 0.001s`
2. In low level control, the base is completely ignored and it does not do any computation. The base only transit messages directly to the actuators and the end effector. The kinematics used to get `feedback.base().tool_twist_linear_x();` and 
`feedback.base().tool_pose_x();` is never computed, so values stay empty. Available informations about the end effector are obtained via the [Feedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/InterconnectCyclic/Feedback.md#) with `feedback.interconnect()`.
3. You can specify a *device identifier* to communicate directly with a specific device, see [this](https://github.com/Kinovarobotics/kortex/blob/master/linked_md/cpp_device_routing.md#other-services). This is an optional parameter and I think the value is 0 by default, so I don't think there is a difference in this case. An example would be when calling `m_pActuatorConfigClient->ReadTorqueCalibration();`. This will ask the calibration for the base (which doesn't have a torque calibration), so it will return 0. For this function, you want to specify the *device identifier* (0 for the base, 1-7 for joints) and call  `m_pActuatorConfigClient->ReadTorqueCalibration(1);`

Best,
Felix

JDchen-Jaden, 2021-11-12:
Thanks! @felixmaisonneuve 

I have no further questions for 2 and 3. 

For 1, I still have the following questions:

**(A)**, In my application, I have to update the velocity command based on the measured EE force in a loop. To this end, the time step between each call of `Refresh` is not exactly 0.001 s. In my case, it is around 0.0012- 0.002 s. So I use the real-time step ( `h` ) instead of 0.001 to calculate the next position: `next_position = current_position + velocity * h`. **It works, but as it results in a very twitchy movement. The robot shakes, and the move is not smooth.**.

**_How to ensure a smooth and accurate move using position control?_** Thanks!

**(B)**, In #42, they can at least observe the drift when sending the velocity command. In my case, the robot **does not even move** if I execute `base_command.mutable_actuators(i)->set_velocity(q_dot_command(i));`. 

I was wondering if the problem is with the hardware. So I check the **high-level joint speed control**. **The robot cannot move to the desired position**.  Please refer to the following example code and the results:

> Example: Start configuration: _all joint position = 0_;
```
void example_send_joint_speeds(k_api::Base::BaseClient* base)
{
    k_api::Base::JointSpeeds joint_speeds;
    std::vector<float> speeds;
    speeds = {-14.0f, 18.0f, -18.0f -6.0f, 0.0f, -4.0f, 0.0f};
        for (size_t i = 0 ; i < speeds.size(); ++i)
        {
            auto joint_speed = joint_speeds.add_joint_speeds();
            joint_speed->set_joint_identifier(i);
            joint_speed->set_value(speeds.at(i));
            joint_speed->set_duration(1);
        }
        base->SendJointSpeedsCommand(joint_speeds);
        // Wait 5 seconds
        std::this_thread::sleep_for(std::chrono::milliseconds(5000));
    // Stop the robot
    std::cout << "Stopping the robot" << std::endl;
    base->Stop();
}
```
> If above speed control works properly, the desired joint position after 5s should be {-70, 90, -90, -30, 0, -20, 0}. However, the obtained results appear as follows. Only Joint 2 moved to the desired position. Joints 4 and 6 did not even move.  Position of Joint 1 and 3 has deviations. Could you please suggest any solutions? Thanks!
[image]

**(C)**, Assuming no gravity effects, is it possible to send position command to specific joints but velocity or torque command to other joints simultaneously? For example, 
```
            q_cmd[i] += (0.002 * q_dot_command[i]);
            base_command.mutable_actuators(0)->set_position(fmod(0.0f, 360.0f));
            base_command.mutable_actuators(1)->set_position(fmod(0.0f, 360.0f));
            base_command.mutable_actuators(2)->set_position(fmod(0.0f, 360.0f));
            base_command.mutable_actuators(3)->set_velocity(0.5f);
            base_command.mutable_actuators(4)->set_velocity(0.5f);
            base_command.mutable_actuators(5)->set_velocity(0.5f);
            base_command.mutable_actuators(5)->set_torque_joint(something);

            base_cyclic->Refresh(base_command);
```

felixmaisonneuve, 2021-11-15:
(A) Have you tried using a constant time-step instead? Sending a position variation based on the delay of the previous cycle means you will ask your arm to accelerate/decelerate every ms. Those acceleration are not perfect, meaning your robot will have a jerky motion. I would assume every message has a perfect 1ms cycle and see from there if the movement is precise and smooth enough.

(B) Can you double check that your joint was correctly set to VELOCITY ControlMode (each joint needs to be set seperately). 

(C) Yes, you need to set each joint ControlMode. I think this might be why you have weird behaviour. By default, it is set to POSITION.

JDchen-Jaden, 2021-11-15:
Thanks! @felixmaisonneuve 

(A), I've tried the constant time-step. Yes, it results in a smoother movement but is extremely slow, which is not correct. That is mainly due to the big divination between the actual time-step to the constant time-step. 

(B), I've double-checked and made sure it is under speed control. The` joint_identifier() `and `speeds `value are updated properly. Then the joint speeds are sent via `SendJointSpeedsCommand()`. I'll reboot the robot and see if it can solve the problem. 
```
    speeds = {-14.0f, 18.0f, -18.0f -6.0f, 0.0f, -4.0f, 0.0f};
        for (size_t i = 0 ; i < speeds.size(); ++i)
        {
            auto joint_speed = joint_speeds.add_joint_speeds();
            joint_speed->set_joint_identifier(i);
            joint_speed->set_value(speeds.at(i));
            joint_speed->set_duration(1);
        }
        base->SendJointSpeedsCommand(joint_speeds);
```
(C), I am a bit confused about your answer. May I double-check two things with you?

- Different actuators can be controlled under different control modes, i.e., speed, position, torque **simultaneously**. All these  commands for each actuator need to be combined as a single command as `base_command` and sent out via `Refresh()` 
- Joint ControlMode is set by `->set_position()`, `->set_velocity()`, and `->set_torque_joint`. 

            q_cmd[i] += (0.002 * q_dot_command[i]);
            base_command.mutable_actuators(0)->set_position(fmod(0.0f, 360.0f));
            base_command.mutable_actuators(1)->set_position(fmod(0.0f, 360.0f));
            base_command.mutable_actuators(2)->set_position(fmod(0.0f, 360.0f));
            base_command.mutable_actuators(3)->set_velocity(0.5f);
            base_command.mutable_actuators(4)->set_velocity(0.5f);
            base_command.mutable_actuators(5)->set_velocity(0.5f);
            base_command.mutable_actuators(6)->set_torque_joint(something);

            base_cyclic->Refresh(base_command);

felixmaisonneuve, 2021-11-16:
(A) Are you sure this time step is coming from the base? It might be difficult for a PC to compute everything and send a message every millisecond. More so if you have other processes running (internet browser, other programs, windows OS doing weird stuff on its own...) which seems to be the case here with your FT300 gripper also running. 

(B) Good

(C) You are right, I misread, you cannot put multiple control modes *simultaneously*. Why would you want to do that? I know some control modes take multiple input from the command. For example,  when in POSITION control mode, actuators will also look at the velocity and torque values, but I am not sure how this works (and if I knew, I am not sure I could share this information). The thing to know is that, in POSITION mode, position values will always be prioritized.

You can set ControlMode with `actuatorConfig->SetControlMode(someControlModeInfo)`, see [ActuatorConfig](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/ActuatorConfig.md). A command can have a position, velocity and torque for each actuator (but values might be ignored if not required).

JDchen-Jaden, 2021-11-16:
(A), No, the output time step is the execution time for a while loop shown in the end. I am a bit confused about how does the base time-step applies here? I am assuming the base time-step you mentioned is exactly 1 ms. I noticed that Kinova also uses `if` flow control in a `while` loop to set time-step to be 0.001 s in the _low-level velocity control example_. I am a bit confused about how does the base time-step apply?  

My understanding is that I have to _(1) read the force data_,  _(2) compute velocity_, _(3) assign the computed velocity to `base_command`_, and  _(4) send command to base_ in a while loop in `main`. To this end, I am very confused about how to keep the base sending the same velocity command at 1k Hz until the updated command is received at 100 Hz. In my case, the `Refresh()` call can only send at 100 Hz, due to the while loop taking around 0.01s. 

I am going to close #91 since it could be better to discuss here. If `callback` works in an asynchronous way, it could be a solution for this problem. But I have trouble implementing it due to the following reason: 
- I wish to read the force data and compute the velocity command in `callback`. However, how to pass the ` joint_speed_command` in the following function to `base_command`?
```
   // read force and compute velocity
   void read_and_compute(const k_api::BaseCyclic::Feedback feedback)
{
    joint_speed_command = some_func;
}

auto lambda_fct_callback = [](const Kinova::Api::Error &err, const k_api::BaseCyclic::Feedback data)
{
    read_and_compute(data);
};

main
{
    while(condition) {
        if(now - last > 1000) // set time-step to be 1 ms
        {
         base_cyclic->Refresh_callback(base_command, lambda_fct_callback, 0);
        }
    }
}
```

```
    start = GetTickUs();
    while(t <= loop_dur)  
    {
        now = GetTickUs();
        t = (double(now - start))/1000000; // to sec
        h = t - t0; // time step
        if(now - last > 10000)
        {
        //(1) read force
        F = come_forceReading_func();  

        // (2) computed velocity based on force 
        q_dot_command = some_velocity_computation_func(F, joint_position, joint_velocity)  

        // (3) update velocity command 
        for(int i = 0; i < actuator_count; i++)
        {
            q_cmd[i] += (0.001 * q_dot_command[i]);
            base_command.mutable_actuators(i)->set_position(fmod(q_cmd[i], 360.0f));
        }

        // (4) send command to base
        feedback = base_cyclic->Refresh(base_command,0);
        t0 = t;
        last = GetTickUs();
        } 
    } 
```

JDchen-Jaden, 2021-11-16:
Hi @felixmaisonneuve 
(C), I wish to implement a robust control algorithm. For stability testing purposes, it would be better to have a closed-form dynamic equation. So, we wish to study a 2 or 3 Dof configuration of Kinova instead of 7-Dof. To this end, it requires some of the actuators to be "locked". My original idea is to send "0" position commands to these joints and torque commands to the rest joints. If it does not allow assigning different control modes for actuators at the same time, do you have further suggestions?

If I specify the actuators to be torque-controlled, and only send command to these actuators, but do nothing to the rest actuators, are the rest actuators can be recognized as "locked' (or idle)? For example
```
// only send torque command to 2nd and 4th joints.
actuator_config->SetControlMode(control_mode_message, 2);
actuator_config->SetControlMode(control_mode_message, 4);
base_command.mutable_actuators(1)->set_velocity(commands[1]);
base_command.mutable_actuators(3)->set_velocity(commands[3]);

Refresh(base_command)
```

felixmaisonneuve, 2021-11-17:
(A) What I am trying to understand is how to mesure your 1.2-2ms. You should be able to send a new command every ms and the base is going to make sure communication with actuators happen at exactly 1000Hz. If you are not able to send a new command once every ms, then you program might not work correctly. 

Your while loop should not take 0.01s, it should take 0.001s since you need to send a command at every ms. 

You could use a global variable to update velocity from callback to the command. The low level actuator control example already has `velocity` as a global variable.

(C)
> My original idea is to send "0" position commands to these joints and torque commands to the rest joints
This is a good idea. 

Maybe there was confusion originally, so I will try to be clear here : 
You can put different ControlMode to different joints. For example, you could set joints 1,3 and 5 in TORQUE ControlMode, while joints 2,4,6 and 7 are in POSITION mode. Is that what you meant when you said simultaneously? 
See the torque control example, only the first actuator is in TORQUE ControlMode, while the other joints are still in POSITION : 
https://github.com/Kinovarobotics/kortex/blob/0aca9b23b0f04d76bf27e3f7b45befc2c77ebed9/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp#L240

What you cannot do is put joint 1 in both POSITION and TORQUE ControlMode at the same time. This is what I understood when you said simultaneously.

Different joints can be in different mode at the same time.

JDchen-Jaden, 2021-11-19:
(A), `callback` method works well. Now,  the command can be sent less than or equal to 1 ms in the while loop.

(C), Thanks for the clarification! Yes, the above answers are what I mean by asking if it is possible to set different control modes for different actuators simultaneously

closed 2021-11-19 by JDchen-Jaden