Kinovarobotics/Kinova-kortex2_Gen3_G3L #91: Kortext PAI asynchronous call to pass 100 Hz data to 1k Hz control loop?

JDchen-Jaden, 2021-11-12:
Hello!

I am working on a project that requires the measured EE force by a 100 Hz F/T sensor to be sent to Kinova, then is used to compute velocity commands. I wish to implement it under the low-level control mode. All computation will be conducted in an asynchronous function to ensure the control frequency is exactly 1k Hz.. 

If my understanding is correct, the `Refresh()` will send the same velocity (within 0.01 s) command every 0.001s until the external force data and velocity command are updated every 0.01 s. To this end, how to pass the updated velocity command obtained in the asynchronous function to the `command` in this function call: `std::future<Feedback> Refresh_async(const Command& command, uint32_t deviceId = 0, const RouterClientSendOptions& options = {false, 0, 3000});`

Thanks. If you think the above approach does not work, could you please suggest a better way?

felixmaisonneuve, 2021-11-15:
Are you able to access your FT-300 force?

Communicating with a FT-300 via Kortex is not supported. You have to use an external cable diretcly connected to your PC, so I am not sure how you are planning to to do all of this

JDchen-Jaden, 2021-11-15:
Hi @felixmaisonneuve 

Yes. FT-300 is connected to the PC. The data is sent to Kinova via UDP. 

As you suggested in #90, the position control requires the real time-step to be close enough to the constant time-step to be set in the code. So I wish to make sure computation time has no impact on command frequency. To this end, it seems that asynchronous calls could be a solution. But I am a bit confused about using the Kortex API Asynchronous mechanism in a way that I asked in the original issue. Could you please check it again and give some suggestions? Thanks!

Best regards,
Jaden

felixmaisonneuve, 2021-11-16:
I'm not sure what you want me to answer. 

Yes, I think it could work using async function call, but since you seem to do only one call to the API, `callbacks` could also be a good option. The async calls are prefered when doing multiple API calls, but it does what you want if I understand correctly. 

With async, you will need to `wait_for` the future at some point, which is why I think callback might be better.

closed 2021-11-16 by JDchen-Jaden