Kinovarobotics/Kinova-kortex2_Gen3_G3L #92: Experimental varification of STABILITY of certian torque-baed control algorithm

JDchen-Jaden, 2021-11-12:
Hello!

Is it possible to implement experimental validation of stability of a robust control algorithm under the low-level torque control? Under the low-level torque control, is there any mechanism that constraints (e.g., hard limitations) torque of each actuator? 

Thanks a lot!

felixmaisonneuve, 2021-11-15:
What are you trying to do exactly, I am not sure I understand what you mean by "experimental validation of stability of a robust control algorithm".

About limitations, there is no limitations in low level control, except hardware limitations. The only reason you cannot send a command to move an actuator to 1000 degrees/s is because the hardware is not able to. You cannot change these limits. 
I am not sure want you want to know here either.

JDchen-Jaden, 2021-11-15:
Thanks! @felixmaisonneuve 

Sorry for the confusion. "No limitations in low-level control" answered all my questions. I wondered if there are internal protection mechanisms (e.g., safe operation speed limitation, collision avoidance, etc.) that will be applied in low-level control. 

Thanks a lot!

closed 2021-11-15 by JDchen-Jaden