Kinovarobotics/Kinova-kortex2_Gen3_G3L #94: Google protobuf error when applying callback mechanism

JDchen-Jaden, 2021-11-22:
Hi!

I have trouble understanding the following error: `google::protobuf::FatalException` , `CHECK failed: (index) < (current_size_)`. I got this error when I implement a simple admittance control. I searched online and found that this error is similar to "index exceeds an array". Then, I tried to understand _"repeated_field.h"_, however, I could not even understand the functionality of this _.h_ file.

After debugging many times, I targeted the issue may being at` callback`. So, I convert my code to a shorter version that is easy for comparing and debug: https://github.com/JDchen-Jaden/Gen3_test/blob/main/12-debug_velocity_test02_noCallback.cpp

You may observe that the robot follows the velocity command (VELOCITY COMMAND) well when running this source code. This is because I applied `Refresh()` rather than `Refresh_callback()`.

If you simply replace **line 308** by `base_cyclic->Refresh_callback(base_command, my_callback, 0);` You should receive the following error.
```
[libprotobuf FATAL /home/jd-vmware/.conan/data/kortex_api_cpp/2.3.0-r.34/kortex/stable/package/9755f981ac44041fa0948f05e49abfa3d3346131/thirdParty/include/google/protobuf/repeated_field.h:1522] 
CHECK failed: (index) < (current_size_): terminate called after throwing an instance of 'google::protobuf::FatalException' what(): 
CHECK failed: (index) < (current_size_): Aborted (core dumped)
```
My callback function simply takes feedback as input and assigns the received data to some global variables. It is important to note that, the velocity command (**line 278**) does not depend on any feedback value. More specifically, this callback function has no contribution to the control. The reason I included it here is that I found passing feedback (though it has no impact on velocity command `q_dot_command`) to callback causes issues.

I hope the statements are clear and the code is helpful for you to understand the issue better. Thanks!

felixmaisonneuve, 2021-11-24:
Hi @JDchen-Jaden,

Protobuf is what we use to serialise the communication between the arm and the computer. *Repeated fields* is the equivalent of an array in protobuf, so yeah, this error is equivalent to an index out of range. 

The repeated field causing the error must be `feedback.actuators` (since it is the only one you are using). I am not sure what is hapenning here. I suppose `feedback.actuators` is empty, but I am not sure why this would happen. I will need to try your code to figure out what is happening, which I'm going to do today. 

In the mean time, you could something like this and see if the error persist
```c++
///////////////////////////////////////////////////////////////
// call back functions
void my_vac(const k_api::BaseCyclic::Feedback feedback){

    for (int i = 0; i < feedback.actuators.size(); i++)
    {
        if (i== 0)
        {
            q[0] = feedback.actuators(0).position()*M_PI/180.0;   // in rad
            q_dot[0] = feedback.actuators(0).velocity()*M_PI/180.0;  // in rad/s
            joint_tau[0] = feedback.actuators(0).torque(); // in N.m
        }
        else if (i==3)
        {
            q[1] = feedback.actuators(3).position()*M_PI/180.0;
            q_dot[1] = feedback.actuators(3).velocity()*M_PI/180.0;
            joint_tau[1] = feedback.actuators(3).torque();
        }
        else if (i == 5)
        {
            q[2] = feedback.actuators(5).position()*M_PI/180.0;
            q_dot[2] = feedback.actuators(5).velocity()*M_PI/180.0;
            joint_tau[2] = feedback.actuators(5).torque();
        }

        //q_dot_command << -5.0, 0.0, 0.0, 5.0, 0.0, 5.0, 0.0;//q_dot_command = {5.0f, 0.0f, 0.0f, 5.0f, 0.0f, 5.0f, 0.0f};//
    }
}
```

If the error really is `feedback.actuators` that is empty, I suppose this should get rid of the error, but it won't be very usefull when you will actually need `q`, `q_dot` and `joint_tau` to be populated to compute your movement. You could also had some kind of verification to see if the `feedback.actuators` is "half-empty". For example if there is only 3 feedback.actuators values
```c++
if (feedback.actuators.size() != 0 && feedback.actuators.size() < 7)
{
    print("feedback.acturators was half empty")
}
```

I don't think this is possible, you either have all the actuators value or none of them.

Anyway, I will try to test this and give you more feedback later today based on the tests I will do.

Best,
Felix

JDchen-Jaden, 2021-11-24:
Hi @felixmaisonneuve 

Thanks a lot! Your first approach solved the problem. No errors appear anymore. Right now, I think the Repeated fields error problem gets entirely resolved because it also works well for the position control, which shared the same error (Repeated fields) as the velocity control before.

For your second suggestion, I saved all 7 actuators' data to` q, q_dot, and joint_tau` instead of only 3, but the error was still there. 

A new error has occurred. After applying your first approach, the robot stopped after a very short period of movement. No errors appeared in the terminal. But if I check the Web App, there is a Warning for either actuator 1 or 4 (it's very random), or both (which case occur is random). The following pictures show the initial/last position of movement and the warning. I hope it is helpful to compare our results and see if the problem is on hardware. Thanks! 

**Initial configuration** 
[image]

**End configuration** 
[image]

**Warnning**
[image]
[image]

I'll continue trying and testing. 

Best regards,
Jaden

felixmaisonneuve, 2021-11-24:
Hi @JDchen-Jaden,

I am glad my solution worked. 

I tested your example and I found some interesting stuff. 

First, even if the joint is in VELOCITY control mode, it seems you need to constantly update your position in your command. 
```c++
if(i==0 || i==3 || i ==5)
{
     base_command.mutable_actuators(i)->set_position(feedback.actuators(i).position());
     base_command.mutable_actuators(i)->set_velocity(q_dot_command[i]);
}
```
In this, commenting `base_command.mutable_actuators(i)->set_position(...)` will make the arm stop after 1 second, I am not sure why. I was certain position was not required in VELOCITY control mode, but it seems we need to send a value for the program to work.

Anyway, this is the error you are currently having. Your `feedback` is never updated when using `Refresh_callback` so the position never changes and the arm stops moving. To fix this, I added `feeback_` in your global variables. That way, I am able to update the feedback in the callback function.
```c++
///////////////////////////////////////////////////////////////
// call back functions
void my_vac(const k_api::BaseCyclic::Feedback feedback){

        feedback_ = feedback;

        // read actuators' information
```

You also need to comment `k_api::BaseCyclic::Feedback feedback;` and change every `feedback` with `feedback_` in your `main`. With this, you should be good to go.

On my side, I was not able to reproduce your original error with the repeated field. The arm would move normally (with the solution above) and the `google::protobuf::FatalException` never occured to me. I don't know why you are receiving empty actuator feedback, it works fine for me. Let's hope it won't be too much of a problem for you in the future, I don't know how much help I can provide when the same code is working for me.

Finally, I do have the warnings too (and the `joint fault` error). Maybe the arm is unable to stop fast enough (when we stop sending velocity command) or something and it creates a warning. I wouldn't touch anything for now since it seems to be working. We'll look into it if it becomes a problem, but for now you should be good to make it work. 

Best,
Felix

JDchen-Jaden, 2021-11-25:
Thanks! @felixmaisonneuve 

Thank you so much for the solution that enables updated `feedback`. It's good to know that the solution you provided works well on your Gen3. I also observed some improvements when applying your solution (please refer to the 3rd attached picture; it can move longer time till 3.7 s; it was less than 1s before). 

However, the same `google::protobuf::FatalException` error occurred again when I applied the same solution. You may see the error occur randomly in terms of time (the values above the error are time in seconds). (_The error also occurred even I commented out the `cout`. Printing out time has no contribution to the error. So I keep the `cout` here for showing when it stops._)

The following are my modification that strictly follows your suggestions. If I did not make any mistakes, something might have gone wrong with our robot control board. I may need to contact Kinova Support. Also, my WebApp shows **User Robot** and **User admin**. I am not sure why it has two user names there. Just in case, I attached the picture and hope it provides more useful information.  

```
//**** Global ******
k_api::BaseCyclic::Feedback feedback_;

// ****callback function****
void my_vac(const k_api::BaseCyclic::Feedback feedback){
    feedback_ = feedback;
          // read actuators' information
    q[i] = feedback.actuators(i).position()*M_PI/180.0;   // in rad
}

void my_callback(const k_api::Error& err,const k_api::BaseCyclic::Feedback& feedback)
{
    my_vac(feedback);
}

main() {
     // all feedback had been updated to feedback_
}
```

[image]

[image]

[image]

[image]

Best regards,
Jaden

felixmaisonneuve, 2021-11-26:
I think the `google::protobuf::FatalException` is not related to the code itself, since it never occured to me, even after a couple hours playing with an arm. With the modifcations you did, I think everything should work and your arm is supposed to move for 4 seconds (based on your condition for the while loop).

What setup are you using? Are you plugged in directly in your arm with an ethernet cable? Are you using a switch? are you using wifi?
Personnally, I was testing with an ethernet cable connectly directly between my computer and the arm. The setup might affect the communication and create errors, so I would try connecting directly to the arm and see if it is still a problem. 

For the User Robot, I suppose you might have created a second user at some point. You can see all users in the webapp in the section "Users". I don't think it does anything, so I wouldn't mind it. 

If you think may have issues, I would make sure your arm is using our most recent firmware version (version 2.3.0-r.227), and, if you are, maybe do a factory reset of the arm. But be carefull, doing a factory reset will wipe every configuration you have made, so you might want to save your actions, protected zones or any other changes before doing it.

Regards,
Felix

JDchen-Jaden, 2021-11-26:
Hi @felixmaisonneuve 

I really appreciate that you spent tons of time testing everything!!

I reset the arm (push the power button for 10 s)  a couple of times yesterday. And it did not work out. 
- The Gen3 is directly connected to the PC **by ethernet cable**.
- The firmware has been updated to the **most recent** version before (please refer to the picture attached).
- I am using **Ubuntu 18.04 on a VMware virtual machine**.
- There is only one user that is **admin** shown there. However, when the error of `google::protobuf::FatalException` occurs, the warning is on **User Robot** rather than **User Admin**. Also, I have no idea the meaning of 6 years since the robot arrived here in 2020. 

Since my advisor worries about the controller board of Gen3, I'll forward this issue to Kinova Support. It would be very helpful if you could also be in the support team to check the problem. Thanks again!

Best regards,
Jaden 

[image]
[image]

[image]

felixmaisonneuve, 2021-11-26:
Hi @JDchen-Jaden,

Can you try using a real Ubuntu computer instead of a VM? 

We have had some problems with our Gen2 arms (older arms) when using a VM. I think it might be worth to try using a real computer, especially since this is communication related.

Best,
Felix

JDchen-Jaden, 2021-11-26:
Hi @felixmaisonneuve 

Our University does not support Dual-boot and Ubuntu. What I can do is to test it on Windows. Meanwhile, I'll dual-boot my laptop and see if it works out. Thanks a lot for the suggestions!

Best regards,
Jaden

JDchen-Jaden, 2021-11-27:
Hi! @felixmaisonneuve,

I think I found the reason that cause the communication issue. I have two network settings for Ubuntu VMware: (1) internet (**ens33**) (2) ethernet (**ens38**). If I disconnect the internet, your solution works well and no errors occurred. However, if I need to build the project using `cmake` I have to reconnect the internet for using Conan.

I have trouble figuring out why the internet connection has an impact on ethernet. Could you please suggest a solution if you also have experience of using Virtual machines. Thanks!

[image]

[image]

[image]

felixmaisonneuve, 2021-11-29:
Hi @JDchen-Jaden,

It is unfortunate that your University does not allow dual-boot. In theory, it is possible to build and run your program on windows. There is a couple options to build on windows that you can find in the [readme](https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/examples#build). When I have to build on windows I normally use mingw32 and I follow the steps described in the subsection *Windows - Using GCC*. However, Windows sometimes start random processes in the background that will slow your execution time. On windows, it is sometimes harder to do low level because of that. 

It is great that you have found a solution by disconnecting one of your interface. Both your interfaces looks good, I don't think there is a problem there. You could maybe look at your VMware network configuration and see if there is something weird with the network bridges. I don't think you will find anything thought, I blame the virtual environment. I think virtualization/network bridges does not keep up with the large amount of packets when you have 2 networks and you are doing low level, so you lose (or maybe they are delayed a bit too long) some packets and it creates your error. I don't know too much about VMs and how virtualization works on your network, so I really cannot help you on that. 

At this point, I don't know if you will be able to fix it entirely, so I would say you have three options : 
1. Use windows. Could be a problem in Low Level because Windows is bad, but that's my personnal opinion :)
2. Find a way to use a real Ubuntu setup instead of a VM
3. Disconnect your network and reconnect it everytime you need to compile.

It is just a mather of what suits you best.

Regards,
Felix

closed 2021-11-29 by JDchen-Jaden