Kinovarobotics/Kinova-kortex2_Gen3_G3L #95: INVALID_USER_SESSION_ACCESS error

otokatli, 2021-12-07:
## Description

While controlling the robot in low-level torque mode, I am occasionally getting INVALID_USER_SESSION_ACCESS error. A similar problem with ros_kortex is also happening when I am controlling my robot with ROS. 

## Version

KortexAPI : 2.3.0

Kortex Device : 2.3.0

felixmaisonneuve, 2021-12-09:
Hi @otokatli,

Can you copy the complete output of your terminal when you receive this error? 

You could try to increase the timeout duration of the connection incativity timeout and see if it helps. 
```
create_session_info.set_connection_inactivity_timeout(60000);
```
You can do a similar thing in ROS. You can either change the default value in the kortex_driver/launch/kortex_driver.launch file
```
<arg name="api_connection_inactivity_timeout_ms" default="60000"/> <!--milliseconds-->
```
or pass the argument when launching the driver
```
roslaunch kortex_driver kortex_driver.launch api_connection_inactivity_timeout_ms:=60000 ip_address:=...
```

Do you have an idea of what seems to be causing this issue? Does this happen when doing a particular sequence or does it seem to happen randomly? 

Finally, what setup are you using? Are you plugged directly to the arm using an ethernet cable? Do you use a switch? What is length of the ethernet cable?

Regards,
Felix

phuicy, 2021-12-15:
[image]

It seems to happen one in five times at random. But sometimes it will do it every time until the robot is reset.

felixmaisonneuve, 2021-12-16:
Hi @phuicy,

Does this also happen when running our low level examples?

phuicy, 2021-12-16:
Yes, but only occasionally

phuicy, 2021-12-16:
and connecting ros_kortex

felixmaisonneuve, 2021-12-16:
This is weird, I have never experienced anything like this, but we have had other users with the same issue. 
We are not able to reproduce it, so it is very hard for us to identify the issue.

What setup are you using? Are you plugged directly to the arm using an ethernet cable? Do you use a switch? What is length of the ethernet cable?

phuicy, 2021-12-16:
Switches, I have tried 1m and 10m Ethernet cables.

It feels like a packet (user/password) synchronization issue.

felixmaisonneuve, 2021-12-16:
Did you try without using the switches? By plugging directly into the arm?

phuicy, 2021-12-18:
I will have a go without the switch. But it is necessary for the use-case.

phuicy, 2022-01-05:
I have tried to use it with a direct connection, no luck same regularity and symptoms of issue.

felixmaisonneuve, 2022-01-05:
Do you have any code sample that I can use to reproduce the error? It is very hard for us to find the cause of this issue without being able to reproduce it.

Do you use the latest version of the SWU (arm's firmware package), of the API and of ROS?

phuicy, 2022-01-06:
It happens with your low level torque Cpp examples.

felixmaisonneuve, 2022-01-07:
That's also what we have been told, but I tried to run the actuator_low_level_velocity_control examples 15-20 times in a row without having issues. 
I also tried the same test while doing stuff in the webapp, and sending `ping` commands to the arm. I was never able to reproduce it. 
At some point I even disconnected the ethernet cable for 5 seconds when the arm started moving  (after moving to the home position) before plugging it back in. The am still managed to finish the movement successfully.

I know I already told you to change the `set_connection_inactivity_timeout`, but you could try to change the `set_session_inactivity_timeout` as well. 
https://github.com/Kinovarobotics/kortex/blob/0aca9b23b0f04d76bf27e3f7b45befc2c77ebed9/api_cpp/examples/200-Actuator_low_level_control/01-actuator_low_level_velocity_control.cpp#L262-L263

You could try different configurations like 
```
    create_session_info.set_session_inactivity_timeout(1000);   // (milliseconds)
    create_session_info.set_connection_inactivity_timeout(1000); // (milliseconds)
```
```
    create_session_info.set_session_inactivity_timeout(1000);   // (milliseconds)
    create_session_info.set_connection_inactivity_timeout(60000); // (milliseconds)
```
```
    create_session_info.set_session_inactivity_timeout(60000);   // (milliseconds)
    create_session_info.set_connection_inactivity_timeout(1000); // (milliseconds)
```
```
    create_session_info.set_session_inactivity_timeout(60000);   // (milliseconds)
    create_session_info.set_connection_inactivity_timeout(60000); // (milliseconds)
```

If you could find a configuration that makes this issue appear more often (or less often) could help us trying to identify what the problem is. Otherwise, it is very difficult for us to work on that without being able to reproduce it.

phuicy, 2022-01-08:
So, interestingly, I have tested my code on 3 of our other Kinova gen3's and the issue is only one of them, the oldest one, although they are all on the same firmware. hmmm. I may try a fresh install.

felixmaisonneuve, 2022-01-10:
That's interesting. Let me know how it goes

phuicy, 2022-01-17:
A factory reset. Didn't make a difference. :(

felixmaisonneuve, 2022-01-18:
When communicating with the arm, a thread is created that sends a message to the arm every 150ms. If the arm does not receive a message before the timeout limit, it will throw this error. I would expect this error to occur if the computer, for a reason or another, is not able to send that message. For example if the thread is killed or if the thread doesn't have CPU time because the load is too heavy for the computer (or VM). 

From what I understand, you have multiple arms and they are all using the same software/firmware package. And running the same program from the same setup will work on the recent arms, but not on the old one. This makes me think the problem might be hardware related (not sure how this is possible). 

After talking with some people, there shouldn't any difference in the hardware used on your old arm compared to your new ones. They should be identical. Maybe there is a defect, I don't know what hardware issue could cause such error however. 

I would suggest for you to contact support@kinova.ca. With your arm's serial number, they will be able to see the history of the arm and maybe see something unusual for one of the pieces in your arm.

I am sorry, but I don't know what more help I could provide here, since the software side doesn't seem to be the issue. 
I will probably close this issue in a couple weeks unless something new comes up

Best regards,
Felix

closed 2022-04-12 by felixmaisonneuve