Kinovarobotics/Kinova-kortex2_Gen3_G3L #96: 2ms of blocking baseCyclic->Refresh(command)

phuicy, 2021-12-15:
## Description

Currently, if you attempt to set torque for all seven joints, 

```cpp
  for (unsigned int i = 0; i < 7; i++) {
    // Save the current actuator position, to avoid a following error
    base_command->mutable_actuators(i)->set_command_id(frame_id);
    base_command->mutable_actuators(i)->set_position(fmod(p[i]*180.0f/M_PI , 360.0f));
    base_command->mutable_actuators(i)->set_torque_joint(command[i]);
  }
  try {
    auto base_feedback = base_cyclic->Refresh(*base_command);

```
the blocking Refresh call blocks for about 2ms.

What is the correct way to approach closed loop control?

* Getting current pose from callback, with proper thread management of course:

```cpp
  for (unsigned int i = 0; i < 7; i++) {
    // Save the current actuator position, to avoid a following error
    base_command->mutable_actuators(i)->set_command_id(frame_id);
    base_command->mutable_actuators(i)->set_position(fmod(p[i]*180.0f/M_PI , 360.0f));
    base_command->mutable_actuators(i)->set_torque_joint(command[i]);
  }
  auto lambda_fct_callback =
      [this](const Kinova::Api::Error &err,
             const Kinova::Api::BaseCyclic::Feedback data) {
        // We are printing the data of the moving actuator just for the example
        // purpose, avoid this in a real-time loop
        this->get(&data);
      };
  try {
    base_cyclic->Refresh_callback(*base_command, lambda_fct_callback, 0);
```

or

* Can you use ```RefreshCommand``` at the end of one loop, and an overlapping ```RefreshFeedback``` at the beginning of the next loop?

Basically, can the protobuf service respond to multiple overlapping commands?

Is this the best way to do 7 joint torque control?

## Version 

KortexAPI : 2.3.0

Kortex Device : 2.3.0

felixmaisonneuve, 2021-12-16:
Hi @phuicy,

The best way to use low level control is definetely by using callbacks. 

Personnaly, I would use `Refresh_callback` like in your first example. I suppose using `RefreshFeedback` on top of `RefreshCommand` is possible and will probably acheive the same result. However, I think it is easier and more convenient to do both the `Command` part and the `Feedback` part in a single command. At this point, I am pretty sure both method are identical and it is up to you to decide what you prefer. I don't think `Refresh` was implemented in a "special" way that makes it more advantageous then using 2 seperate command. I think it combines both functions for ease of use. 

> can the protobuf service respond to multiple overlapping commands?
Why would the protobuf service would overlap commands? The way I see it, (almost) the same amount is sent in both methods. Every message will be sent one after another. I don't see why there could be conflicting commands, but maybe I am missing something.  

Regards,
Felix

phuicy, 2021-12-16:
> Why would the protobuf service would overlap commands?
If the robot is taking 2ms to respond to ```RefreshCommand_Callback```, if i send it another ```RefreshCommand_Callback``` or ```RefreshFeedback_Callback``` in 1ms what would the response be? Blocking on the server side until the ```RefreshCommand_Callback``` has been completed, and then responding to the ```RefreshFeedback_Callback```.
If I am sending ```RefreshCommand_Callback``` at 1ms intervals, but they are taking 2ms to respond, what is happening internally, is the robot responding to commands in 1ms? Is a new command interrupting the old one?

felixmaisonneuve, 2021-12-16:
The second command, `RefreshFeedback_Callback` will also take 2ms before receiving the answer, but it does not add up to the previous `RefreshCommand_Callback` delay.

Lets say so send `RefreshCommand_Callback` at 0ms, then you send `RefreshFeedback_Callback` at 1ms, you will receive  `RefreshCommand_Callback` answer at 2ms and `RefreshFeedback_Callback` answer at 3ms. 

Your 2ms delay comes from the synchronization between all the components. 
Let me try to illustrate what is happening. 
The base has a 1kHz loop, so it checks if it received a command at 0ms, 1ms, 2ms, 3ms.... and sleeps in between.
Now, if you send a command at 0.1ms, the base will only process it at 1ms. Then it will sync with all the actuators (either relay the command or fetch data), and the base will send its callback at about 1.5ms. By the time you received the callback, a little less than 2ms (I think the real number is 1.4-1.6ms) has passed since you sent your original command, and that's where your delay comes from. 

Now that I think of it,  using `RefreshFeedback` on top of `RefreshCommand` might be problematic, since the base will only process 1 command per ms if I am not mistaken. It could be worth to test it and see what happens. 

If you send `Refresh_callback` every ms, you should receive a callback every ms also, but it will be shifted by 1-2ms from your original command. 

Does that make sense?

phuicy, 2021-12-16:
That makes perfect sense, and exactly what I wanted to know.

closed 2021-12-16 by felixmaisonneuve