Kinovarobotics/Kinova-kortex2_Gen3_G3L #97: Robot not listerning to torque low level command

phuicy, 2022-01-05:
## Description

I am sending the torque cyclic commands at 1khz but the robot ignores the torque command input.

## Version

At a minimum, provide the Kortex API and Kortex-enabled device versions.

KortexAPI : 2.9.0-r.15

Kortex Device : latest

## Steps to reproduce

```cpp

void KinovaRobot::set(Eigen7f command) {
  auto frame_id = base_command->frame_id() + 1;
  base_command->set_frame_id(frame_id);
  if (frame_id > 65535)
    base_command->set_frame_id(0);
  auto p = this->get_pose();
  for (unsigned int i = 0; i < KinovaRobot::actuator_count; i++) {
    base_command->mutable_actuators(i)->set_command_id(frame_id);
    base_command->mutable_actuators(i)->set_position(fmod(p[i], 360.0f));
    base_command->mutable_actuators(i)->set_torque_joint(command[i]);
  }
  auto lambda_fct_callback =
      [this](const Kinova::Api::Error &err,
             const Kinova::Api::BaseCyclic::Feedback data) {
        // Set current pose, and velocity
        this->get(&data);
      };

  base_cyclic->Refresh_callback(*base_command, lambda_fct_callback, 0);
}

```
is being run at a steady 1kHz.

After having moved into Torque low level mode:

```cpp
void KinovaRobot::torqueMode() {
  auto servoing_mode = Base::ServoingModeInformation();
  // Set the base in low-level servoing mode
  servoing_mode.set_servoing_mode(Base::ServoingMode::LOW_LEVEL_SERVOING);
  base->SetServoingMode(servoing_mode);
  using namespace std::chrono;
  auto start = high_resolution_clock::now();
  auto base_feedback = base_cyclic->RefreshFeedback();
  auto stop = high_resolution_clock::now();
  auto duration = duration_cast<microseconds>(stop - start);
  RCLCPP_INFO_STREAM(rclcpp::get_logger("Runner"),
                     "Time to get feedback for " << this->name << ":"
                                                 << duration.count() << "us");
  this->get(&base_feedback);

  auto p = get_pose();

  // Initialize each actuator to their current position
  for (unsigned int i = 0; i < actuator_count; i++) {
    // Save the current actuator position, to avoid a following error
    p[i] = fmod(p[i], 360.0f);
    base_command->add_actuators()->set_position(p[i]);
  }
  auto lambda_fct_callback =
      [this](const Kinova::Api::Error &err,
             const Kinova::Api::BaseCyclic::Feedback data) {
        // set cached pose and velocity
        this->get(&data);
      };
  RCLCPP_DEBUG_STREAM(rclcpp::get_logger("Runner"),
                      "First position sending... :" << p);
  // Send a first frame
  base_cyclic->Refresh_callback(*base_command, lambda_fct_callback, 0);

  auto control_mode_message = ActuatorConfig::ControlModeInformation();
  control_mode_message.set_control_mode(ActuatorConfig::ControlMode::TORQUE);
  for (unsigned int i = 1; i <= KinovaRobot::actuator_count; ++i)
    actuator_config->SetControlMode(control_mode_message, i);
}
```

No matter what command I send it the robot seems to just fall under gravity.

Thoughts?

thanks,
Guy

phuicy, 2022-01-05:
OK, no I was wrong. The numbers just have to be much higher than expected. To hold joint 2 in position the monitoring says "12Nm", to get any  motion I need to send 100+ in set_torque_joint()

felixmaisonneuve, 2022-01-05:
Hi @phuicy,

Is your hand-guiding working properly? Also, have you tried our low level torque control example as is to see if it works? 
If hand-guiding and the example are working properly, the error is probably coming from your code.

If they do not work, you could try to set zero torque.

Best,
Felix

closed 2022-01-06 by phuicy