Kinovarobotics/Kinova-kortex2_Gen3_G3L #98: Torque data from kinova Gen3

jocacace, 2022-01-12:
Hi  guys,

I am addressing a strange behaviour using the kinova gen3. In particular, following the example: 01-torque_control_cyclic, I implemented a program to command desired torque to all the joints. After applying a torque command, I read the torque value on the same joint, and such torque differs a lot. I was wondering if the torque() function BaseCyclicClient class was intended to read the overall applied torque (with or without any compensation). 

Consider that I am appling a fixed force (for example 2 N/m), while the torque function returns me numbers ranging from [-1.8;-0.9]

Thanks,

felixmaisonneuve, 2022-01-13:
Hi @jocacace,

Have you tried using the example without modifying it? 
Joints are supposed to follow the command they are sent, so I assume the problem is coming from your code. 

Can you share your code, or part of it, in order for me to help you?

Regards,
Felix

th1991-01, 2022-04-05:
Hi, I also have a similar question. The example, `kortex/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp`, seems to work as expected. However, when I try applying a fixed torque to the last joint, I find that the measured value differs from the commanded one. I use a code modified from the example. The attached code is it.

There may be a problem in my code. I would be grateful if you could give me some advice.

[01-torque_control_cyclic.zip](https://github.com/Kinovarobotics/kortex/files/8417162/01-torque_control_cyclic.zip)

felixmaisonneuve, 2022-04-05:
Hi @th1991-01 

I did some testing on that. The torque received from the actuator's feedback (`base_feedback.actuators(0).torque()`) corresponds to the external torque applied on the actuator. It does not take into account the torque command previously sent. 
You will notice and variation if you manually apply a force on the arm, but you won't see any difference when sending different torque commands. 

Even when you are not sending any command, you might notice a torque value for certain joints caused by gravity (e.g. on joint 2 when the arm is in home position)

Hope this clarifies,
Felix

th1991-01, 2022-04-06:
Thank you so much for your advise.

To understand torque command, I check the below:
* [1] I run my code (attatched above) with command=0 [Nm]. If applying external torque by my hand, `base_feedback.actuators(6).torque()` varies accordingly.
* [2] I run my code with command=1 [Nm]. If stopping the rotation by my hand for a short time, `base_feedback.actuators(6).torque()` and command are almost the same.
* [3] I run my code with command=1 [Nm] without external force other than gravity. The last joint rotates at almost constant speed, and `base_feedback.actuators(6).torque()` is about 0.4 [Nm]. If using command=2 [Nm], then `base_feedback.actuators(6).torque()` is about 0.5 [Nm]. If using command=3 [Nm], then `base_feedback.actuators(6).torque()` is about 0.55 [Nm].

[1] and [2] are clear from your advise, and I can understand them. Thanks.

[3] is still a bit unclear for me. Does the gap between commanded and measured torques under constant velocity rotational motion come from something velocity-dependent such as induced electromotive force? In what form is the torque command applied to the motor? For example, is it current proportional to torque command? Where can I find related information?

felixmaisonneuve, 2022-04-06:
To me, the torque you read in [3] is simply because the torque readings are imprecise during movements. It might be caused by induced electromotive force (and/or other) and it will probably vary from one actuator to another. 

I think torque is proportional to current (I can't quite remember but I am pretty sure it is). I want more into details in the issue #77 

I can also link #38 that provides additional informations that is related to this topic. 

Hope this helps,
Best,
Felix

felixmaisonneuve, 2022-06-08:
Hi @th1991-01,

Do you have other questions regarding this topic?
I will close it in a couple weeks if I don't hear from you.

Best,
Felix

closed 2022-08-11 by felixmaisonneuve